evcc-io/evcc #19717: Prognose Solarertrag stimmt nicht nicht überein

marleyan333, 2025-03-12:
### Describe the bug

Wenn man im Reiter „Vorhersage“ das Feature „Solarprognose anhand der realen Produktionsdaten anpassen“ auswählt erscheint dort korrekterweise ein angepasster Wert. Dieser wird dann jedoch nicht in der Hauptansicht unter „heute verbleibend“ übernommen. Dort bleibt der alte (nicht angepasste) Wert stehen. 
Davon abgesehen funktioniert das Feature bei mir erstaunlich akkurat! 

[image]
[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
…
```

### Log details

```text
…
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.9

andig, 2025-03-12:
Stand heute ist das so.

andig, 2025-03-13:
If we do this next step would be to make this a backend setting if we want to integrate it with the planner (which we wanted to do originally).

closed 2025-03-14 by andig