evcc-io/evcc #19728: **Title:**   Negative Forecast Values When "Adjust Solar Forecast Based on Real Production Data" is Enabled with Open-Meteo and Efficiency Set

danielraffel, 2025-03-12:
### Describe the bug

When the "Adjust solar forecast based on real production data" setting is enabled, the forecasted values can sometimes be negative. This behavior is unexpected and seems to occur when using Open-Meteo as the data source with an efficiency setting of 78%

### Steps to reproduce

1. Enable "Adjust solar forecast based on real production data" in the forecast settings.
2. Use Open-Meteo as the solar forecast data source.
3. Set the efficiency value to 78%.
4. Observe the generated forecast over a period of time.
5. Notice that some forecasted values may fall into the negative range on low forecast days.

### Configuration details

```yaml
solar:
    - type: template
      template: open-meteo
      interval: 1h
      lat: 37.X
      lon: -122.X
      dec: 28
      kwp: 5.6
      az: -90
      efficiency: 78
    - type: template
      template: open-meteo
      interval: 1h
      lat: 37.X
      lon: -122.X
      dec: 27
      kwp: 2
      az: 90
      efficiency: 78
    - type: template
      template: open-meteo
      interval: 1h
      lat: 37.X
      lon: -122.X
      dec: 11
      kwp: 3.2
      az: 0
      efficiency: 78
```

### Log details

```text
[lp-1 ] DEBUG 2025/03/12 11:30:23 charge power: 0W
[lp-1 ] DEBUG 2025/03/12 11:30:23 charge currents: [0 0 0]A
[site ] DEBUG 2025/03/12 11:30:23 grid power: 0W
[site ] DEBUG 2025/03/12 11:30:23 pv 1 power: 613W
[site ] DEBUG 2025/03/12 11:30:23 battery 1 power: -10W
[site ] DEBUG 2025/03/12 11:30:23 battery 1 soc: 71%
[site ] DEBUG 2025/03/12 11:30:23 site power: 90W
[lp-1 ] DEBUG 2025/03/12 11:30:24 charge voltages: [2.1 2.1 2.1]V
[lp-1 ] DEBUG 2025/03/12 11:30:24 charger status: A
[site ] DEBUG 2025/03/12 11:30:24 solar forecast: accumulated 54.106kWh, produced -34.110kWh, scale -0.630
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.9

thecem, 2025-03-12:
Could you pls post TRACE Log for solar-template?

andig, 2025-03-12:
> produced -34.110kWh

Broken pv

closed 2025-03-12 by andig

danielraffel, 2025-03-12:
FWIW this is definitely not a broken PV array.

StefanSchoof, 2025-03-12:
Your PV Production is negative, but evcc want positive values:

> evcc uses positive (+) values for incoming currents (grid consumption, PV generation, house battery discharge)

https://docs.evcc.io/en/docs/reference/configuration/meters

danielraffel, 2025-03-12:
From what I understand, my EVCC is configured correctly, and my PV production shows a positive value when generating power. However, I suspect that setting the efficiency in the solar template might have caused a negative reading WHEN "Adjust solar forecast based on real production data" is enabled . The logs above illustrate this difference, depending on whether the efficiency was set to 78% or left unset. Below are fresh logs from when I am not using efficiency and you can see the data appears OK. It might be worth detecting a scenario where efficiency + historical data contribute to negative numbers and flag this for the user. It is particularly noteworthy when conditions change radically over a few days which is what happened here.

```
[lp-1 ] DEBUG 2025/03/12 13:09:37 charge power: 0W
[lp-1 ] DEBUG 2025/03/12 13:09:37 charge currents: [0 0 0]A
[site ] DEBUG 2025/03/12 13:09:37 grid power: 0W
[site ] DEBUG 2025/03/12 13:09:37 pv 1 power: 733W
[site ] DEBUG 2025/03/12 13:09:37 battery 1 power: -225W
[site ] DEBUG 2025/03/12 13:09:37 battery 1 soc: 70%
[site ] DEBUG 2025/03/12 13:09:37 site power: -125W
[lp-1 ] DEBUG 2025/03/12 13:09:38 charge voltages: [2.1 2.1 2.1]V
[lp-1 ] DEBUG 2025/03/12 13:09:38 charger status: A
[site ] DEBUG 2025/03/12 13:09:38 solar forecast: accumulated 2.185kWh, produced 0.200kWh, scale 0.092
```

premultiply, 2025-03-13:
No?

Battery is charging and some surplus is feed into grid.
Nothing special, nothing to care about.

andig, 2025-03-13:
Lets not confuse things. The forecast looks at pv, nothing else.