evcc-io/evcc #19744: Huawei kaskadierend AC-Laden bei zweitem WR. evcc schaltet AC-Laden aus

detlefheese, 2025-03-13:
Hallo evcc-Gemeinde,

Mir fällt seit einer Woche auf, dass evcc meinen Huawei-Luna-Speicher der an einem WR hängt und von dem kaskadierenden Wechselrichter nur geladen wird, wenn AC-Laden auf an gestellt ist.

Durch die evcc-Steuerung bei "Planladung" oder "Schnell" wird danach das **AC-Laden** WR2->WR1 immer deaktiviert. 

Das führt dazu, dass ich morgens wenn die Sonne mehr aus NO mit dem zweiten WR diesen Anteil ins Netz einspeise anstatt allen erzeugten Strom (WR1+WR2) für Hausbatterie oder Autoladen nutzen zu können.

Wenn ich in der Huawei-App das AC-Laden wieder aktiviere deaktiviert mir dies evcc wieder nach einem geplanten oder Schell-Ladevorgang.

Ich genieße das pausieren der Batterieentladung. Es scheint aber auch das AC-Laden (WR2->WR1) immer wie auszuschalten.

[image]

[image]

Gibt es andere Huawei-Nutzer die auch zwei Wechselrichter haben und Ähnliches beobachten?

Viele Grüße Detlef

CiNcH83, 2025-03-13:
Hast du in der Konfiguration das Flag `forceaccharging` auf `true` gestellt? Siehe [Huawei-Config](https://docs.evcc.io/docs/devices/meters#huawei)...

detlefheese, 2025-03-14:
@CiNcH83 ... das war es!

Heute geht es wunderbar. [image]

Jetzt wird die Erzeugung von beiden Wechselrichtern genutzt.

Ein **wunderbare** Community [image]