evcc-io/evcc #19752: Bind vehicle to a specific charger - UI

RTTTC, 2025-03-13:
I have 2 chargers and 2 vehicles and fairly often stars align just so, that EVCC is unable to recognise vehicles, or does it incorrectly.

My vehicles are configured in UI, while chargers/loadpoints are still in yaml (due to load management not implemented in UI fully yet) - and I am unable to dedicate a vehicle to a specific load point at this time, like it used to be possible when everything was in yaml.

I am always using same vehicles with their own loadpoints, so my feature request is to be able to dedicate a specific charger to a vehicle- either in vehicle config or charger config.
Thank you for the consideration !

andig, 2025-03-13:
That's what "default vehicle" has been available for for years.

closed 2025-03-13 by andig

VolkerK62, 2025-03-13:
I think, it is possible as usual, just the name of the vehicle ist different (id:x)
https://docs.evcc.io/docs/reference/configuration/loadpoints#vehicle
Check `evcc config`, to find out the name.

RTTTC, 2025-03-13:
@VolkerK62 Thank you for the suggestion, for example my Citroen shows up as this, when I check `evcc config`

`db:2 type:template map[accessToken:***** cache:15m capacity:43 icon:car maxCurrent:15 minCurrent:6 password:***** refreshToken:***** template:citroen title:Citroen E-C4 user:***** vin:*****]`

So I have tried to use title, did not work.
Then I have tried to use 
`vehicle: 2` - nope, then :
`vehicle: db:2` and it works. Thank you for your suggestion for this workaround! (not an id:x though)

@andig - my request is to have this vehicle/charger assignment done in UI , since yaml will eventually dissapear anyway.

andig, 2025-03-13:
Ah, why don't you just say so ;)

/cc @naltatis we don't have the default vehicle in loadpoint config UI yet?

VolkerK62, 2025-03-13:
within UI-loadpoint configuration a default vehicle can be defined

[image]

naltatis, 2025-03-14:
I wanted to keep the "add new loadpoint" process clean. But due to this discussion I'd think it would at lease be a good idea to offer an "show all settings" link/button even in the creating flow.