evcc-io/evcc #19761: [UI Config] Cannot add OCPP Charger: connector already registered: 1

aguilaair, 2025-03-13:
### Describe the bug

When adding a Wallbox Commander 2 (FW 6.x), the charger connects to EVCC's OCPP correctly, but then I cannot add it to my dashboard. I get the following error: `create failed: cannot create charger type 'template': cannot create charger type 'ocpp': connector already registered: 1`

### Steps to reproduce

1. Add Charger
2. Select Wallbox FW 6.x
3. Enter station id
4. Click Save, box populates with correct info, popup with error

[image]

### Configuration details

```yaml
[main  ] INFO 2025/03/13 20:40:42 evcc 0.200.9
[main  ] INFO 2025/03/13 20:40:42 using config file: /etc/evcc.yaml
[db    ] INFO 2025/03/13 20:40:42 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2025/03/13 20:40:43 connecting evcc-514717598 at tcp://hass.smart:1883

Konfiguration (/etc/evcc.yaml):

network:
  host: evcc.edm.sh
  port: 7070

interval: 30s

# trial token, valid until 2025-03-24
sponsortoken: *****

mqtt:
  broker: hass.smart:1883
  topic: evcc
  user: *****
  password: *****

circuits:
  - name: main
    title: Main Circuit
    maxPower: 7

vehicles:
  - name: XC60
    type: template
    template: volvo2mqtt
    title: XC60
    capacity: 18.8
    icon: car
    timeout: 720h
    vin: *****
  - name: XC40
    type: template
    template: volvo2mqtt
    title: XC40
    capacity: 78
    icon: car
    timeout: 720h
    vin: *****

meters:
  - name: grid
    type: template
    template: tesla-powerwall
    usage: grid
    host: battery.power
    password: *****
  - name: pv
    type: template
    template: tesla-powerwall
    usage: pv
    host: battery.power
    password: *****
  - name: battery
    type: template
    template: tesla-powerwall
    usage: battery
    host: battery.power
    password: *****
  - name: solar
    type: template
    template: fronius-solarapi-v1
    usage: pv
    host: inverter.power
    user: *****

Version: `0.200.9`
```

### Log details

```text
[ocpp  ] INFO 2025/03/13 20:42:46 charge point connected, registering: garaje1
[garaje-1] WARN 2025/03/13 20:43:12 failed triggering StatusNotification: cannot send request 3658716379, no client garaje exists
[garaje2-1] WARN 2025/03/13 20:56:49 failed triggering StatusNotification: cannot send request 3313779249, no client garaje2 exists
[ocpp  ] INFO 2025/03/13 21:02:39 charge point connected, registering: cambrils
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.200.9 (c2f31f14)

aguilaair, 2025-03-13:
When I check the DB, there are no chargers logged in it, so I'm not sure how connector 1 can already be registered

aguilaair, 2025-03-13:
Using the evcc config assistant works

naltatis, 2025-03-14:
We've addressed similar issues in the last days https://github.com/evcc-io/evcc/issues/19143, https://github.com/evcc-io/evcc/issues/19229.
@andig should OCPP cleanup work right now?

naltatis, 2025-03-14:
@aguilaair thanks. I'll come back to you if we can't reproduce.

andig, 2025-03-15:
@mfuchs1984 du bist hier so super im Thema... könntest Du Dir evtl. mal anschauen was es brauchen würde um den Connector erneut registrieren zu lassen? Im Prinzip gibts zwei Möglichkeiten:
- nach dem Context cancel räumen wir alle Konfiguration wieder ab (das bräuchte man wohl für "anonyme" Charger) oder
- wir erlauben einfach, dass bei erneuter Konfiguration die schon bestehenden Strukturen getroffen werden können

Crown3r, 2025-03-15:
Hey, hier auch gleiches Problem mit einer EON vbox bzw Vestel EVC04. Auch nach Werkseinstellung (Hard-Reset) zurücksetzen und evcc komplett Neuinstallation.

mfuchs1984, 2025-03-15:
@andig das wird leider zeitlich schwierig, kann also nichts versprechen.

aguilaair, 2025-03-15:
> Hey, hier auch gleiches Problem mit einer EON vbox bzw Vestel EVC04. Auch nach Werkseinstellung (Hard-Reset) zurücksetzen und evcc komplett Neuinstallation.

The only way to fix it is to use the .yaml file to configure the chargers

BastianBayer, 2025-03-21:
I´m facing the same problem with an Vestel EVC04 that I want to connect via OCPP. Also the the error code: cannot create charger type 'template': cannot create charger type 'ocpp': connector already registered: 1. I run EVCC as add on in Home assistant and did the config of the battery etc. via the experimental UI so if I open up my evcc.yaml the file is empty. How could I manually add the OCPP connection or is there any way to do that ?

aguilaair, 2025-03-21:
Easiest way is to use the encc assistant to generate the .yaml file for you tbh

Crown3r, 2025-03-22:
Since you have the EVC04, did already tried to connect your wallbox via modbus tcp? I was wondering, if this might work on your device.

SlapJackNpNp, 2025-03-22:
I have the sams problem right now. Running newest evcc in HomeAssistant and trying to add it with the gui.

BastianBayer, 2025-03-22:
> Since you have the EVC04, did already tried to connect your wallbox via modbus tcp? I was wondering, if this might work on your device.

There are several options for the EVC04 and it seems that I have the charger without modbus tcp.

> Easiest way is to use the evcc assistant to generate the .yaml file for you tbh, then you just put that in your empty config file

I have tried to add a new config manually after I deinstalled and installed the add on again. Then everything seemed to work till today. I now get uninterrupted timeouts from the charger. So by adding the OCPP charger manually before making any changes via the ui I recieved some data for a small time. So I will deep dive on that

amtssp, 2025-03-23:
Hi.
I have the same situation trying to add an Orbis Viaris (and also the general OCPP charger) using the GUI I get this error:

"cannot create charger type 'template': cannot create charger type 'ocpp': connector already registered: 1"

[evcc-20250323-083056-trace.log](https://github.com/user-attachments/files/19407878/evcc-20250323-083056-trace.log)

vincko, 2025-03-27:
**TL;DR**  
I found a workaround for this issue with OCPP wallboxes and the Config UI. Follow the steps below to see if it helps!

---

## Workaround Instructions

> **Important:** Do *not* configure the wallbox with OCPP before starting. Correct timing is essential for this workaround.

1. **In evcc’s Config UI, add a new charge point:**
   - Name it however you like.
   - Select one of the OCPP chargers.
   - Add a *dummy* `stationid` (for example, `1`) — don’t use the actual one you’ll set in the wallbox.

2. **Click `Validate & Save`.**

3. **While evcc is still loading, change the `stationid` to the real value** (for example, `amtron`).  
   - In many wallboxes, this can be set in the OCPP settings. Sometimes it's named differently, e.g. `Client ID`.

4. **Wait** even if it looks like the `Validate & Save` action was canceled.  
   - The UI won’t show a loading state anymore, but give it about 15 seconds.

5. **After 15 seconds, you’ll see an error from the original `Validate & Save` request.**  
   - Notice the button is now just `Save` (not `Validate & Save`).

6. **Configure the OCPP settings in your wallbox as usual**, but ensure it doesn’t try to connect yet.
   - That means setting the URL & stationID in the wallbox, but not letting it fully initiate a connection.
   - For some wallboxes this means don't save the settings yet. For others, it means you can save, but not yet restart!

7. **Back in evcc, click `Save`.**

8. **Only now** let the wallbox connect to evcc. You may have to be quick. If it fails, you will have to try again with a new `stationid`.

Done! ✅

---

## How I Discovered This

I ran into the same OCPP connection issue myself. I resolved it by simulating a virtual OCPP wallbox and experimenting with different connection timings using this simulator:  
[OCPP Virtual Charge Point](https://github.com/solidstudiosh/ocpp-virtual-charge-point)

---

## Tested Versions

- `0.202.0`
- `0.201.0`

Using the **generic OCPP1.6J** charger. Hope it helps! Would be keen to hear if it worked for you.

MatzeG, 2025-04-09:
Hello 
is this fixed or should I use the workaround ?

Cheppetto, 2025-04-28:
Still broken, 

Can't test wallbox ocpp capabilities as the workaround does not work for me  / my wallboxes

vincko, 2025-04-28:
On a general note: the workaround is based on my personal experience, not necessarily endorsed by the team behind evcc. 

@Cheppetto 
The goal is the followung:
1. You hit `Save` (step 7)
1. Evcc now waits for the wallbox to connect with the station id entered in step 3 (the real value you want to use) 
1. The wallbox connects to evcc before the check in evcc times out
1. You can continue.

It also took me multiple tries to get right with a wallbox that had to be restarted after changes to the OCPP settings.

What do you get in your case? Does an error appear that there was a timeout?

amtssp, 2025-05-08:
Hi is there any further information that is needed to get to the bottom of this error? 
I will be happy to provide any missing log files. 

The problem is that I'm still unable to add my OCPP charger via UI because of the error:
**"cannot create charger type 'template': cannot create charger type 'ocpp': connector already registered: 1"**

So I find it difficult to troubleshoot another issue I have: https://github.com/evcc-io/evcc/discussions/20903

amtssp, 2025-06-01:
I asked Copilot if the validation process could be the root cause for this error, and it seem that the validation process actually register the connector, so when the save process would like to save it, then it give this error.
Here is the suggestion from Copilot:

`Yes, the error message:

"cannot create charger type 'ocpp': connector already registered:"

can indeed result from the validation process already registering the OCPP charger, so that when you try to save (and thus re-create/register) the same charger, the system detects a duplicate registration and throws this error.

Why does this happen?
Validation Process: When you press "Validate," evcc attempts to actually connect to and register the OCPP charger using the provided configuration. This is done to ensure that the settings are correct and the charger is reachable.
Registration Side Effect: In the OCPP charger implementation, registering a charger (specifically, the charge point/connector) involves adding it to an internal registry or map. If you do this during validation, the internal state is changed.
Save Process: When you then attempt to "Save," the system tries to create/register the same OCPP charger again. Since it was already registered during validation, the code detects this as a duplicate and throws the "already registered" error.

```
Relevant Code Evidence
From the code (see charger/ocpp.go):

Go
cp, err := ocpp.Instance().RegisterChargepoint(id, ...)
if err != nil {
    return nil, err
}
```
If the RegisterChargepoint function finds that a charge point with the same ID is already registered, it will return an error.

Typical Cause
The validation process is not idempotent—it leaves behind a registered charge point.
The save process does not "know" this is just a leftover from validation, so it refuses to register again.
Solutions
Backend Fix: The validation process should clean up any registered charge points after the check, or use a temporary/unique ID for validation.
Frontend/UX Guidance: As a user, avoid pressing "Validate" multiple times before saving, or reload the page to clear temporary state if you encounter this error.
If you are developing or maintaining the system, consider ensuring that validation does not have persistent side effects, or that it unregisters/cleans up after itself.

Summary:
Yes, the error can be caused by the validation process registering the OCPP charger, which then blocks the save process due to the duplicate registration.`

So CoPilot suggest to:
```
Modify the validation logic so that after a successful validation, the OCPP charger or connector is unregistered and cleaned up.

After you call the registration (e.g., RegisterChargepoint), ensure you also call the corresponding "unregister" or removal function before validation completes or if validation fails. 
```

amtssp, 2025-06-26:
Hi [@mfuchs1984 - your conclusion is precisely the same as my analysis using copilot was showing.
The issue seems to be caused by the validation process that actually register the ID during the process, and therefore it is not possible to save the charger later because the ID is already used.

Copilot suggest that the validation process use a clearly-marked temporary ID and the register/unregister function is using this temporary ID during validation. This ensures the real charger registration is never touched during validation. 

I don't know go well enough to actually change the code - but copilot suggest a code change like this:

`func ValidateOCPPConfig(ctx context.Context, config map[string]interface{}) error {
    realId := config["StationId"].(string)
    tempId := fmt.Sprintf("validation-%s", realId)
    config["StationId"] = tempId // Use temp ID for validation

    charger, err := NewOCPPFromConfig(ctx, config)
    if err != nil {
        return err
    }

    // ... Perform validation logic ...

    // Always unregister the temp ID after validation
    ocpp.Instance().UnregisterChargepoint(tempId)

    return nil
}` 

Maybe someone with a deeper knowledge can take a look?

thaake, 2025-07-15:
Same problem here,, tried to switch to ui config, but this prevents my from it.
Staying with yaml for the time being.

andig, 2025-07-31:
Did you try it?

closed 2025-08-03 by andig

networxnet, 2025-12-29:
Same problem here.

````
[main ] INFO 2025/12/29 15:43:12 evcc 0.211.1
[main ] INFO 2025/12/29 15:43:12 using config file: /config/evcc.yaml
[main ] INFO 2025/12/29 15:43:12 using sqlite database: /data/evcc.db
[main ] INFO 2025/12/29 15:43:12 UI listening at :7070
[SN10052201194090-1] DEBUG 2025/12/29 15:43:12 waiting for chargepoint: 5m0s
[ocpp ] DEBUG 2025/12/29 15:43:17 charge point connected: SN10052201194090
[SN10052201194090-1] DEBUG 2025/12/29 15:43:47 failed configuring availability: timeout
[ocpp ] DEBUG 2025/12/29 15:43:47 charge point disconnected: SN10052201194090
[SN10052201194090-1] WARN 2025/12/29 15:43:48 failed triggering StatusNotification: cannot send request 3196447241, no client SN10052201194090 exists
[ocpp ] DEBUG 2025/12/29 15:43:48 charge point connected: SN10052201194090
[ocpp ] DEBUG 2025/12/29 15:44:20 charge point disconnected: SN10052201194090
[main ] FATAL 2025/12/29 15:44:22 charger [db:4] cannot create charger 'db:4': cannot create charger type 'template': cannot create charger type 'ocpp': timeout failed booting site: not found: my_grid
[main ] FATAL 2025/12/29 15:44:22 will attempt restart in: 15m0s
````

Bastel seit Oktober mit evcc und meiner Wallbox. Die ganze Konfiguration über OCPP ist gefühlt etwas wackelig.

Nach Neuinstalltion konnte ich die Wallbox per UI nicht neu konfigurieren, angeblich kein Connect. Im UI wird sie nicht angezeigt, ist aber laut Fehlermeldung beim Neustart angeblich vorhanden. Kein Charger in der yaml angelegt.

andig, 2025-12-29:
Wie immer wenns fertig ist.

networxnet, 2025-12-30:
> Der funktionierende OCPP-Flow über die UI kam erst nach 0.211.1. Mit dem aktuellen Nightly probiert?

Muss die evcc,yaml bei der nightly an einer anderen Stelle liegen? Ich bekomme die nightly nicht gestartet:

`Using config file: 
Using database file: /data/evcc.db
No config file (evcc.yaml) and no database (evcc.db) file configured. Please copy your database to /data/evcc.db or migrate your settings to the database or create a config under .
For details see evcc Home Assistant documentation at https://docs.evcc.io/docs/installation/home-assistant`

Anlegen eines Folders unter addon-configs mit -nightly im Namen ebenfalls ohne Erfolg.

networxnet, 2025-12-30:
Genau so habe ich es auch konfiguriert.

[image]

networxnet, 2025-12-31:
> Wie immer wenns fertig ist.

Ist der funktionierende OCPP-Flow in der neuen 0.300 drin? @andig @premultiply