evcc-io/evcc #19763: Support für Shelly Pro 3EM Switch Add-on

iCodr8, 2025-03-13:
Ich würde gerne das **Shelly Pro 3EM Switch Add-on** benutzen, um den Boost Modus meiner Wärmepumpe zu steuern. Aktuell scheint der Shelly Pro 3EM nur ohne das Switch Add-on benutzt werden zu können.

**Shelly Pro 3EM** als **meter** zu verwenden funktioniert aktuell schon.
Shelly Pro 3EM **Switch Add-on** als **charger** zu verwenden ist _nicht_ möglich.

Beides benutzt die gleiche IP Adresse.

Smartswitch Unterstützung für das Shelly Pro 3EM Switch Add-on:

https://www.shelly.com/de/products/shelly-pro-3em-switch-add-on?srsltid=AfmBOoqmRTf9ZiimkVgDuvucQ7RP0CAupumrZ0QJ3aOsYqy7YjY1hMGX

Ich würde es gerne in etwa so verwenden:

```
chargers:
- name: heatpump_control
  type: template
  template: shelly
  host: 192.168.xxx.xxx
  integrateddevice: true
  heating: true

meters:
- type: template
  template: shelly-pro-3em
  host: 192.168.xxx.xxx
  usage: ext
  name: heatpump_power

loadpoints:
  - title: "Wärmepumpe"
    charger: heatpump_control
    meter: heatpump_power
    phases: 3
```

Vielleicht gibt es auch einen alternativen Vorschlag, aber bisher konnte ich noch keine Lösung dafür finden.

premultiply, 2025-03-14:
Da Fragen wir doch mal @thierolm ob er dazu etwas weiss 🙂

thierolm, 2025-03-14:
@iCodr8 , richte mal das Addon als charger ein, und teile den Output von `evcc -l trace meter`.

thierolm, 2025-03-14:
@premultiply, das ist ein neues Shelly (Add-On) device und ist eigentlich kein Issue, sondern ein Feature request.

iCodr8, 2025-03-14:
Nachdem ich mich nun eine Weile damit beschäftigt habe, habe ich die Lösung gefunden. Sie war deutlich einfacher, als ich gedacht habe.

Die Meisten Shelly Schalter laufen mit der **ID 0** und inzwischen habe ich herausgefunden, dass das das **Shelly Pro 3EM Switch Add-on** auf **ID 100** läuft.

Die Lösung ist somit einfach nur `channel: 100` beim Charger hinzuzufügen und das war's.
Manchmal kann es so einfach sein.

`chargers:
- name: heatpump_control
  type: template
  template: shelly
  host: 192.168.xxx.xxx
  **channel: 100**
  integrateddevice: true
  heating: true

meters:
- type: template
  template: shelly-pro-3em
  host: 192.168.xxx.xxx
  usage: ext
  name: heatpump_power

loadpoints:
  - title: "Wärmepumpe"
    charger: heatpump_control
    meter: heatpump_power
    phases: 3`

iCodr8, 2025-03-14:
Vielleicht würde ein kleiner Hinweis in der Doku dazu helfen.

closed 2025-03-14 by iCodr8