evcc-io/evcc #19771: vehicle soc wird nicht am Ladepunkt übernommen

jgantenberg, 2025-03-14:
### Describe the bug

[image]
Ich habe einen Renault Zoe und eine OCPP Wallbox. Wie im Screenshot zu sehen ist, wird der SOC Wert vom Fahrzeug nicht am Ladepunkt übernommen. Ich habe zwar in der Anzeige den estimierten Wert, der jedoch immer bei 0 anfängt zu schätzen. Wenn ich ein Ladelimit definieren möchte, funktioniert das so nicht.

### Steps to reproduce

1. Fahrzeug lädt

1.Version 0.200.9
2. linux arm (orange Pi)

### Configuration details

```yaml
loadpoints:
  - title: Garage
    charger: EntratekWallbox
    mode: pv
    meter: charge1
    soc:
      poll:
        mode: connected
        interval: 15m
      estimate: true
    enable:
      delay: 2m
      threshold: -500
    disable:
      delay: 10m
      threshold: 1000
```

### Log details

```text
[lp-1  ] DEBUG 2025/03/14 11:24:41 charger status: 
[lp-1  ] DEBUG 2025/03/14 11:38:26 charge power: 10103W
[lp-1  ] DEBUG 2025/03/14 11:38:26 charge currents: [15.2 15.1 15.2]A
[lp-1  ] DEBUG 2025/03/14 11:38:26 detected active phases: 3p
[lp-1  ] DEBUG 2025/03/14 11:38:26 charge total import: 1723.980kWh
[lp-1  ] DEBUG 2025/03/14 11:38:26 charger status: C
[lp-1  ] DEBUG 2025/03/14 11:38:26 soc estimated: 8.54% (vehicle: 0.00%)
[lp-1  ] DEBUG 2025/03/14 11:38:26 vehicle soc: 9%
[lp-1  ] DEBUG 2025/03/14 11:38:26 vehicle range: 37km
[lp-1  ] DEBUG 2025/03/14 11:38:26 smart cost active: 0.34
[renault] TRACE 2025/03/14 11:38:39 GET https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/xxxxxxxxxx/kamereon/kca/car-adapter/v2/cars/xxxxxxxxxx/battery-status?country=DE
[renault] TRACE 2025/03/14 11:38:39 {"data":{"id":"xxxxxxxxxx","attributes":{"timestamp":"2025-03-14T10:37:11Z","batteryLevel":12,"batteryAutonomy":43,"plugStatus":1,"chargingStatus":1.0,"chargingRemainingTime":305}}}
[renault] TRACE 2025/03/14 11:38:39 GET https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/xxxxxxxxxx/kamereon/kca/car-adapter/v1/cars/xxxxxxxxxx/cockpit?country=DE
[renault] TRACE 2025/03/14 11:38:40 {"data":{"id":"xxxxxxxxxx","attributes":{"fuelAutonomy":0,"fuelQuantity":0.0,"totalMileage":32724}}}
[lp-1  ] DEBUG 2025/03/14 11:38:41 charge power: 10103W
[lp-1  ] DEBUG 2025/03/14 11:38:41 charge currents: [15.2 15.1 15.2]A
[lp-1  ] DEBUG 2025/03/14 11:38:41 detected active phases: 3p
[lp-1  ] DEBUG 2025/03/14 11:38:41 charge total import: 1724.010kWh
[lp-1  ] DEBUG 2025/03/14 11:38:41 charger status: C
[lp-1  ] DEBUG 2025/03/14 11:38:41 soc estimated: 8.59% (vehicle: 0.00%)
[lp-1  ] DEBUG 2025/03/14 11:38:41 vehicle soc: 9%
[lp-1  ] DEBUG 2025/03/14 11:38:41 vehicle range: 43km
[lp-1  ] DEBUG 2025/03/14 11:38:41 smart cost active: 0.34
[lp-1  ] DEBUG 2025/03/14 11:38:56 charge power: 10115W
[lp-1  ] DEBUG 2025/03/14 11:38:56 charge currents: [15.2 15.2 15.2]A
[lp-1  ] DEBUG 2025/03/14 11:38:56 detected active phases: 3p
[lp-1  ] DEBUG 2025/03/14 11:38:56 charge total import: 1724.070kWh
[lp-1  ] DEBUG 2025/03/14 11:38:56 charger status: C
[lp-1  ] DEBUG 2025/03/14 11:38:56 soc estimated: 8.69% (vehicle: 0.00%)
[lp-1  ] DEBUG 2025/03/14 11:38:56 vehicle soc: 9%
[lp-1  ] DEBUG 2025/03/14 11:38:56 vehicle range: 43km
[lp-1  ] DEBUG 2025/03/14 11:38:56 smart cost active: 0.34
[lp-1  ] DEBUG 2025/03/14 11:39:11 charge power: 10119W
[lp-1  ] DEBUG 2025/03/14 11:39:11 charge currents: [15.2 15.2 15.2]A
[... 463 lines cut ...]
[lp-1  ] DEBUG 2025/03/14 11:49:11 charger status: C
[lp-1  ] DEBUG 2025/03/14 11:49:11 soc estimated: 11.49% (vehicle: 0.00%)
[lp-1  ] DEBUG 2025/03/14 11:49:11 vehicle soc: 11%
[lp-1  ] DEBUG 2025/03/14 11:49:11 vehicle range: 53km
[lp-1  ] DEBUG 2025/03/14 11:49:11 smart cost active: 0.34
[lp-1  ] DEBUG 2025/03/14 11:49:26 charge power: 10096W
[lp-1  ] DEBUG 2025/03/14 11:49:26 charge currents: [15.2 15.1 15.2]A
[lp-1  ] DEBUG 2025/03/14 11:49:26 detected active phases: 3p
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.9

naltatis, 2025-03-14:
@jgantenberg magst du mal deine gesamte Config posten. Also welches Charger Template hier zum Einsatz kommt.

jgantenberg, 2025-03-14:
hier die vollständige config
```
root@evcc:~# cat /etc/evcc.yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.at.home # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: info
  chargers: debug
  meters: info
  mqtt: info
  ocpp: debug

# unique installation id
plant: "xxxxxxx"

sponsortoken: xxxxxxxx

interval: 15s # control cycle interval

meters:
- name: grid
  type: custom
  energy:
    source: mqtt
    broker: 192.168.96.3:1883
    clientid: evcc
    user: xxxxxxxxxx
    password: xxxxxxxxxx
    topic: esp-solar/sensor/growatt_energy_to_user_total/state
    scale: 1
  power:
    source: mqtt
    broker: 192.168.96.3:1883
    clientid: evcc
    user: xxxxxxxxxx
    password: xxxxxxxxxx
    topic: esp-solar/sensor/growatt_combined_power_from_grid/state
    scale: 1

- name: pv1
  type: custom
  power:    ##Leistungswert aus Umrichter
    source: mqtt
    broker: 192.168.96.3:1883
    clientid: evcc
    user: xxxxxxxxxx
    password: xxxxxxxxxx
    topic: esp-solar/sensor/growatt_pv_power/state
    scale: 1

  energy:   ##Gesamtertrag aus Umrichter
    source: mqtt
    broker: 192.168.96.3:1883
    clientid: evcc
    user: xxxxxxxxxx
    password: xxxxxxxxxx
    topic: esp-solar/sensor/growatt_pv_energy_total/state
    scale: 1

- name: battery1
  type: custom
  power:
    source: mqtt
    broker: 192.168.96.3:1883
    clientid: evcc
    user: xxxxxxxxxx
    password: xxxxxxxxxx
    topic: esp-solar/sensor/growatt_combined_battery_charging/state
    scale: -1

  soc:
    source: mqtt
    broker: 192.168.96.3:1883
    clientid: evcc
    user: xxxxxxxxxx
    password: xxxxxxxxxx
    topic: esp-solar/sensor/growatt_battery_soc/state
    scale: 1

  energy:
    source: mqtt
    broker: 192.168.96.3:1883
    clientid: evcc
    user: xxxxxxxxxx
    password: xxxxxxxxxx
    topic: esp-solar/sensor/growatt_battery_discharge_total/state
    scale: 1

- name: charge1
  type: custom
  power:
    source: mqtt
    broker: 192.168.96.3:1883
    clientid: evcc
    user: xxxxxxxxxx
    password: xxxxxxxxxx
    topic: esp-wb/sensor/sdm630_total_power/state
    scale: 1
  energy:
    source: mqtt
    broker: 192.168.96.3:1883
    clientid: evcc
    user: xxxxxxxxxx
    password: xxxxxxxxxx
    topic: esp-wb/sensor/sdm630_import_active_energy/state
    scale: 1
  currents:
    - source: mqtt
      broker: 192.168.96.3:1883
      clientid: evcc
      user: xxxxxxxxxx
      password: xxxxxxxxxx
      topic: esp-wb/sensor/sdm630_current_l1/state
      scale: 1
    - source: mqtt
      broker: 192.168.96.3:1883
      clientid: evcc
      user: xxxxxxxxxx
      password: xxxxxxxxxx
      topic: esp-wb/sensor/sdm630_current_l2/state
      scale: 1
    - source: mqtt
      broker: 192.168.96.3:1883
      clientid: evcc
      user: xxxxxxxxxx
      password: xxxxxxxxxx
      topic: esp-wb/sensor/sdm630_current_l3/state
      scale: 1

site:
  title: ZuHause
  meters:
    grid: grid
    pv: pv1
    battery: battery1
  residualPower: 500

chargers:
  - name: EntratekWallbox
    type: template
    template: ocpp
    getconfiguration: false
    connecttimeout: 10m
    timeout: 10m

loadpoints:
  - title: Garage
    charger: EntratekWallbox
    mode: pv
    meter: charge1
    soc:
      poll:
        mode: connected
        interval: 15m
      estimate: true
    enable:
      delay: 2m
      threshold: -500
    disable:
      delay: 10m
      threshold: 1000

```

mfuchs1984, 2025-03-14:
Probier mal bei der charger configuration `metervalues: -SoC`

`getconfiguration: false` und `timeout: 10m` sind übrigens laut https://docs.evcc.io/docs/devices/chargers#ocpp-16j-kompatible-wallbox-mit-smart-charging-profil keine valide Optionen, können wohl weg.

mfuchs1984, 2025-03-15:
Kannst du ocpp auch mal auf trace setzen und den Start von evcc + Fahrzeug anstecken + kurz laden aufnehmen?

jgantenberg, 2025-03-15:
```
[lp-1  ] DEBUG 2025/03/15 14:17:54 charge power: 0W
[lp-1  ] DEBUG 2025/03/15 14:17:54 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2025/03/15 14:17:54 charge total import: 1747.870kWh
[lp-1  ] DEBUG 2025/03/15 14:17:54 charger status: A
[ocpp  ] TRACE 2025/03/15 14:18:03 send 0311107101121150094: [2,"1476966218","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2025/03/15 14:18:03 recv 0311107101121150094: [3, "1476966218", {"status": "Accepted"}]
[ocpp  ] TRACE 2025/03/15 14:18:04 recv 0311107101121150094: [2, "0043e28e-67d5-7e0b-0ebd-16cf0ef187c6", "MeterValues", {"connectorId": 1, "meterValue": [{"timestamp": "2025-03-15T13:18:02.000Z", "sampledValue": [{"value": "0.0", "context": "Trigger", "measurand": "Power.Active.Import", "location": "Outlet", "unit": "W"}, {"value": "0.01", "context": "Trigger", "measurand": "Energy.Active.Import.Register", "location": "Outlet", "unit": "kWh"}, {"value": "0.00", "context": "Trigger", "measurand": "Current.Import", "phase": "L1", "location": "Outlet", "unit": "A"}, {"value": "0.00", "context": "Trigger", "measurand": "Current.Import", "phase": "L2", "location": "Outlet", "unit": "A"}, {"value": "0.00", "context": "Trigger", "measurand": "Current.Import", "phase": "L3", "location": "Outlet", "unit": "A"}, {"value": "240.84", "context": "Trigger", "measurand": "Voltage", "phase": "L1", "location": "Outlet", "unit": "V"}, {"value": "239.65", "context": "Trigger", "measurand": "Voltage", "phase": "L2", "location": "Outlet", "unit": "V"}, {"value": "241.32", "context": "Trigger", "measurand": "Voltage", "phase": "L3", "location": "Outlet", "unit": "V"}, {"value": "16.00", "context": "Trigger", "measurand": "Current.Offered", "location": "Outlet", "unit": "A"}, {"value": "3861.11", "context": "Trigger", "measurand": "Power.Offered", "location": "Outlet", "unit": "W"}]}]}]
[ocpp  ] TRACE 2025/03/15 14:18:04 send 0311107101121150094: [3,"0043e28e-67d5-7e0b-0ebd-16cf0ef187c6",{}]
[lp-1  ] DEBUG 2025/03/15 14:18:09 charge power: 0W
[lp-1  ] DEBUG 2025/03/15 14:18:09 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2025/03/15 14:18:09 charge total import: 1747.870kWh
[lp-1  ] DEBUG 2025/03/15 14:18:09 charger status: A
[lp-1  ] WARN 2025/03/15 14:18:09 charger out of sync: expected disabled, got enabled
[lp-1  ] DEBUG 2025/03/15 14:18:09 charger enable
[ocpp  ] TRACE 2025/03/15 14:18:09 send 0311107101121150094: [2,"1900364237","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":16,"stackLevel":8,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2025-03-15T13:17:09Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":0}]}}}]
[ocpp  ] TRACE 2025/03/15 14:18:10 recv 0311107101121150094: [3, "1900364237", {"status": "Accepted"}]
[lp-1  ] DEBUG 2025/03/15 14:18:10 charger disable
[lp-1  ] DEBUG 2025/03/15 14:18:10 wake-up timer: stop
[ocpp  ] TRACE 2025/03/15 14:18:13 send 0311107101121150094: [2,"1312760656","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2025/03/15 14:18:13 recv 0311107101121150094: [3, "1312760656", {"status": "Accepted"}]
[ocpp  ] TRACE 2025/03/15 14:18:13 recv 0311107101121150094: [2, "004407aa-67d5-7e14-bd92-90e5ef29342a", "MeterValues", {"connectorId": 1, "meterValue": [{"timestamp": "2025-03-15T13:18:12.000Z", "sampledValue": [{"value": "0.0", "context": "Trigger", "measurand": "Power.Active.Import", "location": "Outlet", "unit": "W"}, {"value": "0.01", "context": "Trigger", "measurand": "Energy.Active.Import.Register", "location": "Outlet", "unit": "kWh"}, {"value": "0.00", "context": "Trigger", "measurand": "Current.Import", "phase": "L1", "location": "Outlet", "unit": "A"}, {"value": "0.00", "context": "Trigger", "measurand": "Current.Import", "phase": "L2", "location": "Outlet", "unit": "A"}, {"value": "0.00", "context": "Trigger", "measurand": "Current.Import", "phase": "L3", "location": "Outlet", "unit": "A"}, {"value": "240.87", "context": "Trigger", "measurand": "Voltage", "phase": "L1", "location": "Outlet", "unit": "V"}, {"value": "239.62", "context": "Trigger", "measurand": "Voltage", "phase": "L2", "location": "Outlet", "unit": "V"}, {"value": "241.34", "context": "Trigger", "measurand": "Voltage", "phase": "L3", "location": "Outlet", "unit": "V"}, {"value": "16.00", "context": "Trigger", "measurand": "Current.Offered", "location": "Outlet", "unit": "A"}, {"value": "3861.37", "context": "Trigger", "measurand": "Power.Offered", "location": "Outlet", "unit": "W"}]}]}]
[ocpp  ] TRACE 2025/03/15 14:18:13 send 0311107101121150094: [3,"004407aa-67d5-7e14-bd92-90e5ef29342a",{}]
[ocpp  ] TRACE 2025/03/15 14:18:23 send 0311107101121150094: [2,"2408766415","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2025/03/15 14:18:23 recv 0311107101121150094: [3, "2408766415", {"status": "Accepted"}]
[ocpp  ] TRACE 2025/03/15 14:18:24 recv 0311107101121150094: [2, "00442fe7-67d5-7e1f-c2f9-dbc5c9aa4b1c", "MeterValues", {"connectorId": 1, "meterValue": [{"timestamp": "2025-03-15T13:18:22.000Z", "sampledValue": [{"value": "0.0", "context": "Trigger", "measurand": "Power.Active.Import", "location": "Outlet", "unit": "W"}, {"value": "0.01", "context": "Trigger", "measurand": "Energy.Active.Import.Register", "location": "Outlet", "unit": "kWh"}, {"value": "0.00", "context": "Trigger", "measurand": "Current.Import", "phase": "L1", "location": "Outlet", [... cut ...] measurand": "Current.Offered", "location": "Outlet", "unit": "A"}, {"value": "3811.10", "context": "Sample.Periodic", "measurand": "Power.Offered", "location": "Outlet", "unit": "W"}]}]}]
[ocpp  ] TRACE 2025/03/15 14:20:59 send 0311107101121150094: [3,"00468466-67d5-7eba-eb84-45d961421776",{}]
[ocpp  ] TRACE 2025/03/15 14:21:09 send 0311107101121150094: [2,"4270576671","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[lp-1  ] DEBUG 2025/03/15 14:21:09 charge power: 3156W

```

mfuchs1984, 2025-03-15:
Das scheint geklappt zu haben.

jgantenberg, 2025-03-15:
tatsächlich hatte ich das auch gerade gesehen, dass in der UI wieder der korrekte SoC % Wert angezeigt wird und damit auch wieder die Einstellung des Ladelimits. 
Die out of sync Meldungen bekomme ich regelmäßig, da sie nur Warnungen sind, hat das keine Auswirkung auf die Funktion.

closed 2025-03-15 by jgantenberg

mfuchs1984, 2025-03-15:
>Die out of sync Meldungen bekomme ich regelmäßig, da sie nur Warnungen sind, hat das keine Auswirkung auf die Funktion.

Wie gesagt, die kannst du loswerden, siehe oben.