evcc-io/evcc #19789: Falsche Berechnung solar forecast

Maddin2020, 2025-03-14:
### Describe the bug

Hallo zusammen, ich habe nun den Solar Forecast für meine PV Anlagenfelder angelegt. Es kommt aber viel zu viel als vorhersage raus. 

Könnt ihr mir vielleicht sagen woran es liegen könnte. 

`

### Steps to reproduce

-

### Configuration details

```yaml
`solar:
    - type: template
      template: open-meteo
      interval: 1h
      lat: xxx
      lon: xxx
      dec: 21 # 21 Grad
      az: 0 # Süden
      kwp: 9.24
    - type: template
      template: open-meteo
      interval: 1h
      lat: xxx
      lon: xxx
      dec: 10 # 10 Grad
      az: -90 # Osten
      kwp: 2.64
    - type: template
      template: open-meteo
      interval: 1h
      lat: xxx
      lon:xxx
      dec: 10 # 10 Grad
      az: 90 # Westen
      kwp: 2.64
    - type: template
      template: open-meteo
      interval: 1h
      lat: xxx
      lon: xxx
      dec: 10 # 10 Grad
      az: 0 # Süden
      kwp: 1.76
```

### Log details

```text
No Log at the Moment
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

naltatis, 2025-03-14:
@Maddin2020 das sind zu wenig Details um irgend etwas analysieren zu können. Wir haben inzwischen ja das experimentelle Feature mit dem wir die Vorhersagekurve an die reale Produktion anpassen. Evtl. ist das eine Lösung.

closed 2025-03-14 by naltatis

Maddin2020, 2025-03-14:
@naltatis Danke für die schnelle Antwort. Wo finde ich den das experimentelle Feature was du erwähnst und was benötigst du um zu analysieren.

naltatis, 2025-03-14:
Menü > Darstellung > experimentell

Dann siehst du im Vorhersagedialog unten eine checkbox mit der du die Anpassung an/ausschalten kannst. Funktioniert allerdings erst gut, wenn auch etwas Sonne geschienen hat 😄

Maddin2020, 2025-03-14:
Super danke dir für die Antwort ich werde es ausprobieren.