evcc-io/evcc #19792: UI-Konfiguration: charger: cannot create charger 'db:10': invalid charger type:

VolkerK62, 2025-03-15:
### Describe the bug

#19263  ist noch nicht vollumfänglich

Ich habe in UI-Konfiguration einen Ladepunkt angelegt mit einem Shelly als Charger.
Nach Neustart funktioniert alles.
Dann habe ich in der UI-Konfiguration beim Charger eine Änderung vorgenommen (integrateddevice=true). 
Charger gespeichert, Ladepunkt gespeichert.
Nach einem Neustart kommt der Fehler

[image]

Es werden keine Ladepunkte angezeigt. Also keine Chance den Fehler im UI zu beheben.

Wenn ich mittels "evcc config delete" den Charger lösche, bekomme ich in UI-Konfiguration wieder die Ladepunkte angezeigt.

Aber, wenn ich dann den Shelly erneut als Charger hinzufüge und den Ladepunt speichere, kommt nach einem Neustart der Fehler " Charger db:10 not found".
Es wird also die "alte" Verbindung nicht gelöscht.
```
charger
---
db:12 {Type:template Title: Icon: Product:} map[channel:0 host:192.168.178.88 integrateddevice:true password:***** standbypower:15 template:shelly user:*****]

loadpoint
---
db:11 {Type: Title: Icon: Product:} map[charger:db:10 circuit: defaultMode: meter: mode:minpv phasesConfigured:1 soc:map[estimate:false poll:map[interval:3.6e+12 mode:charging]] thresholds:map[disable:map[delay:1.8e+11 threshold:0] enable:map[delay:6e+10 threshold:0]] title:ui-test vehicle:]
```
Man muss den Loadpoint komplett löschen und neu erstellen.

### Steps to reproduce

.

### Configuration details

```yaml
wsl@DESKTOP-7RU7VSU:~$ evcc config
[main  ] INFO 2025/03/15 06:16:33 evcc 0.200.9 (8bb57b53)
[main  ] INFO 2025/03/15 06:16:33 using config file: /etc/evcc.yaml
[db    ] INFO 2025/03/15 06:16:33 using sqlite database: /home/wsl/.evcc/evcc.db
[db    ] TRACE 2025/03/15 06:16:33 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2025/03/15 06:16:33 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/03/15 06:16:33 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/03/15 06:16:33 SELECT * FROM `settings` 16 <nil>
[mqtt  ] INFO 2025/03/15 06:16:33 connecting evcc-21215759 at tcp://192.168.178.25:1883
[mqtt  ] DEBUG 2025/03/15 06:16:33 tcp://192.168.178.25:1883 connected
[db    ] TRACE 2025/03/15 06:16:33 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2025/03/15 06:16:33 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/03/15 06:16:33 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/03/15 06:16:33 SELECT * FROM `configs` WHERE `configs`.`class` = 1 1 <nil>
charger
---
db:10 {Type: Title: Icon: Product:} map[channel:0 host:192.168.178.88 integrateddevice:true password:***** standbypower:15 template:shelly user:*****]

[db    ] TRACE 2025/03/15 06:16:33 SELECT * FROM `configs` WHERE `configs`.`class` = 2 0 <nil>
[db    ] TRACE 2025/03/15 06:16:33 SELECT * FROM `configs` WHERE `configs`.`class` = 3 0 <nil>
[db    ] TRACE 2025/03/15 06:16:33 SELECT * FROM `configs` WHERE `configs`.`class` = 4 0 <nil>
[db    ] TRACE 2025/03/15 06:16:33 SELECT * FROM `configs` WHERE `configs`.`class` = 5 1 <nil>
loadpoint
---
db:11 {Type: Title: Icon: Product:} map[charger:db:10 circuit: meter: mode:minpv phasesConfigured:1 soc:map[estimate:false poll:map[interval:3.6e+12 mode:charging]] thresholds:map[disable:map[delay:1.8e+11 threshold:0] enable:map[delay:6e+10 threshold:0]] title:ui-test vehicle:]

[db    ] TRACE 2025/03/15 06:16:33 SELECT * FROM `configs` WHERE `configs`.`class` = 6 0 <nil>
```

### Log details

```text
wsl@DESKTOP-7RU7VSU:~$ evcc
[main  ] INFO 2025/03/15 06:18:08 evcc 0.200.9 (8bb57b53)
[main  ] INFO 2025/03/15 06:18:08 using config file: /etc/evcc.yaml
[db    ] INFO 2025/03/15 06:18:08 using sqlite database: /home/wsl/.evcc/evcc.db
[db    ] TRACE 2025/03/15 06:18:08 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2025/03/15 06:18:08 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/03/15 06:18:08 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/03/15 06:18:08 SELECT * FROM `settings` 16 <nil>
[mqtt  ] INFO 2025/03/15 06:18:08 connecting evcc-1265398814 at tcp://192.168.178.25:1883
[mqtt  ] DEBUG 2025/03/15 06:18:08 tcp://192.168.178.25:1883 connected
[db    ] TRACE 2025/03/15 06:18:08 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2025/03/15 06:18:08 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/03/15 06:18:08 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[main  ] INFO 2025/03/15 06:18:08 listening at :7070
[db    ] TRACE 2025/03/15 06:18:08 SELECT * FROM `configs` WHERE `configs`.`class` = 5 1 <nil>
[db    ] TRACE 2025/03/15 06:18:08 SELECT * FROM `configs` WHERE `configs`.`class` = 2 0 <nil>
[db    ] TRACE 2025/03/15 06:18:08 SELECT * FROM `configs` WHERE `configs`.`class` = 1 1 <nil>
[main  ] FATAL 2025/03/15 06:18:09 charger cannot create charger 'db:10': invalid charger type:
[main  ] FATAL 2025/03/15 06:18:09 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.9 (8bb57b53)

andig, 2025-03-15:
> Ich habe in UI-Konfiguration einen Ladepunkt angelegt mit einem Shelly als Charger.

Bitte exakte steps to reproduce

andig, 2025-03-15:
> Es werden keine Ladepunkte angezeigt. Also keine Chance den Fehler im UI zu beheben.

@naltatis das sollte egtl. nicht der Fall sein?

andig, 2025-03-15:
@naltatis die Ursache für die fehlende Bearbeitbarkeit hier ist, dass `configureDevices` vor `configureLoadpoint` schon einen Fehler bringt und daher kein Ladepunkt mehr erzeugt wird. Damit sind Charger/Charge Meter dann auch nicht mehr bearbeitbar. Wir müssen wohl nochmal überlegen, in welcher Reihenfolge wir hier vorgehen wollen.

VolkerK62, 2025-03-15:
@andig  brauchst du noch weitere Infos von mir?

VolkerK62, 2025-03-15:
steps to reproduce:

Konfig-UI öffnen
Ladepunkt hinzufügen, Titel vergeben
Wallbox hinzufügen
Democharger auswählen
prüfen
speichern
Ladepunkt speichern
Neustart
Konfig-UI öffnen
den erstellten Ladepunkt bearbeiten
charger bearbeiten
in dern erweiterten Einstellungen 10A und Phasenumschaltung einstellen
prüfen
speichern
Ladepunkt speichern
Neustart
Fehler mit invalid charger type

andig, 2025-03-15:
@naltatis beim zweiten speichern verliert der Charger seinen `type`:

```
db:132 {Type:template Title: Icon: Product:} map[enabled:true phases1p3p:false power:0 status:A template:demo-charger]
db:137 {Type: Title: Icon: Product:} map[enabled:true phases1p3p:false power:0 status:A template:demo-charger]
```

Entweder schickt das UI etwas nicht mit und/oder das Backend muss doch defensiver mit den Updates umgehen.

andig, 2025-03-15:
@naltatis gleiches Problem bei den meters. Sieht aus wir Folgefehler aus https://github.com/evcc-io/evcc/pull/19416.

andig, 2025-03-15:
@VolkerK62 du kannst die kaputten Dinger (hier der Charger) mit

    evcc config delete <id>

löschen.

closed 2025-03-15 by andig