evcc-io/evcc #19794: Ladestrom am Auto begrenzen

topcaser, 2025-03-15:
Hi, ich habe einen MG4. Meine wallbox kann keine Phasen-Umschaltung. Der MG4 lädt also immer 3 phasig und somit komme ich auf minimal 4,1 kW. Wenn man den Ladestrom im Auto (!) begrenzt, passiert folgendes: 

Bei Einstellung "6A" lädt er mit ~2A auf allen 3 Phasen. Ergibt ~1,4kW.

Bei Einstellung "8A" lädt er mit ~2,7A auf allen 3 Phasen. Ergibt ~1,8kW.

Bei Einstellung "16A" lädt er mit ~5,3A auf allen 3 Phasen. Ergibt ~3,7kW.

Somit hätte man mehr Überschuss-Spielraum. Man kann den Ladestrom über mqtt begrenzen. Könnte man das evcc beibringen?

StevieC121176, 2025-03-15:
hast du schon versucht hier den Ladestrom auf weniger als 6A zu stellen?

[image]

andig, 2025-03-15:
Wir werden keine weiteren spezifischen Fahrzeug-APIs ergänzen. Wenn Du das selbst machen willst stell gerne einen PR analog Tesla.