evcc-io/evcc #19802: Teslamate Change in Subscription Pricing – New "Pay As You Go" Model

Berlusconi13, 2025-03-15:
### Describe the bug

Teslamate hat auf ein neues Zahlungsmodell umgestellt und nun taucht der folgende Fehler im Log auf. Die Ladesteuerung funktioniert nicht mehr:

ERROR 2025/03/15 12:27:09 max charge current 7A: Post "https://api.myteslamate.com/api/1/vehicles/XP7YGCEK0RB460704/command/set_charging_amps": context deadline exceeded (Client.Timeout exceeded while awaiting headers)

### Steps to reproduce

Ladung starten mit Tesla and TWC3

### Configuration details

```yaml
Nicht relevant
```

### Log details

```text
Nicht relevant
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.9

andig, 2025-03-15:
> context deadline exceeded

Ist ein Timeout und hat mit Bezahlmodellen nichts zu tun. Bzgl. Bezahlmodell machen wir ein follow-up in den Discussions.

naltatis, 2025-03-15:
@jlestel can the above error be related to your announced change at myteslamate.com?

Berlusconi13, 2025-03-15:
> > context deadline exceeded
> 
> Ist ein Timeout und hat mit Bezahlmodellen nichts zu tun. Bzgl. Bezahlmodell machen wir ein follow-up in den Discussions.

Es gibt keine access und refresh tokens mehr auf teslamate - anscheinend wurde hier was größeres umgestellt 

„To comply with Tesla's API policies, MyTeslaMate must now make API calls on behalf of its users using its own API credentials. As a result, we are adjusting our subscription model to a "Pay As You Go" system to better reflect actual usage.“

andig, 2025-03-15:
> Daher wahrscheinlich auch das timeout

Die URL kann man im Browser aufrufen, also kein timeout.

Berlusconi13, 2025-03-15:
> > Daher wahrscheinlich auch das timeout
> 
> Die URL kann man im Browser aufrufen, also kein timeout.

Es geht doch explizit um den command proxy.
Die tokens sind verschwunden.

Stattdessen erscheint folgendes:

Set the following env vars:
- TESLA_API_HOST=https://api.myteslamate.com
- TESLA_AUTH_HOST=https://api.myteslamate.com
- TESLA_AUTH_PATH=/api/oauth2/v3
- TOKEN=?token=*Mein API-Token*
# Important: need to lower polling interval to not increase your budget
- POLLING_ASLEEP_INTERVAL=180
- POLLING_CHARGING_INTERVAL=180
- POLLING_DRIVING_INTERVAL=120
- POLLING_ONLINE_INTERVAL=180
- POLLING_DEFAULT_INTERVAL=180
- POLLING_MINIMUM_INTERVAL=120

# You need to enable Streaming signals to use streaming on Teslamate
- TESLA_WSS_HOST=wss://streaming.myteslamate.com
- TESLA_WSS_TLS_ACCEPT_INVALID_CERTS=true
- TESLA_WSS_USE_VIN=true

Berlusconi13, 2025-03-15:
Nun erscheint Folgendes.
Es ist alles gepaired.

`max charge current 3A: 500 Internal Server Error: Server error: `POST https://tesla-vehicle-command.myteslamate.com/api/1/vehicles/XP7YGCEK0RB460704/command/set_charging_amps` resulted in a `500 Internal Server Error` response: {"response":null,"error":"vehicle rejected request: your public key has not been paired with the vehicle","error_descrip (truncated...) `

jlestel, 2025-03-15:
Hi guys

Regarding the price update, Tesla urged me to do it before Monday. I considered that this new model allows us to maintain the same price as before if there is no excessive use of the commands:

- I changed the subscriber plan: they therefore have a prorated credit that will be used to pay for their consumption for the next few months.
Obviously, I will refund this credit to those who wish to cancel.

- I highlighted the application creation page: https://app.myteslamate.com/tesla
It has the advantage of not collecting any confidential information compared to the version that didn't comply with Tesla's conditions.

The process isn't ideal yet; it's a version 1! By displaying the Shell commands to guide users, it still seems understandable. I'm thinking of adding Shell and PowerShell files to facilitate the execution of these commands. I'm curious to hear your thoughts on this.

- The MyTeslamate API management page no longer refers to the customer's Tesla API, but rather to the Tesla API used by MyTeslamate: https://app.myteslamate.com/fleet

In conclusion, I'm going to further improve this process, which will allow Tesla to remain compliant and prevent all customers from being unable to use the service.

Moreover, if creating an application is complicated for some (and they want to send commands), it's entirely possible to use only the MyTeslamate API and not have to create an application.

I'm curious to know more about the number of vehicle_data, command, and wake-up messages sent daily.

Nevertheless, our current shared approach seems relevant to me since:
- the customer benefits from their $10 monthly credits with their Tesla API
- only commands are used (and that's the added value of my service).

andig, 2025-03-15:
@jlestel I still don't fully grasp what exactly has changed. Does Tesla no longer allow using the customer's Client/Secret and instead requires you to use the MyTeslamate one now? If that's it, what did they base this on?

jlestel, 2025-03-15:
Exactly, a partner should not make calls with client applications, but only with their own.

Tesla has clarified that the previous approach violated the following sections:
```
§ 5.6.23 – "Accessing, storing, using, or retaining Tesla account credentials (such as a username or password) for any Tesla account other than the Partner’s Tesla account."
→ MyTeslaMate previously collected user credentials to make API calls, which is explicitly prohibited.

§ 5.6.24 – "Accessing, storing, using, or retaining product data or user data obtained through the use of Tesla credentials other than the Partner’s Tesla account."
→ By using user credentials to retrieve and process their data, MyTeslaMate was in violation of this rule.

§ 5.6.16 – "Replaying stored user passwords or credentials for commands that require such credentials in the API call."
→ If stored credentials were used to execute API commands at a later time, this was also considered a violation.
```

So I should not be able to read the Client Secret or access/secret tokens from the client API application.

Finally, they rely on what is visible on the website... and especially in their databases. This was highlighted by their oversight of application creation and command hits.

andig, 2025-03-15:
Oh dear. If I understand the implications correctly then this basically put MyTeslamate under the risk of the API consumers executing more requests than their subscription covers and assigns that risk to MyTeslamate as the owner of the client id? That is indeed very unfortunate. It also takes away the benefit of the 10$ free monthly api calls?

jlestel, 2025-03-15:
> It also takes away the benefit of the 10$ free monthly api calls?

This is only partially affected, as only orders are affected.
Currently, the $10 credit is used directly by EVCC for other calls.

Berlusconi13, 2025-03-15:
If understood correctly, it is currently not possible to charge with evcc in combination with TeslaMate, when charging Tesla with TWC3, right?

andig, 2025-03-15:
I don‘t see why it shouldn’t be as long as MyTeslaMate can work as command proxy. That said: every other, non-crippled, wallbox will be better that the severely restricted and forced-to-cloud-usage, unfortunate TWC. It‘s just a crap solution to a problem that has been solved properly before.

Berlusconi13, 2025-03-16:
I agree that TWC may not be the ideal solution.
What settings should I adjust to get it working again?
Currently, I can’t find any access token, refresh token, or proxy token on teslamate.com

Dragonkit, 2025-03-16:
Ich habe seit gestern genau das gleiche Probleme mit beiden Teslas am TWC. 

500 Internal Server Error` response: {"response":null,"error":"vehicle rejected request: your public key has not been paired with the vehicle","error_descrip (truncated...)

Habe sogar schon die Autos komplett neu verbunden in MyTeslaMate incl. Schlüssel aber keine Verbesserung. 
Ich hoffe es gibt da eine einfache Lösung. Laden geht bei mir aber, nur die Änderungen über EVCC nicht.
Steht halt auf "Schnell" und wir aktuell nur durch das Auto und den TWC zeitlich gesteuert.

Ist es eigentlich mögich EVCC nur zur Steuerung der Hausbatterie und für die Ladelogs zu nutzen ohne die Autos zu steuern ? Das würde mir eigentlich schon ausreichen. Muss ich dann einfach die Autos löschen und als Gastfahrzeuge laufen lassen ? Würde halt schon gerne die Autos sehen aber Steuerung ansich brauche ich aktuell nicht zu nutzen.
PV Überschussladen nutze ich das von Tesla selbst was ja mit Powerwall, TWC und Tesla Fahrzeug super funktioniert.

Berlusconi13, 2025-03-16:
Habe auch schon alles versucht.
Der Weg über myteslamate scheint aktuell nicht mehr zu funktionieren

naltatis, 2025-03-16:
Yes, currently the token generation process does not seem to work. We're in contact with @jlestel to see if there's anything we can support on. We'll also update our documentation or publish an update once the implications of this changes to evcc users are clear.
Until then please be patient.

jlestel, 2025-03-16:
> 500 Internal Server Error` response: {"response":null,"error":"vehicle rejected request: your public key has not been paired with the vehicle","error_descrip (truncated...)

Sorry, was a bug on my side. It's fixed now.

Berlusconi13, 2025-03-16:
There seems to be another issue, when pressing the button „Install my application Key“ on teslamate
[https://ibb.co/gMTGcyYb](url)

CarstenGreb, 2025-03-16:
Hallo zusammen!

Bei mir kommt in evcc jetzt folgende Fehlermeldung. 
401 Unauthorized: You need to subscribe to MyTeslamate Proxy API: https://www.myteslamate.com/billing/subscribeApi

Ich habe das Jahresabo und myteslamate funktioniert. Wenn dem Link folgt, existiert die Seite nicht. 

Hängt das auch mit der Umstellung zusammen?

jlestel, 2025-03-16:
@Berlusconi13 it's fixed.
@CarstenGreb I also fixed the link in error message. It's https://app.myteslamate.com/billing/subscribeApi

Thank you for feedback

andig, 2025-03-16:
Great support, thank you 💚

Berlusconi13, 2025-03-16:
Tks.
I’m keeping my fingers crossed that the command proxies will work again soon🤞

Dragonkit, 2025-03-17:
Also das eine Auto ging jetzt heute morgen ohne Probleme. Gerade das zweite dran gemacht und da kommt die Meldung im Log.

[lp-2 ] INFO 2025/03/17 13:26:53 car connected
[lp-2 ] INFO 2025/03/17 13:26:54 vehicle updated: unknown -> Tesla Model Y
[lp-2 ] INFO 2025/03/17 13:34:53 start charging ->
[lp-2 ] INFO 2025/03/17 15:34:53 stop charging <-
[lp-2 ] INFO 2025/03/17 15:34:53 car disconnected
[lp-2 ] INFO 2025/03/17 15:34:53 vehicle updated: Tesla Model Y -> unknown

[lp-1 ] INFO 2025/03/17 15:51:23 car connected
[lp-1 ] INFO 2025/03/17 15:51:23 start charging ->
[lp-1 ] INFO 2025/03/17 15:51:25 vehicle updated: unknown -> Tesla Model 3
[lp-1 ] WARN 2025/03/17 15:51:25 charger out of sync: expected disabled, got enabled
[lp-1 ] ERROR 2025/03/17 15:51:26 max charge current 14A: 403 Forbidden: account disabled: EXCEEDED_LIMIT
[lp-1 ] WARN 2025/03/17 15:52:23 charger logic error: current mismatch (got 5.1A measured, expected 0A)
[lp-1 ] ERROR 2025/03/17 15:52:24 max charge current 14A: 403 Forbidden: account disabled: EXCEEDED_LIMIT
[lp-1 ] ERROR 2025/03/17 15:53:24 max charge current 14A: 403 Forbidden: account disabled: EXCEEDED_LIMIT
[lp-1 ] ERROR 2025/03/17 15:54:24 max charge current 14A: 403 Forbidden: account disabled: EXCEEDED_LIMIT
[lp-1 ] ERROR 2025/03/17 15:55:24 max charge current 14A: 403 Forbidden: account disabled: EXCEEDED_LIMIT
[lp-1 ] ERROR 2025/03/17 15:56:24 max charge current 14A: 403 Forbidden: account disabled: EXCEEDED_LIMIT
[lp-1 ] ERROR 2025/03/17 15:57:24 max charge current 14A: 403 Forbidden: account disabled: EXCEEDED_LIMIT
[lp-1 ] ERROR 2025/03/17 15:58:24 max charge current 14A: 403 Forbidden: account disabled: EXCEEDED_LIMIT

Er läd aber trotzdem. Jemand eine Idee dazu warum der TeslaMate Account da nun disabled ist ?

biddns, 2025-03-17:
ja, ist bei mir auch so ... und kein zugriff mehr - frag mich wo die ganzen zugriffe herkommen...

TheJDJ13, 2025-03-17:
I started getting this yesterday. Now today I see it before I even had enough sun to start charging.

Garage: charger disable: 429 Too Many Requests: Usage limit exceeded: You have reached your maximum of 100 credits per day.

jlestel, 2025-03-17:
Sorry for the inconvenience, EXCEEDED_LIMIT  is fixed.

I'll remove the credits related to the error you mentioned in a few hours. This will allow you to play commands again.

andig, 2025-03-17:
Using longer `interval` will also reduce the number of commands, so will adding `coarsecurrent` do.

TheJDJ13, 2025-03-17:
> Sorry for the inconvenience, EXCEEDED_LIMIT is fixed.
> 
> I'll remove the credits related to the error you mentioned in a few hours. This will allow you to play commands again.

Thank you.

TheJDJ13, 2025-03-17:
> Using longer `interval` will also reduce the number of commands, so will adding `coarsecurrent` do.

What is coursecurrent and how would I add that?

I don't understand how I could have used so many credits before charging could have started today?

Berlusconi13, 2025-03-17:
> Also das eine Auto ging jetzt heute morgen ohne Probleme. Gerade das zweite dran gemacht und da kommt die Meldung im Log.
>
> [lp-2 ] INFO 2025/03/17 13:26:53 car connected [lp-2 ] INFO 2025/03/17 [...]

Does it work for you again?
How did you get the command proxy running? I can’t find any tokens for that on TeslaMate.

chrassi20, 2025-03-17:
First day and all is blocked because of usage limit... car doesnt stop charging. 
What is the solution? 5min refresh? ... im realy depresed... 
[image]

Dragonkit, 2025-03-18:
> > Auch das Auto geht ein Auto in heute mit heute ohne Probleme. Gerade das Zweit dran gemacht und da kommt Meldung im Log.
> > [lp-2 ] INFO 2025/03/17 13:26:53 mit dem Auto verbunden [lp-2 ] INFO 2025/03/17 13:26:54 Fahrzeug aktualisiert: unbekannt -> Tesla Model Y [lp-2 ] INFO 2025/03/17 13:34:53 Startladung -> [lp-2 ] INFO 2025/03/17 15:34:53 Stop loading .- [lp-2 ] INFO 2025
> > [lp-1 ] INFO 2025/03/17 15:51:23 car connected [lp-1 ] INFO 2025/03/17 15:51:23 Startaufladung -> [lp-1 ] INFO 2025/03/17:51:25 fahrzeugaktuell aktualisiert: unbekannt -> Tesla Model 3 [lp-1 ] WARN 2025/03/17:51:25 Ladegerät aus dem Synchronisations-Ladung: ERKLICH: ER.
> > Er lädt aber. Meines Mal Idee nach der TeslaMate Account da ist bewohnbar
> 
> Funktioniert es wieder für Sie? Wie haben Sie den Befehlsvertreter zum Laufen gebracht? Dafür kann ich auf TeslaMate keine Token finden.

Ich hatte die Token vorher schon in der yaml

CarstenGreb, 2025-03-18:
I just got the error message Credit Limit exceeded again. Do I need to do or adjust anything else?

Berlusconi13, 2025-03-18:
> Ich hatte die Token vorher schon in der yaml

Ich habe nach der Umstellung alles gecleared, mist.

Leer lassen oder einfach was eingeben funktioniert leider auch nicht.
Da ist eine bestimmte Anforderungen an die Zeichenkette, die evcc bemängelt

VickyP1987, 2025-03-18:
>Ich habe nach der Umstellung alles gecleared, mist.
Leer lassen oder einfach was eingeben funktioniert leider auch nicht.
Da ist eine bestimmte Anforderungen an die Zeichenkette, die evcc bemängelt

Ich habe an der yaml nichts verändert, d.h. die Token sind noch drin, das "alte" refresh Token ist offensichtlich ungültig, siehe folgende Meldung im Log:\"/api/1/vehicles/***/command/set_charging_amps\": oauth2: \"login_required\" \"The refresh_token is invalid. Generate a new refresh_token by forcing user to re-authenticate.\

VickyP1987, 2025-03-18:
Tokens müssen neu generiert werden. Seitdem läuft bei mir wieder alles

TheJDJ13, 2025-03-20:
Today I see this after about 90 minutes of charging time:

Garage: max charge current 9A: 429 Too Many Requests: Usage limit exceeded: You have reached your maximum of 100 credits per day.

jbuchner, 2025-03-24:
Can the Tesla integration be modified so that the interval at which commands are sent via the proxy can be set independently of the general update interval? I would like to keep the general interval at 30 seconds but increase the "command interval" to, for example, 300 seconds to save proxy tokens.

andig, 2025-03-24:
> Can the Tesla integration be modified

You mean TWC3? Could probably try to send current control in specific intervals only. Those are not expensive commands though. What's really expensive are wakeups.

jbuchner, 2025-03-24:
> You mean TWC3?

Yes, as you can see from  @TheJDJ13 's comment, the limit of 100 credits per day is reached after 90 minutes of control (which, of course, can be increased in My Teslamate).

It would be great if evcc could be optimized to use commands more efficiently. This could be achieved either by allowing the interval to be adjusted or by only sending a new command when a significant change is necessary.

andig, 2025-03-25:
> It would be great if evcc could be optimized to use commands more efficiently.

See https://github.com/evcc-io/evcc/issues/19802#issuecomment-2749173776. 

> This could be achieved either by allowing the interval to be adjusted 

For time being you can already do this by adjusting the interval.

> or by only sending a new command when a significant change is necessary.

This will make PV charging less efficient. Happy to take a PR for twc3.go. Given the TWC is crippled (and has been from the beginning which only brings us into this situation), it seems this is not something that the core team will be working on.

Dragonkit, 2025-03-25:
Also der TWC3 mag verkrüppelt sein, aber in Verbindung mit Powerwall und Tesla Fahrzeug macht es das PV Überschussladen komplett ohne sonstige Tools. Ich hab es in EVCC auf "schnell" stehen und der Rest macht dann Tesla unter sich aus mit dem Überschussladen. Somit habe ich pro Tag und Auto bisher 1 bzw. 2 Credits verbrauch. Wenn ich dann wegen günstigem Strom mal schnell laden will mach ich der Tesla App kurz den Haken raus bei Überschussladen und gut ist.
Optimal ist das aber trotz allem leider nicht. Nutze EVCC quasi nur noch für das Laden von der Powerwall bei günstigem Strompreis und für die Übersicht der Ladedaten.
Aktuell stehe ich vor der grossen Frage ob ich von Tibber zum Tintenfisch wechsele um da halt feste Zeiten zum laden zu haben. Will aber erstmal abwarten wie sich das Modul 3 auf den Strompreis bei Tibber und Tinterfisch auswirkt.
Könnnte durchaus sein das damit das PV Überschuss laden der Vergangenheit angehört wenn man billiger Strom kaufen kann als es der vom Dach ist. Ich bekomme noch 14,06 Cent für den Verkauft und somit ist das schon nicht weit auseinander.

andig, 2025-03-25:
> Also der TWC3 mag verkrüppelt sein, aber in Verbindung mit Powerwall und Tesla Fahrzeug macht es das PV Überschussladen komplett ohne sonstige Tools. 

Ich wollte auch nichts anderes behaupten. Das ist aber ein weiterer Grund, dafür keine weiteren Sonderlösungen zu finden.

Dragonkit, 2025-03-25:
Bis 30 kwp muss man es nicht mehr bei der Steuer angeben und da es ohne Umsatzsteuer ist muss ich da auch nichts abziehen. Somit kann ich mit den 14,06 als direkten Vergleich rechnen. Also sobald ich unter 14 Cent einkaufen kann ich PV Überschuss teurer :-)

TheJDJ13, 2025-03-25:
> > You mean TWC3?
> 
> Yes, as you can see from [@TheJDJ13](https://github.com/TheJDJ13) 's comment, the limit of 100 credits per day is reached after 90 minutes of charging (which, of course, can be increased in My Teslamate).
> 
> It would be great if evcc could be optimized to use commands more efficiently. This could be achieved either by allowing the interval to be adjusted or by only sending a new command when a significant change is necessary.

Today I reached 100 credits message in about 30 minutes.

naltatis, 2025-03-27:
Just updated our documentation and the explaining blogpost to reflect the recent changes at myteslamate.

https://docs.evcc.io/en/docs/devices/vehicles#tesla
https://docs.evcc.io/en/blog/2025/01/20/tesla-api-update

Migrating this issue to discussions.