evcc-io/evcc #19811: Meter titles should be optional

andig, 2025-03-15:
Regression from https://github.com/evcc-io/evcc/pull/19416

naltatis, 2025-03-15:
I disagree. We're planning to implement more features where this name is used (like https://github.com/evcc-io/evcc/issues/9818). It's perfectly fine to require the user to pick a name for meters that you can have multiple of.
Title is not required for loadpoint or grid-meter since the context and name is clear.

naltatis, 2025-03-16:
Titel ist bei UI-Konfigurierten Ladepunkten und Fahrzeugen heute auch schon ein Pflichtfeld im UI. Würde das bei allen Einträgen von denen es mehrere geben kann im UI auch konistent durchziehen (Identifizierung der einzelnen Kacheln).

closed 2025-03-16 by naltatis