evcc-io/evcc #19818: Config UI: deleting vehicle breaks startup

andig, 2025-03-15:
V 0.200.9
Habe mein Tesla-Fahrzeug (UI-generiert) wegen dem Teslamate-Theater (umständlich, geht...geht nicht wg. erneut geä. Preispolitik, habe Account und Berechtigung gelöscht) per UI gelöscht und ein offline-Fahrzeug angelegt. 
Dabei ist beim Neustart auch gleich der zugeh. go-E Ladepunkt weggeflogen, lässt sich nicht mehr anlegen und bei Neustart kommt ein Fehler:

[main ] FATAL 2025/03/15 16:57:59 loadpoint [db:6] vehicle: not found: db:2
[main ] FATAL 2025/03/15 16:57:59 will attempt restart in: 15m0s 

In meinem yaml ist quasi nix mehr drin ausser Netzwerk, Sponsortoken und Verweis auf db-Pfad.
Sind hier noch Fahrzeug-Reste in der db? Kann man die Reste wegkriegen ohne alles wieder neu anzulegen (PV,Speicher,and. Ladepunkt, and. Fahrzeug)?
Danke Euch!

_Originally posted by @LuRhe in https://github.com/evcc-io/evcc/discussions/19817_

naltatis, 2025-03-15:
@LuRhe du hast vmtl. auch den Ladepunkt schon via UI angelegt und das Fahrzeug dort als Standardfahrzeug hinterlegt, richtig? Kannst du den Ladepunkt im Config UI aufrufen und das Standardfahrzeug dort entfernen?

@andig müssen wir zukünftig automatisch machen. Entweder direkt beim Entfernen oder nicht mehr auffindbare Referenzen beim Start entfernen. In diesem Fall (Vehicle) wäre es total ok es einfach zu entfernen. Ähnliche Themen aber wir aber auch beim Circuit.

LuRhe, 2025-03-15:
Ja, Ladepunkte und Fahrzeuge hab ich komplett über UI angelegt - und ja auch den Tesla da als Standard hinterlegt.
Wie gesagt, im Konfig UI wird der Ladepunkt nicht mehr angezeigt. Bei Neuanlegen prüft er erfolgreich und zählt [db..]Nummer hoch und nach speichern und Neustart ist er wieder weg.

andig, 2025-03-15:
Konkrete Schritte zum reproduzieren wären hilfreich.

LuRhe, 2025-03-15:
Ich denke vor dem Löschen des Fahrzeugs im UI hätte ich es im UI des Ladepunktes als Standardfz. löschen müssen. 
Wäre ein evcc.db output hilfreich - wie kriegt man die db in ein Textfile?
Ansonsten mach ich halt alles neu...
Danke aber für Eure schnelle Reaktion!
Bin grad mega gefrustet mit Tesla Api und BMW - ich glaub ich mach alles wie für Gastfahrzeuge.
Schönes Wochende Euch noch,
Lutz

andig, 2025-03-16:
Das einfachste hier wäre wenn Ladepunkte fehlendes Defaultvehicle nur als Warnung anzeigen würden. Ebenso muss beim Fahrzeug löschen ein Default raus gelöscht werden.

@naltatis der erste Punkt sollte im Nightly mit der letzten Ergänzung bzgl. Referenzen egtl. Schon erledigt sein.

LuRhe, 2025-03-17:
Wow, ihr habt ja schnell eine Lösung implementiert- super & danke für euer Engagement!