evcc-io/evcc #19820: Walbox Funktion nach Evcc Neustart

Crajs20, 2025-03-15:
Die Wallbox Funktion nach Neuverbinbden des Autos ist auf min+pv gesetzt was super funktioniert.

Evcc läuft als docker einwandfrei und wird per watchtower automatisch nachts geupdated.
Habe ich am Abend den Wagen noch verbunden und im Pv oder off Modus, das Laden unterbrochen, wird nach einem Neustart von Evcc die wallbox auf min+pv gesetzt und unnötig Energie aus dem Akku gezogen.

Wunsch wäre, das der letzte Status der wallbox gespeichert wird und entsprechend wieder gesetzt wird und nur nach einem reconnect des Autos der gesetzte Level min+pv gesetzt wird.

Wisley7, 2025-03-15:
Entweder setz es standardmäßig auf pv, oder mach dein Update manuell, dann passiert sowas nicht.

StefanSchoof, 2025-03-15:
Bei evcc und Auto-Update ist zu bedenken, dass evcc immer noch ein 0.x ist und mit dem Update auch Breaking Changes kommen können und evcc dann nach einen Update bei dir gar nicht startet.

closed 2025-03-15 by andig