evcc-io/evcc #19824: Wert von "Geladen" friert ein, wenn der Ladevorgang unterbrochen wird

StefanA-LU, 2025-03-15:
### Describe the bug

Wenn der Ladevorgang unterbrochen wird, weil, z. Bsp. bei der Einstellung "PV" sich eine Wolke vor die PV-Anlage schiebt oder die Ladung mit Umschalten auf "Aus" kurz unterbrochen wird, wird nach dem Wiederstart des Ladens der Wert für "Geladen" nicht mehr addiert, solange das Ladekabel nicht getrennt und wieder eingesteckt wird. Auch nach dem Trennen des Ladekabels fehlen dann diese Anteile in der Statistik.

Die meisten Einstellungen sind mittlerweile in die Datenbank verlegt. Nur grid ist noch in der yaml. Die Wallbox ist eine EM2GO Home mit aktueller Firmware.

[image]

[image]

[image]

### Steps to reproduce

1. Ladenvorgang starten.
2. Unterbrechen mit Umschalten auf "Aus" und 5 Minuten warten.
3. Umschalten auf "PV", Ladevorgang startet wieder.
4. Die Anzeige bleibt aber auf dem alten Stand und ändert sich nicht mehr, bis das Ladekabel getrennt und neu eingesteckt wird. Dann geht es mit 0,0 wieder los.
5. In der Gesamtstatistik fehlen die Lademengen ab 3. bis das Ladekabel neu verbunden wird.
6. 
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 30s # control cycle interval

meters:
- type: template
  template: e3dc-rscp
  usage: grid
  host: 10.0.0.30
  port: 5033
  user: *****
  password: *****
  key: *****
  name: grid1

site:
  title: Haus
  meters:
    grid: grid1

Version: `0.200.9`
```

### Log details

```text
Der Log geht leider erst nach dem Test los. Sind wohl schon wieder zu viele Einträge.
```

### What type of operating system or environment does evcc run on?

Windows

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

V.0.200.9

mfuchs1984, 2025-03-15:
Andere haben das Problem auch
https://github.com/evcc-io/evcc/discussions/17485

andig, 2025-03-15:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2025-03-15 by andig

StefanA-LU, 2025-03-16:
Naja, wenn das die Reaktion auf eine Analyse ist, dann muss ich meine IT-Ingenieurs-Kompetenz hier auch nicht mehr einbringen und meine Zeit verschwenden. Wenn der Logfile den Zeitpunkt nicht mehr beinhaltet, muss man halt bis zum nächsten Mal (heute) warten, bis man das wieder erzeugen kann.

Und der andere Fall scheint die Phasenumschaltung zu betreffen. Weiter ist festzuhalten, dass die EM2GO sauber wieder mit dem Laden beginnt und der interne Zähler auch korrekt wieder addiert. Aber man schließt ja sofort aus, dass es ggf. an der EVCC-Schnittstelle zur EM2GO liegen könnte. Dabei bekommt auch die E3DC mit, dass die EM2GO wieder lädt. Es wäre daher vielleicht einfach hilfreich, dort die Daten wieder auszulesen, wenn sich die Werte am "GRID" ändern. Denn die MODBUS Verbindung ist nicht gestört, das passiert nur bei jedem Neustart von EVCC.

andig, 2025-03-16:
Wir können halt nicht hell sehen. Deshalb gibts Logfiles.

mfuchs1984, 2025-03-22:
Hab mal einen PR gemacht, der die Session Energy entfernt, wie es aus ähnlichen Gründen auch schon bei diversen Wallboxen in der Vergangenheit gemacht wurde.

StefanA-LU, 2025-03-25:
[evcc-20250325-125641-debug.log](https://github.com/user-attachments/files/19448128/evcc-20250325-125641-debug.log)
[evcc-20250325-134414-trace.log](https://github.com/user-attachments/files/19448129/evcc-20250325-134414-trace.log)

StefanA-LU, 2025-03-25:
Hier mal 2 Logs von heute. Da konnte ich es wieder nachstellen. Was mit der 0.201.1 noch auffällt:

Verbindet man das Auto und stellt dann von AUS auf PV, wird max. mit 1,3 KW geladen.

Beendet man das Laden vom Auto aus, entfernt den Stecker und steckt ihn nach 10 Sekunden wieder an, startet der Ladevorgang und wechselt nach oben. Es wurde dann auch auf 3-phasig gewechselt und wieder auf 1-phasig nachdem die PV zurückgegangen ist. Dabei ist dann wieder das Einfrieren der "geladenen Menge" aufgetreten.

In dem Fall war jetzt zuerst von AUS auf PV gestellt und danach das Auto verbunden worden.

StefanA-LU, 2025-03-25:
Heute musste auch der Task neu gestartet werden, weil angeblich ein Anmeldefehler am Grid vorlag. Das war in der Konfiguration auch rot. Bei der EM2GO ist es leider so, dass wenn man den Task neu startet, dass dann die IP-Verbindung / MODBUS nicht sauber getrennt wird. Man muss dort den MODBUS neu starten, dann funktioniert es wieder. Ansonsten bekommt man auch nach Stunden bei jedem Versuch einen timeout Fehler.

StefanA-LU, 2025-03-25:
[evcc-20250325-125641-debug.log](https://github.com/user-attachments/files/19448523/evcc-20250325-125641-debug.log)
[evcc-20250325-134414-trace.log](https://github.com/user-attachments/files/19448522/evcc-20250325-134414-trace.log)

mfuchs1984, 2025-03-25:
Bzgl. des Zählers bitte mit der nightly testen.