evcc-io/evcc #19841: maxAcPower

Geri-G93, 2025-03-16:
### Describe the bug

Hallo
leider muss ich festellen das maxAcPower nicht richtig geregelt wird
setup:
10kw Hybrid Huawei WR (10Ktl-M1) mit 15kwh Luna (5kw entladenleistung)
! 18kwP DC Input (9Süd, 9 West) 
CorsaE mit 11kw OBC
ABL emH01 Wallbox mit OrnoZähler über rs485 konverter

unter 10kw regelt evcc Super, oben drüber garnicht

wie man sieht 10996W pv Power etwa, wiegesagt manchmal hab ich auch 15kw, dann lädt die Batterie vollgas nebenher (10netz+5Batterie)
2553W grid
excessDC 996W
3839W Site
trotzdem ChargePower 10203W

also so richtig gefällt mir das noch nicht, hab jetzt die Wb auf 14A begrenzt erstmal, das geht

bei fragen steh ich selbstverständlich zur verfügung

### Steps to reproduce

HybridWr

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 80

log: debug
levels:
  cache: error
  #mqtt: trace
interval: 30s # control cycle interval
# unique installation id
plant: ausgeblendet
sponsortoken: ausgeblendet

# mqtt message broker
mqtt:
  broker: 127.0.0.1:1889
  topic: evcc # root topic for publishing, set empty to disable publishing
  clientid: EVCC
  user: EVCC
  password: evcc2023
modbusproxy:
  - port: 503
    uri: 192.168.155.9:502    #Huawei sun2000_1

  
meters:
  - name: ChargerMeter
    type: template
    template: orno
    usage: charge
    
     ##Modbus TCP Waveshare_1 ORNO
     #RS485 via adapter (Modbus RTU)
    modbus: rs485serial
    id: 1
    device: COM3 # USB-RS485 Adapter Adresse
    baudrate: 9600 # Prüfe die Geräteeinstellungen, typische Werte sind 9600, 19200, 38400, 57600, 115200
    comset: "8E1" # Kommunikationsparameter für den Adapter  
    #modbus: rs485tcpip
[... 577 lines cut ...]
[site ] DEBUG 2025/03/16 13:17:26 grid power: 3638W
[site ] DEBUG 2025/03/16 13:17:26 battery 1 power: -9W
[site ] DEBUG 2025/03/16 13:17:26 pv 1 power: 10996W
[site ] DEBUG 2025/03/16 13:17:26 pv 1 excess DC: 996W
[site ] DEBUG 2025/03/16 13:17:26 grid currents: [6.46 6.69 13.1]A
[site ] DEBUG 2025/03/16 13:17:26 battery 1 soc: 100%
[site ] DEBUG 2025/03/16 13:17:26 site power: 4925W
[site ] DEBUG 2025/03/16 13:17:26 solar forecast: accumulated 61.149kWh, produced 62.310kWh, scale 1.019
```

### What type of operating system or environment does evcc run on?

Windows

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.9

Chris45671, 2025-03-16:
Würde deine E-Mail und die VIN aus den Daten rausnehmen.

mfuchs1984, 2025-03-16:
Du hast einen aktiven Ladeplan, dann wird natürlich Vollgas geladen.

closed 2025-03-16 by andig

Geri-G93, 2025-03-16:
Jetzt wo ihr das sagt, lt tibber war es da am günstigsten, stimmt sry.
Iobroker setzt beim anstecken anhand des Arbeitsplans die Abfahrtszeit
Werd das bei genug Überschuss auf später schieben
Werd ich Mittwoch nochmal probieren ohne