evcc-io/evcc #19845: `residualpower` verhindert `battery boost`, wenn zu hoch eingestellt

VolkerK62, 2025-03-16:
Wenn `residualpower` zu hoch eingestellt ist, funktioniert `battery boost` nicht richtig.

Wenn `battery boost`aktiv ist, sollte `residualpower` bei der Berechnung unberücksichtigt bleiben

Meine Batterie kann max 1,2kW, das reicht so gerade um den Boost zu nutzen
Zum Log:
16:08:08 Residualpower auf 500W eingestellt
16:08:38 battery boost aktiviert
16:09:43 Nach der Startphase wird auf 6A runtergeregelt und der disable timer startet
16:10:46 Residualpower auf 0W eingestellt
16:11:03 disable timer verschwindet 
Gegenprobe
16:11:54 Residualpower wieder auf 500W gestellt
16:12:23 disable timer startet wieder

```
[site  ] DEBUG 2025/03/16 16:07:42 ----
[lp-1  ] DEBUG 2025/03/16 16:07:42 charge power: 0W
[lp-1  ] DEBUG 2025/03/16 16:07:42 charge currents: [0 0 0]A
[site  ] DEBUG 2025/03/16 16:07:42 grid power: -3W
[site  ] DEBUG 2025/03/16 16:07:42 ext 1 power: 0W
[site  ] DEBUG 2025/03/16 16:07:42 battery 1 power: -300W
[site  ] DEBUG 2025/03/16 16:07:42 battery 1 soc: 73%
[site  ] DEBUG 2025/03/16 16:07:42 aux 1 power: 0W
[site  ] DEBUG 2025/03/16 16:07:42 ext 2 power: 0W
[site  ] DEBUG 2025/03/16 16:07:42 grid powers: [97 4 -104]W
[site  ] DEBUG 2025/03/16 16:07:42 pv 1 power: 757W
[site  ] DEBUG 2025/03/16 16:07:42 grid currents: [1.1 1.16 -0.99]A
[site  ] DEBUG 2025/03/16 16:07:42 site power: -248W
[lp-1  ] DEBUG 2025/03/16 16:07:42 charge voltages: [232 0 0]V
[lp-1  ] DEBUG 2025/03/16 16:07:42 detected connected phases: 1p
[lp-1  ] DEBUG 2025/03/16 16:07:42 charge total import: 2787.600kWh
[lp-1  ] DEBUG 2025/03/16 16:07:42 charger status: B
[site  ] DEBUG 2025/03/16 16:07:43 solar forecast: accumulated 43.137kWh, produced 26.900kWh, scale 0.624
[lp-1  ] DEBUG 2025/03/16 16:07:49 set charge mode: pv
[site  ] DEBUG 2025/03/16 16:07:49 ----
[lp-1  ] DEBUG 2025/03/16 16:07:49 set enable threshold: 0
[lp-1  ] DEBUG 2025/03/16 16:07:49 set disable threshold: 0
[lp-1  ] DEBUG 2025/03/16 16:07:49 charge power: 0W
[lp-1  ] DEBUG 2025/03/16 16:07:49 charge currents: [0 0 0]A
[site  ] DEBUG 2025/03/16 16:07:50 pv 1 power: 749W
[site  ] DEBUG 2025/03/16 16:07:50 ext 1 power: 0W
[site  ] DEBUG 2025/03/16 16:07:50 battery 1 power: -327W
[site  ] DEBUG 2025/03/16 16:07:50 battery 1 soc: 73%
[site  ] DEBUG 2025/03/16 16:07:50 aux 1 power: 0W
[site  ] DEBUG 2025/03/16 16:07:50 grid power: 39W
[site  ] DEBUG 2025/03/16 16:07:50 ext 2 power: 0W
[site  ] DEBUG 2025/03/16 16:07:50 grid powers: [96 23 -80]W
[site  ] DEBUG 2025/03/16 16:07:50 grid currents: [1.12 1.17 -0.95]A
[site  ] DEBUG 2025/03/16 16:07:50 site power: -233W
[lp-1  ] DEBUG 2025/03/16 16:07:50 charge voltages: [232 0 0]V
[lp-1  ] DEBUG 2025/03/16 16:07:50 detected connected phases: 1p
[lp-1  ] DEBUG 2025/03/16 16:07:50 charge total import: 2787.600kWh
[lp-1  ] DEBUG 2025/03/16 16:07:50 charger status: B
[lp-1  ] DEBUG 2025/03/16 16:07:50 pv charge current: 1.01A = 0A + 1.01A (-233W @ 1p)
[site  ] DEBUG 2025/03/16 16:07:50 solar forecast: accumulated 43.139kWh, produced 26.900kWh, scale 0.624
[site  ] DEBUG 2025/03/16 16:07:52 ----
[lp-1  ] DEBUG 2025/03/16 16:07:52 charge power: 0W
[lp-1  ] DEBUG 2025/03/16 16:07:52 charge currents: [0 0 0]A
[site  ] DEBUG 2025/03/16 16:07:52 ext 1 power: 0W
[site  ] DEBUG 2025/03/16 16:07:52 grid power: 36W
[site  ] DEBUG 2025/03/16 16:07:52 battery 1 power: -286W
[site  ] DEBUG 2025/03/16 16:07:52 battery 1 soc: 73%
[site  ] DEBUG 2025/03/16 16:07:52 grid powers: [92 25 -81]W
[site  ] DEBUG 2025/03/16 16:07:52 ext 2 power: 0W
[site  ] DEBUG 2025/03/16 16:07:52 pv 1 power: 749W
[site  ] DEBUG 2025/03/16 16:07:52 grid currents: [1.17 1.17 -0.946]A
[site  ] DEBUG 2025/03/16 16:07:52 aux 1 power: 0W
[site  ] DEBUG 2025/03/16 16:07:52 site power: -196W
[site  ] DEBUG 2025/03/16 16:07:53 solar forecast: accumulated 43.140kWh, produced 26.900kWh, scale 0.624
[... 578 lines cut ...]
[lp-1  ] DEBUG 2025/03/16 16:13:02 charge currents: [0 0 0]A
[site  ] DEBUG 2025/03/16 16:13:02 grid power: -1004W
[site  ] DEBUG 2025/03/16 16:13:02 aux 1 power: 0W
[site  ] DEBUG 2025/03/16 16:13:02 ext 1 power: 0W
[site  ] DEBUG 2025/03/16 16:13:02 pv 1 power: 710W
[site  ] DEBUG 2025/03/16 16:13:02 grid powers: [-1174 257 -88]W
[site  ] DEBUG 2025/03/16 16:13:02 ext 2 power: 0W
[site  ] DEBUG 2025/03/16 16:13:02 grid currents: [-5.04 1.39 -1.01]A
[site  ] DEBUG 2025/03/16 16:13:02 battery 1 power: 1085W
[site  ] DEBUG 2025/03/16 16:13:02 battery 1 soc: 73%
[site  ] DEBUG 2025/03/16 16:13:02 site power: 581W
[lp-1  ] DEBUG 2025/03/16 16:13:02 charge voltages: [232 0 0]V
[lp-1  ] DEBUG 2025/03/16 16:13:02 detected connected phases: 1p
[lp-1  ] DEBUG 2025/03/16 16:13:03 charge total import: 2787.800kWh
[lp-1  ] DEBUG 2025/03/16 16:13:03 charger status: B
[lp-1  ] DEBUG 2025/03/16 16:13:03 charge final energy: 13.600kWh
[lp-1  ] INFO 2025/03/16 16:13:03 stop charging <-
[lp-1  ] DEBUG 2025/03/16 16:13:03 charge total import: 2787.800kWh
[lp-1  ] DEBUG 2025/03/16 16:13:03 soc estimated: 63.87% (vehicle: 63.00%)
[lp-1  ] DEBUG 2025/03/16 16:13:03 vehicle soc: 64%
[lp-1  ] DEBUG 2025/03/16 16:13:03 vehicle soc limit: 90%
[lp-1  ] DEBUG 2025/03/16 16:13:03 vehicle range: 239km
[site  ] DEBUG 2025/03/16 16:13:03 solar forecast: accumulated 43.239kWh, produced 26.900kWh, scale 0.622
[site  ] DEBUG 2025/03/16 16:13:12 ----
```

premultiply, 2025-03-16:
Das ist IMHO das Problem was auch in #19385 konfigurationsabhängig auftritt.

premultiply, 2025-03-16:
Schau mal bitte mein letzter Commit in #19385.

VolkerK62, 2025-03-16:
Code lesen ist jetzt nicht meine Kernkompetenz, aber du hast meine Intention verstanden.

mfuchs1984, 2025-04-27:
Ich denke das sorgt auch für Probleme bei charger oder Autos, die nur in 1A Schritten geregelt werden. Dann wird versucht um 230* Phasen Watt zu boosten, vom boost geht aber die residualpower weg, deshalb wird die nächste Ampere Stufe nicht erreicht, siehe https://github.com/evcc-io/evcc/discussions/20892#discussioncomment-12959779

closed 2025-04-27 by premultiply