evcc-io/evcc #19851: EVCC support for SW SG ready for Weishaupt BiBlock

rafdev18, 2025-03-16:
It would be excellent if EVCC supports SG ready over Modbus. Several heat pumps do have modbus support and you can read and write directly the needed registers.

I have this today implemented in home assistant but would prefer a more centralised approach in EVCC.

You can easily set a new temperature with a modbus write_register call. Basically you read the current set temperature for hot water in the register and if there is excess sun energy available you set the hot water temperature to +5 C.

Attached the code I use in home assistant on register 42103 which is storing the hot water temperature.

    - service: modbus.write_register
      data:
        hub: wpump
        address: 42103
        slave: 1
        value: 500

Modbus support would be needed in EVCC and then you define the IP address for the heatpump on the local network and you can set hot water temperature higher to store the sun energy.

Happy to test and help in the development with additional code examples from HA -

I think the nice thing about this implementation is that all Modbus enabled heat pumps would work Viessmann, Weishaupt, ... and no additional HW is needed just the SW support in evcc.

andig, 2025-03-17:
Preferably we would set a mode (like SG Ready boost) instead of fiddling with temperatures, but worst case this would be possible. 

@deadrabbit87 would you want to update the broken Weishaupt template?

closed 2025-03-29 by github-actions[bot]

deadrabbit87, 2025-03-30:
Sorry for the delay, I was in a hurry to expand my PV. 

I'm not sure what to do here? The Modbus registers of the Weishaupt heat pump are only readable.

andig, 2025-03-30:
Ouch. Then that‘s a no-go. ALternative may be to use physical SG Ready signaling, i.e. have 2 Shellys to drive the SG1 and 2 lines.

andig, 2026-09-07:
See https://github.com/evcc-io/evcc/pull/33193, untested.