evcc-io/evcc #19853: EVCC OpenDTU: Enable filtering for specific inverter data

robmroi, 2025-03-16:
**Is your feature request related to a problem? Please describe.**\
**Ist dein Funktionswunsch mit einem Problem verbunden? Bitte beschreibe es.**

I have an unusual setup where I have two different Hoymiles inverters connected to OpenDTU, but only one of them actually feeds into my power circuit. The current OpenDTU integration in EVCC does not allow selecting a specific inverter, meaning I get data from both inverters combined, which is not useful for my case.

Ich habe eine ungewöhnliche Konfiguration: Ich habe zwei verschiedene Hoymiles-Wechselrichter mit OpenDTU verbunden, aber nur einer davon speist in meinen Stromkreis ein. Die aktuelle OpenDTU-Integration in EVCC erlaubt es nicht, einen spezifischen Wechselrichter auszuwählen, sodass ich die Daten beider Wechselrichter kombiniert bekomme – was für meinen Anwendungsfall nicht hilfreich ist.

---

**Describe the solution you'd like**\
**Beschreibe die gewünschte Lösung**

I would like to have an option in EVCC to specify a particular inverter ID when integrating OpenDTU (in the web interface). Instead of retrieving data via:

```
<OpenDTU IP>/api/livedata/status
```

EVCC should allow querying a specific inverter with:

```
<OpenDTU IP>/api/livedata/status?inv=138XXXXXXX
```

(where "138XXXXXXX" is the desired inverter ID). The data path in the JSON response would then change from:

```
/total/Power
```

to:

```
/inverters/0/AC/0/Power
```

Ich möchte, dass EVCC die Möglichkeit bietet, eine spezifische Wechselrichter-ID bei der OpenDTU-Integration (in der Weboberfläche) anzugeben. Statt die Daten über:

```
<OpenDTU IP>/api/livedata/status
```

abzurufen, sollte EVCC erlauben, einen bestimmten Wechselrichter mit:

```
<OpenDTU IP>/api/livedata/status?inv=138XXXXXXX
```

abzufragen (wobei "138XXXXXXX" die gewünschte Wechselrichter-ID ist). Der Datenpfad in der JSON-Antwort würde sich dann von:

```
/total/Power
```

zu:

```
/inverters/0/AC/0/Power
```

ändern.

---

**Describe alternatives you've considered**\
**Beschreibe alternative Lösungen, die du in Betracht gezogen hast**

A workaround would be to manually filter out the unwanted inverter data outside of EVCC, but this requires additional scripting and is not an elegant solution. Another alternative is modifying OpenDTU itself, but this is not ideal since OpenDTU works correctly in other applications.

Ein Workaround wäre, die unerwünschten Wechselrichter-Daten außerhalb von EVCC manuell zu filtern, aber das erfordert zusätzliche Skripterstellung und ist keine elegante Lösung. Eine weitere Alternative wäre, OpenDTU selbst zu modifizieren, aber das ist nicht ideal, da OpenDTU in anderen Anwendungen bereits korrekt funktioniert.

---

**Additional context**\
**Zusätzlicher Kontext**

I have attached two screenshots:

1. OpenDTU UI showing the inverter selection.
2. API request in Firefox showing the modified URL and JSON response.

Ich habe zwei Screenshots angehängt:

1. OpenDTU-Benutzeroberfläche mit der Wechselrichterauswahl.
2. API-Abfrage in Firefox mit geänderter URL und JSON-Antwort.

[image]
[image]

This feature would help users with multiple inverters connected to OpenDTU but only one relevant for EVCC. Thanks for considering this request!

Diese Funktion würde Nutzern helfen, die mehrere Wechselrichter mit OpenDTU verbunden haben, aber nur einen für EVCC nutzen möchten. Vielen Dank fürs Berücksichtigen dieses Vorschlags!

StefanSchoof, 2025-03-16:
Das opendtu Template ist Recht übersichtlich: https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/hoymiles-opendtu.yaml

Du solltest also in der Lage sein ein custom meter zu bauen, dass dein Problem löst. Wenn es sich universell ändern lässt sich das in das template einbauen. Seit kurzem ist es auch möglich ein Template einfach zu testen: https://github.com/evcc-io/evcc/issues/19489

andig, 2025-03-17:
If you want to add an optional setting- happy to take a PR!

robmroi, 2025-03-17:
> Das opendtu Template ist Recht übersichtlich: https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/hoymiles-opendtu.yaml
>
> Du solltest also in der Lage sein ein custom meter zu [...]

Danke! Damit sollte ich das bei mir auf jeden Fall hinbekommen

robmroi, 2025-03-17:
> If you want to add an optional setting- happy to take a PR!

First, I’ll create the custom template for my setup. When I have the time, I’ll look into adding the optional setting in the UI. Since I haven’t contributed to this project before, I first need to familiarize myself with how everything works.