evcc-io/evcc #19854: Config Fehler nach update auf 0.201.0 cannot create charger type 'template': cannot create charger type 'easee'

svenp, 2025-03-16:
### Describe the bug

Nach dem update startet evcc nicht mehr mit folgendem Fehler:
charger: [easee_rot] cannot create charger 'easee_rot': cannot create charger type 'template': cannot create charger type 'easee': unexpected status: 400 (Bad Request)

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
root@evcc:~# cat /etc/evcc.yaml 
network:
  schema: http
  host: 192.168.10.103
  port: 7070

interval: 15s # control cycle interval
sponsortoken: 
#SEMP_BASE_URL: http://192.168.10.103:7070
# trial token, valid until 2025-01-02

# sponsor token enables optional features (request at https://cloud.evcc.io)
#sponsortoken: eYiI6I

# log settings
log: info
levels:
  core: info #debug
  lp-1: info #debug
  lp-2: info #debug
  lp-3: info
# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/evcc-io/config#meters
meters:
- name: sma-grid
  type: template
  template: sma-energy-meter
  usage: grid
  host: 192.168.10.8 # IP-Adresse oder Hostname
- name: sma-battery
  type: template
  template: sma-si-modbus
  usage: battery
# Modbus TCP
  modbus: tcpip
  id: 3
  host: 192.168.10.10 # Hostname
  port: 502 # Port
  capacity: 32 # Akkukapazität in kWh (optional)
  watchdog: 60s # optional
  chargepower: 4600 # Maximale Ladeleistung für erzwungenes Laden
#  power: Power # default values, optionally override
- name: grid
  type: template
  template: discovergy
  usage: grid
  user: s # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
  password: # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
  meter: 1E
- name: sma-pv6
  type: template
  template: sma-inverter-modbus
  usage: pv
  # Modbus TCP
[... 348 lines cut ...]
Mar 16 20:23:24 evcc systemd[1]: Stopped evcc.service - evcc.
Mar 16 20:23:24 evcc systemd[1]: Started evcc.service - evcc.
Mar 16 20:23:24 evcc evcc[4340]: [main  ] INFO 2025/03/16 20:23:24 evcc 0.201.0
Mar 16 20:23:24 evcc evcc[4340]: [main  ] INFO 2025/03/16 20:23:24 using config file: /etc/evcc.yaml
Mar 16 20:23:24 evcc evcc[4340]: [db    ] INFO 2025/03/16 20:23:24 using sqlite database: /var/lib/evcc/evc>
Mar 16 20:23:24 evcc evcc[4340]: [main  ] INFO 2025/03/16 20:23:24 listening at :7070
Mar 16 20:23:24 evcc evcc[4340]: [main  ] FATAL 2025/03/16 20:23:24 charger [easee-rot] cannot create charg>
Mar 16 20:23:24 evcc evcc[4340]: [main  ] FATAL 2025/03/16 20:23:24 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.201.0

jpakusch, 2025-03-16:
Same issue here.

orthoenergy, 2025-03-16:
Same issue

pauljo2022, 2025-03-16:
sami issue

NFea-r, 2025-03-16:
same issue

svenp, 2025-03-16:
apt install evcc=0.200.9

after downgrade to 0.200.9 evcc ist starting without any error.

orthoenergy, 2025-03-16:
restarted successful after evcc migrate -r just by yaml

methesecond, 2025-03-16:
I will never ever update this s. Don’t know if you test the software.

methesecond, 2025-03-16:
Again. Every update is a catastrophe.

NFea-r, 2025-03-16:
> apt install evcc=0.200.9
> 
> after downgrade to 0.200.9 evcc ist starting without any error.

Unfortunately this does not work for me...

Tedfierro, 2025-03-16:
Downgrade auf 0.200.6 behebt es vorerst...

NFea-r, 2025-03-16:
> Downgrade auf 0.200.6 behebt es vorerst...

Auch bei 0.200.6 funktioniert es bei mir nicht.
Bei mir läuft das ganze auf einem RaspberryPi uns Raspberry Pi OS. apt get update hat heute neben den evcc Update von 0.200.9 auf 0.201.0 auch noch jede menge andere Updates gezogen da das Bookworm Repository von 12.9 auf 12.10 geändert wurde. Aber da wir hier alle mit Easee Cahragern Probleme haben, nehme ich nicht an, dass es daran liegt

Tedfierro, 2025-03-16:
Ja, liegt an Easee... Im Easee Portal kann ich mich nicht anmelden - Fehler wohl nicht auf EVCCs Seite.

svenp, 2025-03-16:
> Ja, liegt an Easee... Im Easee Portal kann ich mich nicht anmelden - Fehler wohl nicht auf EVCCs Seite.

Dann wäre das error handling auf evcc Seite zwischen den Versionen 0.200.9 und 0.20.10 gerändert worden, das die 0.200.9 bei mir ja normal startet

Tedfierro, 2025-03-16:
Ich bekomme diverse Fehler im Easee Portal. An der Easee Integration hat sich doch gar nichts geändert oder bin ich falsch?

NFea-r, 2025-03-16:
Bekomme auch sporadisch Fehler auf der Easee-Seite. Anscheinend kann man Glück haben und einen funktionierenden Slot bei Easee erwischen, dann ist die evcc Version egal.

svenp, 2025-03-17:
> Bekomme auch sporadisch Fehler auf der Easee-Seite. Anscheinend kann man Glück haben und einen funktionierenden Slot bei Easee erwischen, dann ist die evcc Version egal.

Wenn das die Ursache ist, sollte evcc das Error handling anpassen. Eine nicht erreichbare Seite kann in dem Fall ja eine Fehlermeldung erzeugen, darf aber meiner Meinung nach nicht dazu führen das EVCC überhaupt nicht mehr startet

scharrin2, 2025-03-17:
same issue

andig, 2025-03-17:
Die Easee Cloud API ist gestört: https://status.easee.com

nicx, 2025-03-17:
@andig eine Idee: Wäre es technisch möglich bei bestehender Störung einer Cloud-API zb. den Charger dann in EVCC als "gestört" darzustellen anstatt dass EVCC gar nicht erst startet? Bei mir ist EVCC inzwischen so zentral (durch zb den modbus-proxy), dass ich auf den Charger verzichten kann, nicht aber auf EVCC selbst ;)

GrimmiMeloni, 2025-03-17:
Easee fixed their cloud earlier this morning.
TL;DR no evcc problem. Closing.

closed 2025-03-17 by GrimmiMeloni