evcc-io/evcc #19866: max charge current 4A: invalid current 4

Jambola77, 2025-03-17:
### Describe the bug

I have the above error message in the logfile, but have no min or max charge setting in the config file.

### Steps to reproduce

Start evcc with my config ;-)

### Configuration details

```yaml
network:
        schema: http
        host: 127.0.0.1
        port: 7070
interval: 30s

log: info
levels:
#  core: debug
#  lp-1: debug
  core: warn
  lp-1: warn

mqtt:
  broker: 192.168.1.100:1883
  topic: evcc

meters:
- name: SDM630
  type: custom
  power:
    source: http
    uri: http://192.168.1.123/cm?cmnd=Status0
    jq: .StatusSNS.SDM630.Power_all

- name: pv
  type: custom
  power:
    source: mqtt
    broker: localhost:1883
    timeout: 30s
    topic: solarview/WR0/PAC
    scale: 1

- name: mikrowr
  type: custom
  power:
    timeout: 30s
    source: mqtt
    broker: localhost:1883
    topic: tele/DDS238/SENSOR
    jq: .ENERGY.Power

- name: grid
  type: custom
  power:
    source: mqtt
    broker: localhost:1883
    timeout: 30s
    topic: cerbo/ACPOWER

- name: mybat
  type: custom
power:
    source: mqtt
    broker: localhost:1883
    timeout: 30s
    topic: cerbo/DCPOWER
    scale: -1
  soc:
    source: mqtt
    broker: localhost:1883
    timeout: 30s
    topic: cerbo/SoC

vehicles:
- name: ID5
  type: template
  template: id
  title: ID5
  capacity: 79
  user: <deleted>
  password: <deleted>

chargers:
- type: template
  template: wallbe
  host: 192.168.1.251
  port: 502
  name: wallbox1

loadpoints:
- title: Carport
  charger: wallbox1
  meter: SDM630
  mode: pv
  vehicle: ID5
  disable:
    delay: 1m
    threshold: -200

site:
- title: SPHome
  residualPower: 200
  meters:
          pv:
            - pv
            - mikrowr
          grid: grid
          battery: mybat
```

### Log details

```text
[site ] DEBUG 2025/03/17 10:04:52 ----
[lp-1 ] DEBUG 2025/03/17 10:04:52 charge power: 0W
[site ] DEBUG 2025/03/17 10:04:52 grid power: -307W
[site ] DEBUG 2025/03/17 10:04:52 pv 1 power: 4931W
[site ] DEBUG 2025/03/17 10:04:52 pv 2 power: 301W
[site ] DEBUG 2025/03/17 10:04:52 pv power: 5232W
[site ] DEBUG 2025/03/17 10:04:52 battery 1 power: -2628W
[site ] DEBUG 2025/03/17 10:04:52 battery 1 soc: 41%
[site ] DEBUG 2025/03/17 10:04:52 site power: -2735W
[lp-1 ] DEBUG 2025/03/17 10:04:52 charger status: B
[lp-1 ] DEBUG 2025/03/17 10:04:52 pv charge current: min 4A > 3.96A (-2735W @ 3p, battery: false)
[lp-1 ] ERROR 2025/03/17 10:04:52 max charge current 4A: invalid current 4 

[...]

[site ] DEBUG 2025/03/17 10:05:22 ----
[lp-1 ] DEBUG 2025/03/17 10:05:22 charge power: 0W
[site ] DEBUG 2025/03/17 10:05:22 grid power: -1751W
[site ] DEBUG 2025/03/17 10:05:22 battery 1 power: -2666W
[site ] DEBUG 2025/03/17 10:05:22 battery 1 soc: 41%
[site ] DEBUG 2025/03/17 10:05:22 pv 2 power: 301W
[site ] DEBUG 2025/03/17 10:05:22 pv 1 power: 4939W
[site ] DEBUG 2025/03/17 10:05:22 pv power: 5240W
[site ] DEBUG 2025/03/17 10:05:22 site power: -4217W
[lp-1 ] DEBUG 2025/03/17 10:05:22 charger status: B
[lp-1 ] DEBUG 2025/03/17 10:05:22 pv charge current: 6.11A = 0A + 6.11A (-4217W @ 3p)
[lp-1 ] DEBUG 2025/03/17 10:05:22 max charge current: 6.11A
[lp-1 ] DEBUG 2025/03/17 10:05:22 charger enable
[lp-1 ] DEBUG 2025/03/17 10:05:22 wake-up timer: start
[site ] DEBUG 2025/03/17 10:05:52 ----
[lp-1 ] DEBUG 2025/03/17 10:05:52 charge power: 0W
[site ] DEBUG 2025/03/17 10:05:52 grid power: -1998W
[site ] DEBUG 2025/03/17 10:05:52 pv 1 power: 4952W
[site ] DEBUG 2025/03/17 10:05:52 pv 2 power: 301W
[site ] DEBUG 2025/03/17 10:05:52 pv power: 5253W
[site ] DEBUG 2025/03/17 10:05:52 battery 1 power: -2486W
[site ] DEBUG 2025/03/17 10:05:52 battery 1 soc: 41%
[site ] DEBUG 2025/03/17 10:05:52 site power: -4284W
[lp-1 ] DEBUG 2025/03/17 10:05:52 charger status: B
[lp-1 ] DEBUG 2025/03/17 10:05:52 pv charge current: 6.21A = 0A + 6.21A (-4284W @ 3p)
[lp-1 ] DEBUG 2025/03/17 10:05:52 max charge current: 6.21A
[site ] DEBUG 2025/03/17 10:06:22 ----
[lp-1 ] DEBUG 2025/03/17 10:06:22 charge power: 4033W
[site ] DEBUG 2025/03/17 10:06:22 grid power: 197W
[site ] DEBUG 2025/03/17 10:06:22 pv 2 power: 301W
[site ] DEBUG 2025/03/17 10:06:22 pv 1 power: 4954W
[site ] DEBUG 2025/03/17 10:06:22 pv power: 5255W
[site ] DEBUG 2025/03/17 10:06:22 battery 1 power: -871W
[site ] DEBUG 2025/03/17 10:06:22 battery 1 soc: 41%
[site ] DEBUG 2025/03/17 10:06:22 site power: -474W
[lp-1 ] DEBUG 2025/03/17 10:06:22 charger status: C
[lp-1 ] INFO 2025/03/17 10:06:22 start charging ->
[lp-1 ] DEBUG 2025/03/17 10:06:22 wake-up timer: stop
[lp-1 ] DEBUG 2025/03/17 10:06:23 soc estimated: 47.00% (vehicle: 47.00%)
[lp-1 ] DEBUG 2025/03/17 10:06:23 vehicle soc: 47%
[lp-1 ] DEBUG 2025/03/17 10:06:23 vehicle soc limit: 80%
[lp-1 ] DEBUG 2025/03/17 10:06:23 vehicle range: 196km
[lp-1 ] DEBUG 2025/03/17 10:06:23 pv charge current: 6.9A = 6.21A + 0.687A (-474W @ 3p)
[lp-1 ] DEBUG 2025/03/17 10:06:23 max charge current: 6.9A
[site ] DEBUG 2025/03/17 10:06:52 ----
[lp-1 ] DEBUG 2025/03/17 10:06:52 charge power: 4287W
[site ] DEBUG 2025/03/17 10:06:52 grid power: 12W
[site ] DEBUG 2025/03/17 10:06:52 pv 1 power: 4959W
[site ] DEBUG 2025/03/17 10:06:52 pv 2 power: 300W
[site ] DEBUG 2025/03/17 10:06:52 pv power: 5259W
[site ] DEBUG 2025/03/17 10:06:52 battery 1 power: -409W
[site ] DEBUG 2025/03/17 10:06:52 battery 1 soc: 41%
[site ] DEBUG 2025/03/17 10:06:52 site power: -197W
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.201.0

Jambola77, 2025-03-17:
Some more errors in the GUI:

[image]

VolkerK62, 2025-03-17:
Check the UI-Loadpoint Settings
I guess, the minimum is set to 4A

StefanSchoof, 2025-03-17:
And 4A is to low. You need at least 6A.

Jambola77, 2025-03-17:
Thanks, now I have found it in the UI and changed to 6A.

closed 2025-03-17 by Jambola77