evcc-io/evcc #19868: Problems with enyaq connect

rpourafzal-herbers, 2025-03-17:
### Describe the bug

Hi @andig,

since a few days i get this problem, so that there is no SoC status of enyaq and my enyaq is shown "offline":

"[main  ] ERROR 2025/03/17 10:03:19 creating vehicle enyaq80 failed: cannot create vehicle type 'template': cannot create vehicle type 'enyaq': invalid character 'c' looking for beginning of object key string
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'enyaq': invalid character 'c' looking for beginning of object key string
Capacity: 80.0kWh
Features: [Offline Retryable]"

evcc ist up-to-date (evcc 0.131.12)

Regards
Ramin

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
vehicles:
- type: template
  template: enyaq
  title: Enyaq iV 80
  user: <user>
  password: <password>
  vin: TMB****
  capacity: 80
  phases: 3
  mode: pv
  # minSoC: 5
  # targetSoC: 80
  minCurrent: 6
  maxCurrent: 16
  cloud: false
  name: enyaq80
```

### Log details

```text
[main  ] ERROR 2025/03/17 10:03:19 creating vehicle enyaq80 failed: cannot create vehicle type 'template': cannot create vehicle type 'enyaq': invalid character 'c' looking for beginning of object key string
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'enyaq': invalid character 'c' looking for beginning of object key string
Capacity: 80.0kWh
Features: [Offline Retryable]
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.12

rpourafzal-herbers, 2025-03-17:
No it is working, newer evcc version is now installed.. Or is it the nightly build?

Mar 17 10:28:37 coolness-services evcc[51445]: [main  ] INFO 2025/03/17 10:28:37 evcc 0.201.0 (f801592b)

# evcc -v
evcc version 0.201.0 (f801592b)

StefanSchoof, 2025-03-17:
> evcc ist up-to-date (evcc 0.131.12)

Aktuell ist 0.201.0 und in dem Bereich wurde was gefixt

StefanSchoof, 2025-03-17:
> No it is working

Not working or now working?

zimbo86, 2025-03-17:
I am on 200.9 and my SOC is not refreshed when car connects.
Will try 201.0

```
[lp-1 ] DEBUG 2025/03/17 10:26:38 charge total import: 704.735kWh
[lp-1 ] DEBUG 2025/03/17 10:26:38 charger status: B
[lp-1 ] DEBUG 2025/03/17 10:26:38 charge start energy: 704.735kWh
[lp-1 ] INFO 2025/03/17 10:26:38 car connected
[lp-1 ] DEBUG 2025/03/17 10:26:38 pv timer elapse
[lp-1 ] DEBUG 2025/03/17 10:26:38 pv timer inactive
[lp-1 ] DEBUG 2025/03/17 10:26:38 charge total import: 704.735kWh
[lp-1 ] DEBUG 2025/03/17 10:26:40 soc estimated: 0.00% (vehicle: 0.00%)
[lp-1 ] DEBUG 2025/03/17 10:26:40 vehicle soc: 0%
[lp-1 ] DEBUG 2025/03/17 10:26:40 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/03/17 10:26:40 vehicle range: 0km
[lp-1 ] DEBUG 2025/03/17 10:26:40 max charge current: 16A
[lp-1 ] DEBUG 2025/03/17 10:26:40 charger enable
[lp-1 ] DEBUG 2025/03/17 10:26:40 wake-up timer: start
[site ] DEBUG 2025/03/17 10:26:44 set priority soc: 10
[site ] DEBUG 2025/03/17 10:27:04 ----
[lp-1 ] DEBUG 2025/03/17 10:27:04 charge power: 10447W
[lp-1 ] DEBUG 2025/03/17 10:27:04 charge currents: [15.2 15 15.6]A
[site ] DEBUG 2025/03/17 10:27:04 battery 1 power: 1896W
[site ] DEBUG 2025/03/17 10:27:04 battery 1 soc: 94%
[site ] DEBUG 2025/03/17 10:27:04 pv 1 power: 5514W
[site ] DEBUG 2025/03/17 10:27:05 grid power: 4189W
[site ] DEBUG 2025/03/17 10:27:06 grid powers: [149 1669 1820]W
[site ] DEBUG 2025/03/17 10:27:08 grid currents: [3.74 7.32 7.98]A
[site ] DEBUG 2025/03/17 10:27:08 site power: 6085W
[lp-1 ] DEBUG 2025/03/17 10:27:08 vehicle odometer: 3163km
[lp-1 ] DEBUG 2025/03/17 10:27:08 charge total import: 704.798kWh
[lp-1 ] DEBUG 2025/03/17 10:27:08 charger status: C
[lp-1 ] DEBUG 2025/03/17 10:27:08 charge start energy: 704.798kWh
[lp-1 ] INFO 2025/03/17 10:27:08 **start charging ->**
[lp-1 ] DEBUG 2025/03/17 10:27:08 wake-up timer: stop
[lp-1 ] DEBUG 2025/03/17 10:27:09 soc estimated: 0.00% (vehicle: 0.00%)
[lp-1 ] DEBUG 2025/03/17 10:27:09 vehicle soc: 0%
[lp-1 ] DEBUG 2025/03/17 10:27:09 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/03/17 10:27:09 **vehicle range: 0km**
[site ] DEBUG 2025/03/17 10:27:34 ---- 
............
[lp-1 ] INFO 2025/03/17 10:30:08 stop charging <-
[lp-1 ] DEBUG 2025/03/17 10:30:09 charge total import: 705.253kWh
[lp-1 ] DEBUG 2025/03/17 10:30:15 vehicle status: B
[lp-1 ] DEBUG 2025/03/17 10:30:15 vehicle soc: 71%
[lp-1 ] DEBUG 2025/03/17 10:30:15 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/03/17 10:30:15 vehicle range: 280km
[lp-1 ] DEBUG 2025/03/17 10:30:15 pv charge current: 3.78A = 0A + 3.78A (-2605W @ 3p)
[lp-1 ] DEBUG 2025/03/17 10:30:15 pv enable timer reset
[lp-1 ] DEBUG 2025/03/17 10:30:15 pv timer inactive
[site ] DEBUG 2025/03/17 10:30:34 ----
```

and the vehicle Trace:
``` 
[enyaq ] TRACE 2025/03/17 10:26:38 POST https://mysmob.api.connect.skoda-auto.cz/api/v1/authentication/refresh-token?tokenType=CONNECT
[enyaq ] TRACE 2025/03/17 10:26:39 {"token":"xyz"}
[enyaq ] TRACE 2025/03/17 10:26:39 GET https://mysmob.api.connect.skoda-auto.cz/api/v1/charging/***
[enyaq ] TRACE 2025/03/17 10:26:40 {"isVehicleInSavedLocation":false,"status":{"battery":{}},"settings":{"targetStateOfChargeInPercent":100,"batteryCareModeTargetValueInPercent":80,"preferredChargeMode":"MANUAL","availableChargeModes":["MANUAL"],"chargingCareMode":"DEACTIVATED","autoUnlockPlugWhenCharged":"OFF","maxChargeCurrentAc":"MAXIMUM","batterySupport":"ENABLED"},"carCapturedTimestamp":"2025-03-17T09:25:26Z","errors":[{"type":"STATUS_OF_CHARGING_NOT_AVAILABLE","description":"Status of charging is not available."},{"type":"STATUS_OF_CONNECTION_NOT_AVAILABLE","description":"Status of connection is no available."}]}
[enyaq ] TRACE 2025/03/17 10:27:08 GET https://mysmob.api.connect.skoda-auto.cz/api/v1/vehicle-health-report/warning-lights/***
[enyaq ] TRACE 2025/03/17 10:27:08 {"capturedAt":"2025-03-17T09:20:25.748Z","mileageInKm":3163,"warningLights":[{"category":"ASSISTANCE","defects":[]},{"category":"COMFORT","defects":[]},{"category":"BRAKE","defects":[]},{"category":"ELECTRIC_ENGINE","defects":[]},{"category":"LIGHTING","defects":[]},{"category":"TIRE","defects":[]},{"category":"OTHER","defects":[]}]}
[enyaq ] TRACE 2025/03/17 10:29:39 GET https://mysmob.api.connect.skoda-auto.cz/api/v2/air-conditioning/***
[enyaq ] TRACE 2025/03/17 10:29:42 {"state":"OFF","runningRequests":[],"targetTemperature":{"temperatureValue":22.0,"unitInCar":"CELSIUS"},"airConditioningAtUnlock":false,"windowHeatingEnabled":false,"steeringWheelPosition":"LEFT","seatHeatingActivated":{"frontLeft":false,"frontRight":false},"chargerConnectionState":"CONNECTED","chargerLockState":"LOCKED","windowHeatingState":{"front":"OFF","rear":"OFF","unspecified":"INVALID"},"timers":[{"id":1,"enabled":false,"time":"00:00","type":"ONE_OFF","selectedDays":["SATURDAY"]},{"id":2,"enabled":false,"time":"00:00","type":"ONE_OFF","selectedDays":["SATURDAY"]}],"carCapturedTimestamp":"2025-03-17T09:25:26Z","outsideTemperature":{"temperatureValue":3.5,"temperatureUnit":"CELSIUS","carCapturedTimestamp":"2025-03-17T09:25:24.259Z"},"errors":[]}
[enyaq ] TRACE 2025/03/17 10:30:09 GET https://mysmob.api.connect.skoda-auto.cz/api/v1/charging/***
[enyaq ] TRACE 2025/03/17 10:30:10 {"isVehicleInSavedLocation":false,"status":{"chargingRateInKilometersPerHour":0.0,"chargePowerInKw":0.0,"state":"CONNECT_CABLE","battery":{"remainingCruisingRangeInMeters":280000,"stateOfChargeInPercent":71}},"settings":{"targetStateOfChargeInPercent":100,"batteryCareModeTargetValueInPercent":80,"preferredChargeMode":"M [... cut ...]
[enyaq ] TRACE 2025/03/17 10:45:39 {"isVehicleInSavedLocation":false,"status":{"chargingRateInKilometersPerHour":18.0,"chargePowerInKw":3.6,"remainingTimeToFullyChargedInMinutes":370,"state":"CHARGING","chargeType":"AC","battery":{"remainingCruisingRangeInMeters":283000,"stateOfChargeInPercent":72}},"settings":{"targetStateOfChargeInPercent":100,"batteryCareModeTargetValueInPercent":80,"preferredChargeMode":"MANUAL","availableChargeModes":["MANUAL"],"chargingCareMode":"DEACTIVATED","autoUnlockPlugWhenCharged":"OFF","maxChargeCurrentAc":"MAXIMUM","batterySupport":"ENABLED"},"carCapturedTimestamp":"2025-03-17T09:38:03Z","errors":[]}
```

rpourafzal-herbers, 2025-03-17:
> > No it is working
> 
> Not working or now working?

Its working with evcc version 0.201.0, but is this now a nightly build?

zimbo86, 2025-03-17:
> > > No it is working
> > 
> > 
> > Not working or now working?
> 
> Its working with evcc version 0.201.0, but is this now a nightly build?

no its not:

[image]

Should I open a separate issue for my problem a post above so that we can close this issue?

andig, 2025-03-17:
> evcc ist up-to-date (evcc 0.131.12)

Not really ;)

closed 2025-03-17 by andig

rpourafzal-herbers, 2025-03-17:
> > evcc ist up-to-date (evcc 0.131.12)
> 
> Not really ;)

I don't know why, but I updated all packages via apt several times a week, today I executed the setup script for unstable releases, perhaps it fixed this problem.