evcc-io/evcc #19870: Add sensibo as control for heatpumps

timpetersson81, 2025-03-17:
Sensibo can control a various of heatpumps. It works as a remote control for the heatpump, and I use to crontrol my heatpump today.

https://support.sensibo.com/api/

andig, 2025-03-17:
Having had a quick look at the api I'm unclear how this would apply. How would you (in terms of `sgready`) set boost vs. normal mode?

timpetersson81, 2025-03-17:
I switch off doing high peak today, and on when surplus or low peak. Just in normal mode. Boost mode can be used as you suggest, but is will make it a lot noisy. I would prefer just turning up or down the temperature if I can those.

andig, 2025-03-17:
How exactly would you do any of that?

closed 2025-03-21 by andig

timpetersson81, 2025-03-21:
> How exactly would you do any of that, especially setting normal or boost mode? What do you mean by noisy?

Me heatpump is air to air, so boost speeds up the fan and make noise. Today I just use home assistant automations to turn on and off the heatpump in high and low peak hours.

andig, 2025-03-21:
> I just use home assistant automations to turn on and off the heatpump in high and low peak hours.

Given https://github.com/evcc-io/evcc/issues/19870#issuecomment-2729780945 is unanswered I've closed here. Please feel free to propose a template PR for what you have in mind. Core team won't work on this.