evcc-io/evcc #19873: OVMS Cannot unmarshal string

spikey-de, 2025-03-17:
### Describe the bug

Hi folks,

since yesterday's update I get the message

vehicle soc: json: cannot unmarshal string into Go struct field ChargeResponse.staleambient of type int

in combination with a OVMS car. In Dexters Web portal all data are correct, so I assume any error between the the server and local EVCC.

Already known?

Spikey

### Steps to reproduce

1. vehicle soc: json: cannot unmarshal string into Go struct field ChargeResponse.staleambient of type int ERROR during start
2.
3.
...

### Configuration details

```yaml
Running in HA
```

### Log details

```text
[lp-1 ] ERROR 2025/03/17 11:31:48 vehicle soc: json: cannot unmarshal string into Go struct field ChargeResponse.staleambient of type int
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

201.0

CaptainRoot, 2025-03-17:
Same Error here, may be related to https://github.com/evcc-io/evcc/pull/19796

VolkerK62, 2025-03-17:
https://github.com/evcc-io/evcc/discussions/19867#discussioncomment-12522832

andig, 2025-03-17:
Could you share a trace log?

andig, 2025-03-17:
Closed in 97b4b9aeda1a1b0ab63d37512da9385c6d4c1776

closed 2025-03-17 by andig