evcc-io/evcc #19879: EVCC Ladeenergieanzeige im Ladevorgang fehlerhaft

BastiZ0, 2025-03-17:
### Describe the bug

Beim manuellen sowie automatischen Start des Ladevorgangs kommt es regelmäßig zu keiner Aktualisierung der geladenen Energie in der Übersicht des Ladepunktes. 

Gleichzeitig wird auch der Ladevorgang nicht korrekt in der Ladevorgängeübersicht angezeigt. 

Dieser Fehler tritt nicht jedes Mal auf. Er kann meist mit einem Neustart des Addons in Home Assistant behoben werden. Allerdings kommt es beim Umschalten von 1 auf 3-Phasenbetrieb oder auch umgekehrt wieder zum selbigen Fehler. 

Es kommt vor, dass die Lademenge nach einer unbestimmten Zeit dann doch anfängt hochzuzählen. Die Ladeleistung und die Energie, die in der Übersicht beim Energiezähler der Wallbox stehen, sind korrekt. Nur die geladene Energie und damit auch die Verhältnisse von Solar, Netz, Co2 etc, stimmen nicht.

### Steps to reproduce

1. Starten des Ladevorgangs
2. Fehler tritt direkt mit sehr hoher Wahrscheinlichkeit auf
3. neustart kann Problem beheben, muss aber nicht.

### Configuration details

```yaml
gibt es nicht mehr. meine migrierte war

network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: 192.168.79.1
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# trial token, valid until 2065-10-09
sponsortoken: 

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true
plant: 
# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
[... 266 lines cut ...]
  #   priority: <priority>
  #   tags: <tags>
```

### Log details

```text
wird nachgereicht
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.201.0

BastiZ0, 2025-03-17:
Hier der Trace log.

[evcc-20250317-150129-trace.log](https://github.com/user-attachments/files/19288452/evcc-20250317-150129-trace.log)

andig, 2025-03-17:
Zeigt das Logfile das Problem?

BastiZ0, 2025-03-17:
Das kann ich dir leider nicht genau sagen. Was ich sehe ist, dass LP-1, was vermutlich der Loadpoint der Sungrow Wallbox ist, tatsächlich mit dem charge total import langsam entsprechend der Energie, weiterzählt, so wie es sein soll.

Die Anzeige geschieht dies aber nicht. Da bleibt die Anzeige "eingefroren" und alle relevanten Berechnungen ebenfalls. Ich weiß, du magst keine Screenshots, nur zur Sicherstellung, dass ich nicht missverstanden werde.

[image]

Hier sind es 1,2kWh. Während des Ladevorgangs zeigt er mir die korrekte Leistung an, zählt hier aber nicht weiter. Das ist nicht immer so und kann durch Neustart von EVCC oft behoben werden.

tantive, 2025-03-17:
Ich kann den Fehler bestätigen. Tritt bei mir mit der Sungrow Wallbox ebenfalls auf.

andig, 2025-03-17:
Es braucht ein Log vom Fehler. Irgendeins hilft leider nicht.

BastiZ0, 2025-03-19:
Das verstehe ich. Irgendwoher muss das ja aber kommen und ich würde den Fehler gerne ausmerzen. :)

Ich habe das jetzt gerade nochmal getestet. Mit der go-e Wallbox funktioniert es ganz normal. Die Ladeenergie wird direkt auf der Übersicht neben der Leistung angezeigt und zählt auch hoch, egal ob die Phase umgeschaltet wird. Habe gerade mehrmals umgeschaltet und er hat trotzdem weitergezählt.

Bei der Sungrow Wallbox hat er am Anfang auch zufällig mal angefangen den "Geladen" - Counter zu zählen. Nach der Umschaltung ist er allerdings bei 4kWh stehen geblieben. Obwohl in der TRACE Log der LP-1 Counter (charge total import) weitergezählt hat. Das Umschalten war um 12:23. Um 12:34 Uhr habe ich nochmals umgeschaltet, in der Übersicht weiter 4kWh, der Zähler LP-1 im Log hat sich allerdings von 1326,653 auf 1327,256 erhöht. Insgesamt fehlen mir seit dem Umschalten und nicht registrieren 2,5kWh bei "Geladen" und damit diese auch in der Übersicht, bevor ich zum Go-e gewechselt bin.

[image]

Ich würde dir gern sagen, wo der Fehler liegt. Im Tracelog sehe ich nur das beide Wallboxen "normal" zählen. Wobei der Go-e deutlich öfter die charge total import aktualisiert im gegensatz zur Sungrow WB. 

Woher bezieht EVCC denn die Werte, die auf der Übersichts(Home)seite angezeigt werden? Wo muss ich nachschauen, ob die korrekt aktualisiert werden. 

Ich hänge dir trotzdem nochmal 2 Tracelogs dran. Zeitstempel wie folgt:

12:23 -> auf einphasig bis 12:34 (charge total import von 1326,653 auf 1327,256) -keine Änderung in der Übersicht

[evcc-20250319-124133-trace.log](https://github.com/user-attachments/files/19340887/evcc-20250319-124133-trace.log)

12:55 wechsel auf go-e charger. Laden ab 12:57Uhr 

[evcc-20250319-130803-trace.log](https://github.com/user-attachments/files/19340890/evcc-20250319-130803-trace.log)

andig, 2025-03-19:
> Woher bezieht EVCC denn die Werte, die auf der Übersichts(Home)seite angezeigt werden? Wo muss ich nachschauen, ob die korrekt aktualisiert werden.

Gute Idee. Schau mal bitte unter /api/state?jq=.loadpoints und dann  `chargeTotalImport` und/oder `chargedEnergy`. Zählen die beide weiter?

BastiZ0, 2025-03-20:
> Gute Idee. Schau mal bitte unter /api/state?jq=.loadpoints und dann `chargeTotalImport` und/oder `chargedEnergy`. Zählen die beide weiter?

Gerade getestet. Verändern tut sich entsprechend der Leistung nur `chargeTotalImport`. `chargedEnergy` bleibt bei `0`, sowie alle `session` ausgaben bei `null`

BastiZ0, 2025-03-21:
Habe eben noch einmal ein kW geladen. Verändert habe ich nichts direkt, allerdings HA und damit EVCC neu gestartet. Er hat dieses mal sporadisch die Energie gezählt. Unter der API ist nun auch `"chargedEnergy":1035` einsehbar und wird auch im GUI angezeigt. Wie gesagt tritt dieses Problem nur bei der Sungrow Wallbox auf.

andig, 2025-03-21:
Es gibt zwei Möglichkeiten:

- die Sungrow liefert falsche/keine Werte
- evcc verarbeitet sie falsch

Das kannst Du im Logfile (trace) raus finden. Falls SG falsche Werte liefert können wir schauen den Wert evtl. nicht aus der WB zu nehmen oder Sungrow löst das Problem.

Analyse welcher Fall es ist würde ich gerne Dir überlassen, die Werkzeuge sind da.

aspengler, 2025-03-22:
Ich habe ebenfalls eine Sungrow und bin hierher verwiesen worden.

Habe evcc mit "lp-1: trace" neu gestartet und eben mal unseren Cupra Born mit Min+PV angefangen zu laden (14:36 Uhr). Um 15:29 Uhr auf preisorientiertes Schnellladen (1p -> 3p) umgeschaltet. Um 15:53 Uhr hat evcc wegen Erreichens der 80% SOC aufgehört zu laden.

Das ist die Detailansicht des Ladevorgangs in evcc:

[image]

Das Log (nur für lp-1) habe ich angehängt. Auffällig sind IMHO folgende Zeilen:

Mar 22 14:41:20 homepi evcc[10677]: [lp-1  ] DEBUG 2025/03/22 14:41:20 charge total import: 166.007kWh
Mar 22 14:41:30 homepi evcc[10677]: [lp-1  ] DEBUG 2025/03/22 14:41:30 charge total import: 162.784kWh

und später noch

Mar 22 15:34:10 homepi evcc[10677]: [lp-1  ] DEBUG 2025/03/22 15:34:10 charge total import: 165.184kWh
Mar 22 15:34:20 homepi evcc[10677]: [lp-1  ] DEBUG 2025/03/22 15:34:20 charge total import: 164.696kWh

[evcc_20250322_charge.log](https://github.com/user-attachments/files/19403873/evcc_20250322_charge.log)

andig, 2025-03-23:
Welcher der beiden Fälle ist es denn? Wir können die Ladeenergie gerne entfernen wenn die WB die nicht richtig liefert.

aspengler, 2025-03-23:
Gute Frage: mir scheint es falsch zu sein, wenn die Wallbox binnen 10 Sekunden gut 3 kWh Strom "vergisst"
Allerdings hatte ich auch schon Fälle, da sahen die Werte der (gleichen) Wallbox ok aus, aber der "Geladen" Wert war einfach der Wert des vorherigen Vorgangs (#19883)

Tue mich da schwer mit einer definitiven Aussage...

BastiZ0, 2025-03-23:
> Es gibt zwei Möglichkeiten:
> 
> * die Sungrow liefert falsche/keine Werte
> * evcc verarbeitet sie falsch

Tatsächlich liefert die Wallbox bei mir plausible "live" Werte unter `/api/state?jq=.loadpoints.` Ich habe in HA zusätzlich einen Rest Sensor angelegt, der sich über eure API den `chargeTotalImport` abgreift. Dieser ist übereinstimmend mit meinem Shelly Sensor, der die 3 Phasen misst. Es ist bei mir also nicht die WB, die falsche Werte in der API liefert. `chargedEnergy` wird als Attribut mit abgegriffen. Dieser zählt nicht bei jedem Ladevorgang mit oder bleibt bei Phasenwechsel "stehen". 

Es liegt m.E. also irgendwie an der "Verarbeitung" der Werte, die die WB zur Verfügung stellt.

> Allerdings hatte ich auch schon Fälle, da sahen die Werte der (gleichen) Wallbox ok aus, aber der "Geladen" Wert war einfach der Wert des vorherigen Vorgangs ([#19883](https://github.com/evcc-io/evcc/discussions/19883))

@aspengler 
Dieses Problem, welches du beschreibst, tritt bei mir auf, wenn der `chargedEnergy` Wert einfach stehen bleibt. Dann kann es auch vorkommen, dass in der Übersicht der Ladevorgang doppelt eingetragen wird. Das `chargeTotalImport`, also der Energiezähler der WB hoch und runterläuft, das hatte ich tatsächlich bisher nicht. Da scheint es ein Problem mit deiner WB oder der Übertragung zu geben. Ich habe sie über ein RS485 to ETH TCP/IP im Netzwerk und mit HA ( EVCC ) verbunden. Ich habe auf einem Pi4 auch EVCC einmal zur Probe installiert, um auszuschließen, dass es am HA System liegt. Auch hier sind die Werte der WB übereinstimmend mit den Werten in HA.

andig, 2025-03-23:
> Dieser ist übereinstimmend mit meinem Shelly Sensor

Du bist sicher, dass Du das durchgängig beobachtet hast?

> Dieser zählt nicht bei jedem Ladevorgang mit oder bleibt bei Phasenwechsel "stehen".

Dann brauchts davon bitte ein Logfile in dem das nachvollziehbar wäre. Ohne finden wir die Ursache nicht.

BastiZ0, 2025-03-23:
> Du bist sicher, dass Du das durchgängig beobachtet hast?

Ja, definitiv. Ich saß am Rechner und habe manuell die `/api/state?jq=.loadpoints` im Browser aktualisiert, während ich die WB über EVCC durch mehrere Szenarien gebracht habe. Unteranderem beim Umschalten auf 1P-Betrieb. Die WB hat unter `chargeTotalImport` weitergezählt und `chargedEnergy` ist stehengeblieben. Deckt sich wie gesagt auch mit den Aufzeichnungen von HA und den Attributen, die ich dafür angelegt habe.

Ich hatte die Woche schon mal 2 Logs rausgezogen, wo genau dieses Verhalten zu sehen war. Ich hoffe ich komme nachher dazu, diesen Abschnitt nochmal aufzuarbeiten und dir dann zur Verfügung zu stellen.

mfuchs1984, 2025-03-27:
Wenn die Session Energy unzuverlässig ist, könnt ihr sie ja in evcc Mal raus nehmen. Beispiel https://github.com/evcc-io/evcc/pull/20001/files

steppix, 2025-04-02:
Ich habe ein ähnliches Problem mit der Sungrow Wallbox. Bei mir wird die Ladeenergie als Netzbezug "erkannt", obwohl ich heute überhaupt keinen Netzbezug hatte. Die Statistik ist hinüber und ich habe keine Ahnung, was ich da machen kann. In der Konfiguration von evcc sehe ich, das kein Netzbezug stattgefunden hat. evcc bekommt korrekte Daten und zeigt diese in der Konfiguration auch an.

BastiZ0, 2025-04-12:
@andig 

Würde gern mit deiner Unterstützung versuchen, das Problem mit der SUNGROW Wallbox zu finden, damit ihr diesen ausmerzen könnt.. Was kann ich tun, um hier eine weitere Eingrenzung, woher dieser Verarbeitungsfehler kommt, vorzunehmen?

tantive, 2025-04-19:
Ich habe meine Wallbox gestern auf 1.7.186 aktualisiert. Sieht momentan besser/konsistenter aus. Ich beobachte das mal in den nächsten Tagen.

andig, 2025-04-27:
Dann hoffen wir mal, dass es damit gelöst ist...

closed 2025-04-27 by andig

BastiZ0, 2025-05-04:
Leider nein! Problem besteht in Gänze weiterhin. Kein Rest des Ladezählers nach erneutem Anstecken der Wallbox ans Fahrzeug & Energieanzeige zählt nicht immer von beginn an.

andig, 2025-05-04:
Ticket bei Sungrow geöffnet?

hameno, 2025-05-07:
Ich denke das ist kein Problem vom charger selbst. Der Session Zähler in der HA Integration zählt korrekt. Ich glaube evcc hat irgendwie ein Problem mit dem reset zu 0 beim Start einer Session. Ich werde mir das in den kommenden Tagen mal anschauen. 
Konnte es jedenfalls heute wieder beobachten:

[image]
[image]
[image]

andig, 2025-05-07:
>können wir das hier wieder aufmachen?

Geschlossen wurde es weil es keinen Ansatz einer Lösung gab. Wenn es lösungsrelevante Erkenntnisse gibt gerne. "Aufmachen" alleine hilft leider nicht.

hameno, 2025-05-07:
Ich werde das hier lokal debuggen

andig, 2025-05-08:
> d.h. der Fehler liegt meines Erachtens nach irgendwo im nicht-sungrow spezifischen Teil

Der PR fixt allerdings Sungrow mit entsprechenden Nebenwirkungen.

hameno, 2025-05-08:
Ja, für einen anderen fix reichen meine Kenntnisse in Go und evcc leider noch nicht aus. Und die möglichen Nebenwirkungen für andere charger sind mir riskant.

andig, 2025-05-08:
Nur "root cause" lösen, im Ladepunkt. Deine beiden Hinweise _was_ dazu führt waren schonmal Gold wert, vielen Dank!

PR #21115 (2025-05-08): Sungrow: Remove ChargeRater implementation
The charged session energy resets on each start/stop command. evcc expects this to not reset until the car is disconnected. This breaks assumptions in loadpoint regarding the session charged energy and causes incorrect values to be calculated.

Removing the implementation causes evcc to calculate the session energy based on the MeterEnergy implementation (total energy used by wallbox) which then correctly calculates session energy.

Fixes #19879