evcc-io/evcc #19884: Preisgrenze nach Neustart von "keine" nach Neustart von System auf <=0,0 umgesetzt

jove01, 2025-03-17:
### Describe the bug

EVCC ist bis auf eine Ausnahme im UI definiert.

Ein Fahrzeug ist in evcc.yaml definiert
`vehicles:
  - name: Explorer
    type: custom
    title: Explorer
    capacity: 77 
    soc:
      source: http
      uri: http://1`

"Keine" ist bei angeschlossenem Fahrzeug definiert. Danach "Restart" im Konfiguration-UI. Hiernach springt die Anzeige auf Gastfahrzeug  wie angezeigt um.

[image]

Dann wieder auf Preisgrenze "keine" gestellt.

[image]

### Steps to reproduce

1. s.o.
2.
3.
...

### Configuration details

```yaml
s.o.
```

### Log details

```text
[site ] DEBUG 2025/03/17 18:58:46 grid power: -0W
[site ] DEBUG 2025/03/17 18:58:46 grid currents: [-0.207 0.518 -0.329]A
[site ] DEBUG 2025/03/17 18:58:46 pv 1 power: 0W
[site ] DEBUG 2025/03/17 18:58:46 battery 1 power: 490W
[site ] DEBUG 2025/03/17 18:58:46 battery 1 soc: 95%
[site ] DEBUG 2025/03/17 18:58:46 site power: 590W
[lp-1 ] DEBUG 2025/03/17 18:58:46 charge voltages: [239 237 239]V
[lp-1 ] DEBUG 2025/03/17 18:58:46 detected connected phases: 3p
[lp-1 ] DEBUG 2025/03/17 18:58:46 charge total import: 582.798kWh
[lp-1 ] DEBUG 2025/03/17 18:58:46 charger status: B
[lp-1 ] DEBUG 2025/03/17 18:58:46 vehicle api refresh
[main ] DEBUG 2025/03/17 18:58:46 vehicle status: B (Explorer)
[main ] DEBUG 2025/03/17 18:58:46 vehicle status: B (Mokka E)
[main ] WARN 2025/03/17 18:58:46 vehicle status: >1 matches, giving up
[site ] DEBUG 2025/03/17 18:58:46 solar forecast: accumulated 0.010kWh, produced 0.000kWh, scale 0.000
[site ] DEBUG 2025/03/17 18:59:15 ----
[lp-2 ] DEBUG 2025/03/17 18:59:15 charge power: 0W
[lp-1 ] DEBUG 2025/03/17 18:59:15 charge power: 4W
[lp-1 ] DEBUG 2025/03/17 18:59:16 charge currents: [0.07 0.01 0.01]A
[site ] DEBUG 2025/03/17 18:59:16 grid power: -0W
[site ] DEBUG 2025/03/17 18:59:16 grid currents: [-0.212 0.476 -0.333]A
[site ] DEBUG 2025/03/17 18:59:16 pv 1 power: 0W
[site ] DEBUG 2025/03/17 18:59:16 battery 1 power: 491W
[site ] DEBUG 2025/03/17 18:59:16 battery 1 soc: 95%
[site ] DEBUG 2025/03/17 18:59:16 site power: 591W
[lp-2 ] DEBUG 2025/03/17 18:59:16 charger status: A
[site ] DEBUG 2025/03/17 18:59:16 solar forecast: accumulated 0.010kWh, produced 0.000kWh, scale 0.000
[site ] DEBUG 2025/03/17 18:59:45 ----
[lp-2 ] DEBUG 2025/03/17 18:59:45 charge power: 0W
[lp-1 ] DEBUG 2025/03/17 18:59:45 charge power: 4W
[lp-1 ] DEBUG 2025/03/17 18:59:46 charge currents: [0.07 0.01 0.01]A
[site ] DEBUG 2025/03/17 18:59:46 grid power: -0W
[site ] DEBUG 2025/03/17 18:59:46 grid currents: [-0.216 0.48 -0.337]A
[site ] DEBUG 2025/03/17 18:59:46 pv 1 power: 0W
[site ] DEBUG 2025/03/17 18:59:46 battery 1 power: 459W
[site ] DEBUG 2025/03/17 18:59:46 battery 1 soc: 95%
[site ] DEBUG 2025/03/17 18:59:46 site power: 559W
[lp-1 ] DEBUG 2025/03/17 18:59:46 charge voltages: [239 238 239]V
[lp-1 ] DEBUG 2025/03/17 18:59:46 detected connected phases: 3p
[lp-1 ] DEBUG 2025/03/17 18:59:46 charge total import: 582.798kWh
[lp-1 ] DEBUG 2025/03/17 18:59:46 charger status: B
[site ] DEBUG 2025/03/17 18:59:46 solar forecast: accumulated 0.011kWh, produced 0.000kWh, scale 0.000
[lp-1 ] DEBUG 2025/03/17 19:00:02 set smart cost limit: <nil>
[lp-1 ] DEBUG 2025/03/17 19:00:05 set smart cost limit: <nil>
[lp-2 ] DEBUG 2025/03/17 19:00:05 set smart cost limit: <nil>
[site ] DEBUG 2025/03/17 19:00:15 ----
[lp-2 ] DEBUG 2025/03/17 19:00:15 charge power: 0W
[lp-1 ] DEBUG 2025/03/17 19:00:15 charge power: 4W
[lp-1 ] DEBUG 2025/03/17 19:00:16 charge currents: [0.07 0.01 0.01]A
[site ] DEBUG 2025/03/17 19:00:16 grid power: -0W
[site ] DEBUG 2025/03/17 19:00:16 grid currents: [-0.207 0.51 -0.321]A
[site ] DEBUG 2025/03/17 19:00:16 battery 1 power: 486W
[site ] DEBUG 2025/03/17 19:00:16 battery 1 soc: 95%
[site ] DEBUG 2025/03/17 19:00:16 pv 1 power: 0W
[site ] DEBUG 2025/03/17 19:00:16 site power: 586W
[lp-2 ] DEBUG 2025/03/17 19:00:16 charger status: A
[site ] DEBUG 2025/03/17 19:00:16 solar forecast: accumulated 0.011kWh, produced 0.000kWh, scale 0.000
[site ] DEBUG 2025/03/17 19:00:45 ----
[lp-2 ] DEBUG 2025/03/17 19:00:45 charge power: 0W
[lp-1 ] DEBUG 2025/03/17 19:00:45 charge power: 4W
[lp-1 ] DEBUG 2025/03/17 19:00:46 charge currents: [0.07 0.01 0.01]A
[site ] DEBUG 2025/03/17 19:00:46 grid power: -0W
[site ] DEBUG 2025/03/17 19:00:46 pv 1 power: 0W
[site ] DEBUG 2025/03/17 19:00:46 battery 1 power: 496W
[site ] DEBUG 2025/03/17 19:00:46 grid currents: [-0.232 0.463 -0.346]A
[site ] DEBUG 2025/03/17 19:00:46 battery 1 soc: 95%
[site ] DEBUG 2025/03/17 19:00:46 site power: 596W
[lp-1 ] DEBUG 2025/03/17 19:00:46 charge voltages: [239 238 238]V
[lp-1 ] DEBUG 2025/03/17 19:00:46 detected connected phases: 3p
[lp-1 ] DEBUG 2025/03/17 19:00:46 charge total import: 582.798kWh
[lp-1 ] DEBUG 2025/03/17 19:00:46 charger status: B
[site ] DEBUG 2025/03/17 19:00:46 solar forecast: accumulated 0.011kWh, produced 0.000kWh, scale 0.000
[site ] DEBUG 2025/03/17 19:01:15 ----
[lp-2 ] DEBUG 2025/03/17 19:01:15 charge power: 0W
[lp-1 ] DEBUG 2025/03/17 19:01:16 charge power: 4W
[lp-1 ] DEBUG 2025/03/17 19:01:16 charge currents: [0.07 0.01 0.01]A
[site ] DEBUG 2025/03/17 19:01:16 grid power: 42W
[site ] DEBUG 2025/03/17 19:01:16 grid currents: [-0.208 0.542 -0.325]A
[site ] DEBUG 2025/03/17 19:01:16 pv 1 power: 0W
[site ] DEBUG 2025/03/17 19:01:16 battery 1 power: 460W
[site ] DEBUG 2025/03/17 19:01:16 battery 1 soc: 95%
[site ] DEBUG 2025/03/17 19:01:16 site power: 602W
[lp-2 ] DEBUG 2025/03/17 19:01:16 charger status: A
[site ] DEBUG 2025/03/17 19:01:16 solar forecast: accumulated 0.011kWh, produced 0.000kWh, scale 0.000
[site ] DEBUG 2025/03/17 19:01:45 ----
[lp-2 ] DEBUG 2025/03/17 19:01:45 charge power: 0W
[lp-1 ] DEBUG 2025/03/17 19:01:46 charge power: 4W
[lp-1 ] DEBUG 2025/03/17 19:01:46 charge currents: [0.07 0.01 0.01]A
[site ] DEBUG 2025/03/17 19:01:46 grid power: -0W
[site ] DEBUG 2025/03/17 19:01:46 pv 1 power: 0W
[site ] DEBUG 2025/03/17 19:01:46 grid currents: [-0.207 0.568 -0.317]A
[site ] DEBUG 2025/03/17 19:01:46 battery 1 power: 483W
[site ] DEBUG 2025/03/17 19:01:46 battery 1 soc: 95%
[site ] DEBUG 2025/03/17 19:01:46 site power: 583W
[lp-1 ] DEBUG 2025/03/17 19:01:46 charge voltages: [238 238 238]V
[lp-1 ] DEBUG 2025/03/17 19:01:46 detected connected phases: 3p
[lp-1 ] DEBUG 2025/03/17 19:01:46 charge total import: 582.798kWh
[lp-1 ] DEBUG 2025/03/17 19:01:46 charger status: B
[site ] DEBUG 2025/03/17 19:01:46 solar forecast: accumulated 0.012kWh, produced 0.000kWh, scale 0.000
[site ] DEBUG 2025/03/17 19:02:15 ----
[lp-2 ] DEBUG 2025/03/17 19:02:15 charge power: 0W
[lp-1 ] DEBUG 2025/03/17 19:02:16 charge power: 4W
[lp-1 ] DEBUG 2025/03/17 19:02:16 charge currents: [0.07 0.01 0.01]A
[site ] DEBUG 2025/03/17 19:02:16 grid power: -0W
[site ] DEBUG 2025/03/17 19:02:16 grid currents: [-0.232 0.496 -0.346]A
[site ] DEBUG 2025/03/17 19:02:16 pv 1 power: 0W
[site ] DEBUG 2025/03/17 19:02:16 battery 1 power: 496W
[site ] DEBUG 2025/03/17 19:02:16 battery 1 soc: 95%
[site ] DEBUG 2025/03/17 19:02:16 site power: 596W
[lp-2 ] DEBUG 2025/03/17 19:02:16 charger status: A
[site ] DEBUG 2025/03/17 19:02:16 solar forecast: accumulated 0.012kWh, produced 0.000kWh, scale 0.000
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.201.0  nightly nicht geprüft

naltatis, 2025-03-17:
Hi @jove01, ich kann das nicht reproduzieren. Bitte schreib noch mal Schritt für Schritt was du wo einstellst und wo beobachtest.

jove01, 2025-03-17:
1. 

[image]

2.

[image]

3.

[image]

4.

[image]

5.

[image]

6.

[image]

`main ] INFO 2025/03/17 21:04:07 evcc 0.201.0
[main ] INFO 2025/03/17 21:04:07 using config file: /config/evcc.yaml
[db ] INFO 2025/03/17 21:04:07 using sqlite database: /data/evcc.db
[mqtt ] INFO 2025/03/17 21:04:07 connecting evcc_mqtt at tcp://localhost:1883
[mqtt ] DEBUG 2025/03/17 21:04:07 tcp://localhost:1883 connected
[main ] INFO 2025/03/17 21:04:07 listening at :7070
[tapo ] DEBUG 2025/03/17 21:04:07 SMART.TAPOPLUG P110 connected (fw:1.3.1 Build 240621 Rel.162048,hw:1.0,mac:AC-15-A2-DD-67-2C)
[lp-1 ] DEBUG 2025/03/17 21:04:07 set title: Wallbox
[lp-1 ] DEBUG 2025/03/17 21:04:07 set priority: 0
[lp-1 ] DEBUG 2025/03/17 21:04:07 set smart cost limit: 0.0
[lp-1 ] DEBUG 2025/03/17 21:04:07 set thresholds: {Enable:{Delay:1m0s Threshold:0} Disable:{Delay:3m0s Threshold:0}}
[lp-1 ] DEBUG 2025/03/17 21:04:07 set plan energy: 0kWh @ 0001-01-01 00:53:28 +0053 LMT
[lp-1 ] DEBUG 2025/03/17 21:04:07 set session energy limit: 0
[lp-1 ] DEBUG 2025/03/17 21:04:07 set session soc limit: 80
[lp-1 ] DEBUG 2025/03/17 21:04:07 set soc config: {Poll:{Mode:charging Interval:1h0m0s} Estimate:0xc0010944ec}
[lp-1 ] DEBUG 2025/03/17 21:04:07 set default mode: off
[lp-1 ] DEBUG 2025/03/17 21:04:07 set phases: 3
[lp-2 ] DEBUG 2025/03/17 21:04:07 set title: Steckdose
[lp-2 ] DEBUG 2025/03/17 21:04:07 set priority: 0
[lp-2 ] DEBUG 2025/03/17 21:04:07 set smart cost limit: <nil>
[lp-2 ] DEBUG 2025/03/17 21:04:07 set thresholds: {Enable:{Delay:1m0s Threshold:0} Disable:{Delay:3m0s Threshold:0}}
[lp-2 ] DEBUG 2025/03/17 21:04:07 set plan energy: 0kWh @ 0001-01-01 00:53:28 +0053 LMT
[lp-2 ] DEBUG 2025/03/17 21:04:07 set session energy limit: 0
[lp-2 ] DEBUG 2025/03/17 21:04:07 set session soc limit: 0
[lp-2 ] DEBUG 2025/03/17 21:04:07 set soc config: {Poll:{Mode:charging Interval:1h0m0s} Estimate:0xc001095a3c}
[lp-2 ] DEBUG 2025/03/17 21:04:07 set default mode: off
[lp-2 ] DEBUG 2025/03/17 21:04:07 set phases: 1
[lp-1 ] DEBUG 2025/03/17 21:04:07 charge total import: 582.804kWh
[site ] INFO 2025/03/17 21:04:08 site config:
[site ] INFO 2025/03/17 21:04:08 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2025/03/17 21:04:08 grid: power ✓ energy ✓ currents ✓
[site ] INFO 2025/03/17 21:04:08 pv 1: power ✓ energy ✓ currents ✗
[site ] INFO 2025/03/17 21:04:08 battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✓
[site ] INFO 2025/03/17 21:04:08 vehicles:
[site ] INFO 2025/03/17 21:04:08 vehicle 1: range ✓ finish ✗ status ✓ climate ✗ wakeup ✗
[site ] INFO 2025/03/17 21:04:08 vehicle 2: range ✓ finish ✗ status ✓ climate ✓ wakeup ✓
[site ] INFO 2025/03/17 21:04:08 tariffs:
[site ] INFO 2025/03/17 21:04:08 grid: priceforecast
[site ] INFO 2025/03/17 21:04:08 feed-in: ✗
[site ] INFO 2025/03/17 21:04:08 co2: ✗
[site ] INFO 2025/03/17 21:04:08 solar: ✓
[lp-1 ] INFO 2025/03/17 21:04:08 loadpoint 1:
[lp-1 ] INFO 2025/03/17 21:04:08 mode: off
[lp-1 ] INFO 2025/03/17 21:04:08 charger: power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1 ] INFO 2025/03/17 21:04:08 meters: charge ✓
[lp-1 ] INFO 2025/03/17 21:04:08 charge: power ✓ energy ✓ currents ✓
[... 58 lines cut ...]
[site ] DEBUG 2025/03/17 21:05:09 battery 1 power: 1363W
[site ] DEBUG 2025/03/17 21:05:09 battery 1 soc: 70%
[site ] DEBUG 2025/03/17 21:05:09 site power: 1463W
[lp-1 ] DEBUG 2025/03/17 21:05:09 charge voltages: [240 239 239]V
[lp-1 ] DEBUG 2025/03/17 21:05:09 detected connected phases: 3p
[lp-1 ] DEBUG 2025/03/17 21:05:09 charge total import: 582.804kWh
[lp-1 ] DEBUG 2025/03/17 21:05:09 charger status: B
[lp-1 ] DEBUG 2025/03/17 21:05:09 vehicle api refresh
[main ] DEBUG 2025/03/17 21:05:09 vehicle status: B (Explorer)
[main ] DEBUG 2025/03/17 21:05:09 vehicle status: B (Mokka E)
[main ] WARN 2025/03/17 21:05:09 vehicle status: >1 matches, giving up
[site ] DEBUG 2025/03/17 21:05:39 ----
[lp-2 ] DEBUG 2025/03/17 21:05:39 charge power: 0W
[lp-1 ] DEBUG 2025/03/17 21:05:39 charge power: 4W
[lp-1 ] DEBUG 2025/03/17 21:05:39 charge currents: [0.07 0.01 0.01]A
[site ] DEBUG 2025/03/17 21:05:39 grid power: -0W
[site ] DEBUG 2025/03/17 21:05:39 pv 1 power: 0W
[site ] DEBUG 2025/03/17 21:05:39 battery 1 power: 1361W
[site ] DEBUG 2025/03/17 21:05:39 grid currents: [-1.36 2.8 -1.45]A
[site ] DEBUG 2025/03/17 21:05:39 battery 1 soc: 70%
[site ] DEBUG 2025/03/17 21:05:39 site power: 1461W
[lp-2 ] DEBUG 2025/03/17 21:05:39 charger status: A
`

Ich hoffe das hilft!

naltatis, 2025-03-17:
Danke. Konnte es nachstellen und fixen ☝