evcc-io/evcc #19887: PV Energie Fronius Solar API ist um Faktor 1000 zu viel

phrozen77, 2025-03-17:
### Describe the bug

Da hat sich wohl ein kleiner Fehler eingeschlichen:

evcc sagt 36TWh, Wechselrichter sagt (richtigerweise) 36MWh...

[image]

[image]

Ja, schön wärs 😆 

evcc 0.201.0

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
irrelevant
```

### Log details

```text
irrelevant
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

stan23, 2025-03-17:
Ist bei mir auch so.
Ob es daher kommt?

https://github.com/evcc-io/evcc/pull/19192#issuecomment-2729096021

andig, 2025-03-17:
Gerne PR

closed 2025-03-18 by andig