evcc-io/evcc #19905: Änderung des suspec-inverter.yaml, oder Einführung eines sunspec-smartmeter.yaml

HanseKa, 2025-03-18:
Ich betreibe zwei PV-Anlagen, PV neu, und PV alt (bestehend u. a. aus 3 SMA SunnyBoy Wechselrichtern). PV neu ist mit einem Fronius Symo Gen24 6.0 plus und einem Fronius Smartmeter TS65A-3 am Einspeisepunkt ausgestattet. PV alt ist mit einem zweiten TS65A-3 per modbus mit dem Symo Gen24 verbunden. Diese PV alt kann ich aber weder mit dem fronius-gen24.yaml, noch mit dem sunspec-inverter.yaml in EVCC über die UI konfigurieren. 

Für die Lösung des Problems habe ich zwei Vorschläge:

1. ergänzen des templates sunspec-inverter.yaml um folgende zwei Einträge: 
    unter {{- if eq .usage "pv" }}
              power 
                  value
                     - 213:W
     und

              energy
                  value
                      - 213:TotWhExp

2. Einführung eines neuen sunspec-smartmeter.yaml mit folgendem Inhalt:
   
template: sunspec-smartmeter
products:
  - description:
      de: SunSpec Smartmeter
      en: SunSpec Smartmeter
group: generic
params:
  - name: usage
    choice: ["grid", "pv"]
  - name: modbus
    choice: ["tcpip", "rs485"]
  - name: capacity
    advanced: true
render: |
  type: custom
  {{- if eq .usage "grid" }}
  # sunspec model 20x (int+sf)/ 21x (float) meter
  power:
    source: sunspec
    {{- include "modbus" . | indent 2 }}
    value:
      - 201:W
      - 211:W
      - 203:W
      - 213:W
  energy:
    source: sunspec
    {{- include "modbus" . | indent 2 }}
    value:
      - 201:TotWhImp
      - 211:TotWhImp
      - 203:TotWhImp
      - 213:TotWhImp
    scale: 0.001
  currents:
    - source: sunspec
      {{- include "modbus" . | indent 4 }}
      value:
        - 201:AphA
        - 211:AphA
        - 203:AphA
        - 213:AphA
    - source: sunspec
      {{- include "modbus" . | indent 4 }}
      value:
        - 203:AphB
        - 213:AphB
    - source: sunspec
      {{- include "modbus" . | indent 4 }}
      value:
        - 203:AphC
        - 213:AphC
  voltages:
    - source: sunspec
      {{- include "modbus" . | indent 4 }}
      value:
        - 201:PhVphA
        - 211:PhVphA
        - 203:PhVphA
        - 213:PhVphA
    - source: sunspec
      {{- include "modbus" . | indent 4 }}
      value:
        - 203:PhVphB
        - 213:PhVphB
    - source: sunspec
      {{- include "modbus" . | indent 4 }}
      value:
        - 203:PhVphC
        - 213:PhVphC
  powers:
    - source: sunspec
      {{- include "modbus" . | indent 4 }}
      value:
        - 201:WphA
        - 211:WphA
        - 203:WphA
        - 213:WphA
    - source: sunspec
      {{- include "modbus" . | indent 4 }}
      value:
        - 203:WphB
        - 213:WphB
    - source: sunspec
      {{- include "modbus" . | indent 4 }}
      value:
        - 203:WphC
        - 213:WphC
  {{- end }}
  {{- if eq .usage "pv" }}
  power:
    source: sunspec
    {{- include "modbus" . | indent 2 }}
    value:
      - 101:W
      - 111:W
      - 103:W
      - 113:W
      - 213:W
  energy:
    source: sunspec
    {{- include "modbus" . | indent 2 }}
    value:
      - 101:TotWhExp
      - 111:TotWhExp
      - 103:TotWhExp
      - 113:TotWhExp
      - 213:TotWhExp
    scale: 0.001
  {{- end }}

Bisher habe ich das Problem mit Hilfe der evcc.yaml gelöst. PV alt ist mit dem template sunspec-inverter.yaml eingebunden, allerdings mit usage:grid und name:pv1. Diese Möglichkeit gibt es aber, wie ich eingangs schon erwähnt habe, bei der Konfiguration über die UI nicht. Oder hat jemand einen Vorschlag oder Idee, ohne zusätzliche Hardware, die PV alt sauber einzubinden?

benesolar hat dahingehend schon einmal ein issue oder einen pull request gestartet (finde ich gerade nicht). Wurde aber, mangels Notwendigkeit, geclosed.

Es wäre top wenn meine Vorschläge nochmals geprüft werden würden.

Ich habe mir übrigens mit diesem sunspec-smartmeter.yaml ein binary gebuildet. EVCC funktioniert damit wie gewünscht.

andig, 2025-03-19:
> Diese PV alt kann ich aber weder mit dem fronius-gen24.yaml, noch mit dem sunspec-inverter.yaml in EVCC über die UI konfigurieren.

Was hat das denn mit einem zusätzlichen Energiemesswert zu tun? Mir ist nicht klar, was da nicht geht und was du überhaupt konfigurieren willst. Eine zweite PV? Batterie? Einen zweiten Netzzähler???

HanseKa, 2025-03-19:
Ich möchte eine zweite PV Anlage (PV alt) konfigurieren!  Die Energiewerte dieser PV -Anlage liefert das zweite Fronius Smartmeter T65A-3, mit der modbus Adresse 201! Deshalb der Wunsch der Änderung, bzw. der Einführung eines sunspec-smartmeter.yaml.Von meinem iPhone gesendet

andig, 2025-03-19:
> 213:W

Ah die 2xx Models gibts nur unter `grid`. Das passt bei Dir nicht weil es für `pv` nicht angezeigt wird.

@premultiply weiteres Template? Ich hadere allerdings mit der Energierichtung. Bei grid haben wir `TotWhImp`.

HanseKa, 2025-03-19:
Die Energierichtung passt mit TotWhExp schon, weil ich bei Verwendung pv ja einspeise, bzw. liefere. Meine „PV alt“ hat aktuell 10.444,598 kWh produziert seit das Fronius Smartmeter eingebaut wurde (Jan 2024)! Wieso das Smartmeter auch einen Bezug von 0,119 kWh bei TotWhImp anzeigt, weiss wahrscheinlich nur der Hersteller. 

Hier die Abfrage der ID201:

evcc sunspec 192.168.2.153:502 -i 201
[main  ] INFO 2025/03/19 16:55:02 evcc 0.200.9
--------- Model 1 common ---------
Mn            Fronius    string
Md   Smart Meter TS 6    string
Opt          PV alt H    string
Vr                1.4    string
SN         3682862508    string
DA             201.00    uint16
Pad            0x00d5    pad
--------- Model 213 ac_meter ---------
A                           -8.42    float32
AphA                        -0.94    float32
AphB                       -3.363    float32
AphC                       -4.117    float32
PhV                     233.26666    float32
PhVphA                      234.3    float32
PhVphB                      230.7    float32
PhVphC                      234.8    float32
PPV                           404    float32
PPVphAB                     402.7    float32
PPVphBC                     403.1    float32
PPVphCA                     406.2    float32
Hz                           49.9    float32
W                          1742.3    float32
WphA                         16.8    float32
WphB                          748    float32
WphC                        977.4    float32
VA                         1918.1    float32
VAphA                       186.7    float32
VAphB                       767.1    float32
VAphC                       964.3    float32
VAR                          -410    float32
VARphA                       -186    float32
VARphB                     -164.6    float32
VARphC                      -59.3    float32
PF                         -0.973    float32
PFphA                      -0.088    float32
PFphB                      -0.977    float32
PFphC                      -0.998    float32
TotWhExp            1.0444598e+07    float32
TotWhExpPhA                   n/a    float32
TotWhExpPhB                   n/a    float32
TotWhExpPhC                   n/a    float32
TotWhImp                      119    float32
TotWhImpPhA                   n/a    float32
TotWhImpPhB                   n/a    float32
TotWhImpPhC                   n/a    float32
TotVAhExp                     n/a    float32
TotVAhExpPhA                  n/a    float32
TotVAhExpPhB                  n/a    float32
TotVAhExpPhC                  n/a    float32
TotVAhImp                     n/a    float32
TotVAhImpPhA                  n/a    float32
TotVAhImpPhB                  n/a    float32
TotVAhImpPhC                  n/a    float32
TotVArhImpQ1                  n/a    float32
TotVArhImpQ1phA               n/a    float32
TotVArhImpQ1phB               n/a    float32
TotVArhImpQ1phC               n/a    float32
TotVArhImpQ2                  n/a    float32
TotVArhImpQ2phA               n/a    float32
TotVArhImpQ2phB               n/a    float32
TotVArhImpQ2phC               n/a    float32
TotVArhExpQ3                  n/a    float32
TotVArhExpQ3phA               n/a    float32
TotVArhExpQ3phB               n/a    float32
TotVArhExpQ3phC               n/a    float32
TotVArhExpQ4                  n/a    float32
TotVArhExpQ4phA               n/a    float32
TotVArhExpQ4phB               n/a    float32
TotVArhExpQ4phC               n/a    float32
Evt                    0x00000000    bitfield32

HanseKa, 2025-03-29:
Ich bin anscheinend nicht der einzige mit diesem Problem! https://github.com/evcc-io/evcc/discussions/19526#discussioncomment-12640662

HanseKa, 2025-04-07:
Ich möchte noch einmal die Notwendigkeit der Änderung, bzw. der Neueinführung verdeutlichen.

Die Fronius Smartmeter können, z.B. ivm Fronius Symo Gen24 PLus, unterschiedlich konfiguriert werden!
Dies beginnt schon damit, wo das primäre Smartmeter verbaut ist. Sitzt es am Einspeisepunkt, oder wird es im Verbrauchszweig installiert! Dieses Smartmeter bekommt intern die Modbus Adr. 1 oder auch 200.

Alle weiteren Smartmeter können dann entweder als Erzeuger- oder als Verbraucherzähler konfiguriert werden. Modbus Adr. >1 oder >200! Bei Verwendung als Erzeugerzähler liefert das Smartmeter dann die erzeugte Energie eines Wechselrichters, eines AC-Speichers, eines Wechselrichters+AC-Speichers, eines Windrads oder eines Block-Heizkraftwerks (BHKW).

Wird es als Verbraucherzähler eingesetzt kann es den Verbrauch eines Elektrofahrzeugs, einer Wärmepumpe, einer Heizung, einer Pumpe oder einer Weißware aufzeichnen.

Auf diese Weise lassen sich, wie in meinem Fall, auch ohne weitere Installationsarbeiten oder weitere Hardware, bestehende PV-Anlagen in evcc integrieren.

Sollen diese Smartmeter künftig auch über die UI konfigurierbar sein, führt kein Weg an diesem Topic vorbei. Lasse mich gerne eines besseren belehren, wenn dem nicht so ist!

andig, 2025-04-08:
@premultiply klingt so als bräuchten wir ein `sunspec-meter` dass für `pv` und `grid` geeignet wäre.

@HanseKa mangels andersweitigem Feedback: wärst Du so nett einen PR zu machen? Mit dem inverter sollten wir es m.E. nicht mischen.

benesolar, 2025-04-08:
Ich hätte hier bereits einen Pull Request vorgeschlagen: https://github.com/evcc-io/evcc/pull/19624

closed 2025-04-21 by premultiply