evcc-io/evcc #19907: EVCC startet nicht bei Fehlern (Passwort falsch)

Cyrelion, 2025-03-18:
### Describe the bug

Ich habe bei Inexogy (früher Discovergy) mein Passwort geändert, nach einem neustart von HA / EVCC klappt der start aber nicht mehr da er bei dem Fehler:

[main  ] FATAL 2025/03/18 12:01:33 meter [db:20] cannot create meter 'db:20': cannot create meter type 'template': cannot create meter type 'discovergy': unexpected status: 401 (Unauthorized)

fest hängt und ich somit auch nicht dass Passwort im Webinterface ändern kann.

### Steps to reproduce

1. Inexogy Zähler konfigurieren
2. Passwort ändern
3. Neu starten
...

### Configuration details

```yaml
evcc.yaml fast leer da alles in db konfiguriert.
```

### Log details

```text
[main  ] FATAL 2025/03/18 12:01:33 meter [db:20] cannot create meter 'db:20': cannot create meter type 'template': cannot create meter type 'discovergy': unexpected status: 401 (Unauthorized)
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

Cyrelion, 2025-03-18:
Meine Lösung war nun über den HA container auf die DB zuzugreifen und das Passwort manuell in der SQLite zu ändern. Das ist allerdings nicht für jeden Anwender möglich :-(

andig, 2025-03-18:
Ja, derzeit ist das noch so. Wir arbeiten dran.

closed 2025-03-18 by andig

picki72, 2025-03-18:
Same here caused by unavailability of sensor data from the midbus integration of my Solaredge inverter:

[main  ] FATAL 2025/03/18 15:53:43 meter [db:5] cannot create meter 'db:5': cannot create meter type 'template': cannot create meter type 'custom': power: sunspec model not found: [{203 0 W}]