evcc-io/evcc #19909: seit Update kein 3-phasiges Laden mehr mit cFos Wallbox möglich

bastiitsab, 2025-03-18:
Ich hab komischerweise das gleiche Problem mit eienr CFOS Wallbox. Seit dem Update geht keine Umschaltung auf 3 Phasen mehr. Bis auf das Update gab es keine Änderungen.

[image]

_Originally posted by @bastiitsab in https://github.com/evcc-io/evcc/discussions/19899#discussioncomment-12537344_

GHMattZ, 2025-03-18:
- Habe das gleiche Problem mit einer GOe Wallbox (22kW flex, V3). Mit 0201.0 schaltet er nicht auf 3 Phasen um. Auch nicht wenn ich ihn zu 3 Phasen zwinge und Schnellladen mache. Downgrade auf 0200.9 und es funktioniert wieder. Fahrzeug ist ein Audi etron, falls es relevant wäre.

bastiitsab, 2025-03-18:
> * Downgrade auf 0200.9 und es funktioniert wieder. 

Kann ich bestätigen

HanseKa, 2025-03-18:
Das gleiche mit einer openWB pro! Lädt nur mehr einphasig! Automat. Umschaltung hat bisher tadellos funktioniert. Auch wenn ich manuell auf dreiphasig einstelle, lädt die Wallbox nur einphasig!

andig, 2025-03-18:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

bastiitsab, 2025-03-18:
@andig Das Issue wurde aus einer Discussion eröffnet, dadurch ist es nicht durch den normalen Prozess gegangen. Auch wenn das Issue nicht der Formalie entsprach, handelt es sich hier offensichtlich um kein Einzelschicksal wofür unbedingt meine Logs nötig wären. Da ich ein Rollback gemacht habe, kann ich derzeit keine Logs liefern. 
Vielleicht kann @GHMattZ , @Jonnyblue85 @HanseKa logs hierfür liefern und ein neues Issue öffnen, ich kann es auch nicht reopen.

GHMattZ, 2025-03-18:
Hab das ganze auf Home Assistant laufen und bin jetzt auch wieder bei der alten Version. Kenne mich mit der ganzen Materie nicht genug aus, um die Infos liefern zu können. Meine Config ist seit  ca. 3 Monaten unverändert und lief einwandfrei bis zum letzten Update.
Wenn es irgendwo eine einfache Anleitung gibt was ich machen soll/muss/kann, kann ich versuchen es zu liefern.

TobiasHuber1980, 2025-03-18:
> Das gleiche mit einer openWB pro! Lädt nur mehr einphasig! Automat. Umschaltung hat bisher tadellos funktioniert. Auch wenn ich manuell auf dreiphasig einstelle, lädt die Wallbox nur einphasig!

kann ich bei meiner open wb pro nicht nachvollziehen. gestartet 3-phasig (schnell), runter 1-phasig (min-pv) und wieder auf 3-phasig (schnell); ohne probleme.

andig, 2025-03-19:
> handelt es sich hier offensichtlich um kein Einzelschicksal wofür unbedingt meine Logs nötig wären

Ob dein Log ist egal. Kein Log wird nicht bearbeitet. Wir können NICHT hellsehen.

andig, 2025-03-19:
Da mittlerweile nich ein Issue dazu gibt geschlossen.

closed 2025-03-19 by andig