evcc-io/evcc #19914: Easee Home schaltet ständig einfach ab

Schneeschuher, 2025-03-18:
### Describe the bug

Hallo Zusammen,

ich habe ständig das Problem das meine Easee Home Wallbox das Laden nach ein paar Minuten abbricht.
Sie wartet dann auf das Fahrzeug, wie im Screenshot zu sehen.
Am anderen Ende steckt ein VW ID3. Bisher hatte ich null Probleme mit Laden.

[image]

### Steps to reproduce

Fahrzeug anstecken und auf schnell laden klicken...
Bei jedem Laden.

### Configuration details

```yaml
Meine yaml unter /addon_configs/49686a9f_evcc/evcc.yaml ist komischerweise leer.
```

### Log details

```text
011]A
[site ] DEBUG 2025/03/18 12:37:29 grid power: -4984W
[site ] DEBUG 2025/03/18 12:37:29 grid currents: [0 0 0]A
[site ] DEBUG 2025/03/18 12:37:29 pv 1 power: 5401W
[site ] DEBUG 2025/03/18 12:37:29 pv 2 power: 3115W
[site ] DEBUG 2025/03/18 12:37:29 pv power: 8516W
[site ] DEBUG 2025/03/18 12:37:29 battery 1 power: 0W
[site ] DEBUG 2025/03/18 12:37:29 battery 1 soc: 100%
[site ] DEBUG 2025/03/18 12:37:29 site power: -4984W
[lp-1 ] DEBUG 2025/03/18 12:37:29 charge total import: 3959.486kWh
[lp-1 ] DEBUG 2025/03/18 12:37:29 charger status: B
[lp-1 ] INFO 2025/03/18 12:37:29 stop charging <-
[lp-1 ] DEBUG 2025/03/18 12:37:29 charge total import: 3959.486kWh
[lp-1 ] DEBUG 2025/03/18 12:37:30 vehicle soc: 44%
[lp-1 ] DEBUG 2025/03/18 12:37:30 vehicle soc limit: 80%
[lp-1 ] DEBUG 2025/03/18 12:37:30 vehicle range: 191km
[site ] DEBUG 2025/03/18 12:37:30 solar forecast: accumulated 6.469kWh, produced 16.200kWh, scale 2.504
[site ] DEBUG 2025/03/18 12:37:59 ----
[lp-1 ] DEBUG 2025/03/18 12:37:59 charge power: 0W
[lp-1 ] DEBUG 2025/03/18 12:37:59 charge currents: [0.009 0.01 0.011]A
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.201.0

andig, 2025-03-18:
Was sagt denn der Easee Support?

/cc @GrimmiMeloni

Schneeschuher, 2025-03-18:
@andig Den habe ich noch nicht eingebunden, da die Box ja vorher ohne evcc nie dieses Problem hatte.

GrimmiMeloni, 2025-03-18:
Bitte mal ein debug log mit Easee auf `trace` davon teilen.

Schneeschuher, 2025-03-18:
@GrimmiMeloni Vielen Dank Dir... Meinst du das Log?

[evcc-20250318-184003-trace.log](https://github.com/user-attachments/files/19324501/evcc-20250318-184003-trace.log)

Schneeschuher, 2025-03-18:
@andig & @GrimmiMeloni Wenn ich über evcc die Box ausschalte und dann erneut in den Schnellmodus wechsle, scheint es zu funktionieren. D.h. evcc wechselt nicht selbstständig zur entsprechenden Ladequelle. Sondern wenn diese sich ändert, geht die Box in den Wartezustand, wenn ich dann auf Aus und wieder anschalte, startet evcc korrekt.

GrimmiMeloni, 2025-03-18:
> [@GrimmiMeloni](https://github.com/GrimmiMeloni) Vielen Dank Dir... Meinst du das Log?

Ja, aber ein vollständiges. Dieses Log enthält u.a. nicht die Logzeilen mit den API Calls, und ist daher wertlos. Wenn Du kein ungefiltertes Log öffentlich teilen möchtest, kannst Du es mir auch direkt an grund.heuboden0v@icloud.com senden.

Schneeschuher, 2025-03-19:
@GrimmiMeloni Schick ich dir heute zu... Vielen Dank...

Schneeschuher, 2025-03-19:
@GrimmiMeloni Vielen Dank dir für Deine Mühe. Als technischer Betreiber war bei Easee, sehr versteckt, Tibber eingetragen. Das könnte es gewesen sein.

GrimmiMeloni, 2025-03-19:
Ja, das wird es sein. Hatten wir schon mal an anderer Stelle.

Schneeschuher, 2025-03-20:
Hallo,

darf ich dir nochmal ein aktuelles Log schicken.
Inzwischen sind alle Störfaktoren raus.
Technischer Betreiber nun Easee. Es hängt sonst keinerlei App etc. dran.
Leider besteht das Problem immer noch.
Kannst du mal auf das aktuelle Log schauen, ob du noch einen Hinweis siehst.

Für Rückfragen stehe ich gerne zur Verfügung.

Viele Grüße
Alex Schwarz

Schneeschuher, 2025-03-20:
> Klaro.

Vielen Dank dir...
Ich hatte es in der letzten Mail schon angehängt. Hier nochmal anbei.
Ich glaube dem Problem näher zu kommen.
Es könnte sein das Tibber Pulse der Verursacher ist.
Ich bekomme relativ oft die Meldung "grid power: outdated".

Viele Grüße
Alex Schwarz

GrimmiMeloni, 2025-03-20:
@Schneeschuher auch das neue Log zeigt das selbe Bild. Der Charger quittiert weiterhin Kommandos ohne das evcc etwas gesendet hat - teilweise ist die letzte Aktion von evcc 5 Minuten oder länger her. 

Du kannst ja mal bei Easee anfragen, ob sie wissen welcher Client die folgenden CommandResponses ausgelöst hat:

```
[easee ] TRACE 2025/03/20 12:54:42 CommandResponse XXXXXXXX: {SerialNumber:XXXXXXXX ID:29 Timestamp:2025-03-20 11:54:41.489097 +0000 UTC DeliveredAt:2025-03-20 11:54:39.985 +0000 UTC WasAccepted:true ResultCode:0 Comment: Ticks:638780684814890970}
[easee ] TRACE 2025/03/20 12:59:55 CommandResponse XXXXXXXX: {SerialNumber:XXXXXXXX ID:29 Timestamp:2025-03-20 11:59:54.702221 +0000 UTC DeliveredAt:2025-03-20 11:59:53.233 +0000 UTC WasAccepted:true ResultCode:0 Comment: Ticks:638780687947022210}
[easee ] TRACE 2025/03/20 13:48:10 CommandResponse XXXXXXXX: {SerialNumber:XXXXXXXX ID:29 Timestamp:2025-03-20 12:48:09.382569 +0000 UTC DeliveredAt:2025-03-20 12:48:07.82 +0000 UTC WasAccepted:true ResultCode:0 Comment: Ticks:638780716893825690}
```