evcc-io/evcc #19917: Add "targetSocTolerance" per vehicle

grogi, 2025-03-18:
**Is your feature request related to a problem? Please describe.**

The BMS in my car (Mini Electric) is a bit imprecise and changes its mind when charging stops.

Let's say I have the Target SOC for the car set to 80%. The BMS, while charging is progress, might report SOC at 80%. This triggers the EVCC to stop charging, because the desired SOC is reached. 

Unfortunately, the moment the charging stops, the reported SOC might drop a bit, to 76% for instance. This triggers the EVCC to restart charging. 

**Describe the solution you'd like**

I would like to be able to specify the `targetSocTolerance`. The charging will only be started if the `reportedSoc` is below `targetSoc` - `targetSocTolerance`. 

For instance, if I have specified the `targetSocTolerance` to 5 and 'targetSoc' to 80, the charging only starts when reported SOC is below 75%. Once started, it will however not stop until the 80% is reached. Now, if the BMS reports 76% after the charging stopped, this will not resume charging, as the value is within the tolerance. 

The `targetSocTolerance` value defaults to 0 - this value doesn't change the current behavior at all if it is not specified.

**Describe alternatives you've considered**

I have used Home Assistant to dynamically manage the `targetSoc`. This kind-of works, but starts to be a bit tedious now with the additional solar panels... 

**Additional context**

andig, 2025-03-19:
I don‘t think we‘d want to have this workaround in evcc. Worst case charging restarts a few times once a day- that should not be an issue.

andig, 2025-03-19:
You can also try if the estimatoe can help in this case.

closed 2025-03-19 by andig

grogi, 2025-03-19:
> Is it really the car reporting 80% or the evcc estimator? A log file would show.

Yes, it is the car...

I don't have a EVCC log at hand, but values history chart from HA. 
At 5:20 the BMW/Mini integration reports 81%, the EVCC is switched to OFF to stop charging. The next pulling cycle car reports 78%. 

[image]