evcc-io/evcc #19924: Umschalten auf 3-phasig funktioniert nach Update auf 0.201.0 nicht mehr

Jonnyblue85, 2025-03-18:
### Describe the bug

Seit dem Update von 0.200.9 auf 0.201.0 ist es so, dass einige Wallboxen nur noch einphasig laden. Die Wallboxen stehen auf EVCC auch auf 1-phasig. Ein umschalten auf Automatisch oder 3-phasig in evcc hat keine Auswirkung. Starten, stoppen, umschalten auf "Schnell" auch nicht. 

Ein umschalten auf automatisch oder 3-phasig an den Wallboxen (in meinem Fall go-e charger) funktioniert, danach funktioniert es wieder. Ob das automatische umschalten klappt konnte ich noch nicht ausprobieren.

Laut https://github.com/evcc-io/evcc/issues/19909 betrifft es auch andere Wallboxhersteller.

Mein Setup: 2x go-e charger, Model Y und Enyaq. Fehler tritt bei beiden auf.

Leider scheint der Fehler nur direkt nach dem Update aufzutauchen, sobald die Wallboxen umgestellt wurden lässt sich der Fehler nicht mehr reproduzieren - da ich aber nicht alleine bin, scheint es schon ein Problem zu sein.

### Steps to reproduce

1. PV laden startet automatisch
2. Es ist mehr als 3,6kW verfügbar
3. Auto sollte auf 3-phasig umschalten, macht es aber nicht
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxx

interval: 15s # control cycle interval

sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

# mqtt message broker
mqtt:
  broker: 192.168.11.21:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  user: xxx
  password: xxx

meters:
- type: template
  template: sonnenbatterie 
  usage: grid  
  host: 192.168.11.15  
  port: 8080  
  name: grid1
- type: template
  template: sonnenbatterie 
  usage: pv  
  host: 192.168.11.15  
  port: 8080  
  name: pv-sued
- type: template
  template: sonnenbatterie 
  usage: battery  
  host: 192.168.11.15  
  port: 8080
  token: xxx
  name: battery-sonnen
[... 1135 lines cut ...]
[lp-1  ] DEBUG 2025/03/18 21:50:39 available power -826W < 4140W min 3p threshold
[lp-1  ] DEBUG 2025/03/18 21:50:39 start phase scale1p timer
[lp-1  ] DEBUG 2025/03/18 21:50:39 phase scale1p in 3m0s
[lp-1  ] DEBUG 2025/03/18 21:50:39 pv charge current: 0A = 16A + -17A (11713W @ 3p)
[lp-1  ] DEBUG 2025/03/18 21:50:39 projected site power 8953W >= 0W disable threshold
[lp-1  ] DEBUG 2025/03/18 21:50:39 pv disable timer start: 3m0s
[lp-1  ] DEBUG 2025/03/18 21:50:39 pv disable in 3m0s
[lp-1  ] DEBUG 2025/03/18 21:50:39 max charge current: 6A
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.201.0

andig, 2025-03-19:
Nicht nachvollziehbar. Im Logfile wird 3p geladen.

christian-moeller, 2025-03-19:
Mein Ladeverhalten (auch go-e) ist seit 201.0 auch vermurkst. Gerade lädt das Auto mit 860W, 3phasig, und 3.4kW werden eingespeist. Sowas hatte ich noch nie!

[image]

GHMattZ, 2025-03-19:
Stell dich den Log dazu ein.

andig, 2025-03-19:
> charge currents: [15.9 16 16]A

Eindeutig 3p. Problem leider nicht nachvollziehbar.

Horgul123, 2025-03-19:
Mit der V0.210.0 schaltet er bei mir auch nicht mehr die Phasen um. Hier umstellung von PV 1-phasig auf schnell 3-Phasig im Log. Mit V0.209.9 geht es ohne Probleme. go-e Wallbox

[evcc-20250319-125710-debug.log](https://github.com/user-attachments/files/19341104/evcc-20250319-125710-debug.log)

andig, 2025-03-19:
Aus https://github.com/evcc-io/evcc/issues/19924#issuecomment-2736342019:

> [lp-1  ] DEBUG 2025/03/19 12:43:19 set charge mode: now
> [lp-1  ] DEBUG 2025/03/19 12:43:42 set phases: 0

@naltatis hier hätte ich eigentlich erwartet, dass wir 3p sehen sollten.

In https://github.com/evcc-io/evcc/issues/19924#issuecomment-2736545066 sehe ich gar keine Phasenumschaltung, auch nicht nach 

> [lp-1  ] DEBUG 2025/03/19 12:54:50 set charge mode: now

andig, 2025-03-19:
Gefunden, der Fehler ist hier: https://github.com/evcc-io/evcc/pull/19263/files#diff-c456526d4bd9022a488ce32de208b40d86016e2c40b0c81a2dffe557c186377dR191. Die Ladepunkteigenschaften für 1p3p dürfen erste gesetzt werden nachdem der Charger zugeordnet ist. Das habe ich vertüddelt.

andig, 2025-03-19:
Closed in 49a597a5da11928ad8968f4ea26549620fff3578. Könnt ihr bitte in 30min das Nightly probieren?

jheinitz, 2025-03-19:
Moin @andig ! 

Bei mir war das gestern genau anders herum. Die Sonne ging unter, aber meine Easse hat fleißig mit 3 Phasen und 7A weiter geladen. Ich hätte erwartet, dass dort der Timer los läuft, um von 3p auf 1p zu reduzieren.

Behebt die Korrektur das auch in diese Richtung?

Danke und Grüße

Jonnyblue85, 2025-03-19:
Ich bin auf HA und kann da soweit ich weiß glaube ich die nightly nicht ausprobieren, aber hört sich ja so an als wärs das gewesen. Danke!

closed 2025-03-20 by andig

StefanSchoof, 2025-03-20:
@Jonnyblue85 Gibt es mittlerweile: https://github.com/evcc-io/evcc/discussions/2896#discussioncomment-12207879

PixelFresser, 2025-03-21:
Hi,

ich hatte das selbe Problem und nun das letzte nightly installiert.
Leider schaltet scheint die Umschaltung nach wie vor nicht richtig zu funktionieren.

Die einzige Änderung was hinzu kommt, ist sie Umstellung von der "klassischen" Fronius Wattpliot Anbindung auf die neue OCPP.
Diese hatte ich auch nur durchgeführt da ich eben diese 1p3p Umschaltungsproblematik hatte.

`evcc  | [main  ] INFO 2025/03/21 13:31:46 evcc 0.201.1 (9a18c95f)
evcc  | [main  ] INFO 2025/03/21 13:31:46 using config file: /etc/evcc.yaml
evcc  | [db    ] INFO 2025/03/21 13:31:46 using sqlite database: /root/.evcc/evcc.db
evcc  | [mqtt  ] INFO 2025/03/21 13:31:46 connecting evcc-1173357484 at tcp://192.168.10.15:1883
evcc  | [mqtt  ] DEBUG 2025/03/21 13:31:46 tcp://192.168.10.15:1883 connected
evcc  | [main  ] INFO 2025/03/21 13:31:46 listening at :7070
evcc  | [91001230-1] DEBUG 2025/03/21 13:31:46 waiting for chargepoint: 5m0s
evcc  | [ocpp  ] DEBUG 2025/03/21 13:32:00 charge point connected: 91001230
evcc  | [mini  ] DEBUG 2025/03/21 13:32:02 identity.Login - database token found
evcc  | [91001230-1] DEBUG 2025/03/21 13:32:02 recovered transaction: 1742537866
evcc  | [lp-1  ] DEBUG 2025/03/21 13:32:03 charge total import: 3072.849kWh
evcc  | [site  ] INFO 2025/03/21 13:32:04 site config:
evcc  | [site  ] INFO 2025/03/21 13:32:04   meters:      grid ✓ pv ✓ battery ✓
evcc  | [site  ] INFO 2025/03/21 13:32:04     grid:      power ✓ energy ✓ currents ✓
evcc  | [site  ] INFO 2025/03/21 13:32:04     pv 1:      power ✓ energy ✓ currents ✗
evcc  | [site  ] INFO 2025/03/21 13:32:04     pv 2:      power ✓ energy ✓ currents ✗
evcc  | [site  ] INFO 2025/03/21 13:32:04     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✓
evcc  | [site  ] INFO 2025/03/21 13:32:04   vehicles:
evcc  | [site  ] INFO 2025/03/21 13:32:04     vehicle 1: range ✓ finish ✗ status ✓ climate ✓ wakeup ✓
evcc  | [site  ] INFO 2025/03/21 13:32:04     vehicle 2: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
evcc  | [site  ] INFO 2025/03/21 13:32:04   tariffs:
evcc  | [site  ] INFO 2025/03/21 13:32:04     grid:      priceforecast
evcc  | [site  ] INFO 2025/03/21 13:32:04     feed-in:   pricestatic
evcc  | [site  ] INFO 2025/03/21 13:32:04     co2:       ✗
evcc  | [site  ] INFO 2025/03/21 13:32:04     solar:     ✓
evcc  | [lp-1  ] INFO 2025/03/21 13:32:04 loadpoint 1:
evcc  | [lp-1  ] INFO 2025/03/21 13:32:04   mode:        pv
evcc  | [lp-1  ] INFO 2025/03/21 13:32:04   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
evcc  | [lp-1  ] INFO 2025/03/21 13:32:04   meters:      charge ✓
evcc  | [lp-1  ] INFO 2025/03/21 13:32:04     charge:    power ✓ energy ✓ currents ✓
evcc  | [site  ] DEBUG 2025/03/21 13:32:04 set buffer soc: 95
evcc  | [site  ] DEBUG 2025/03/21 13:32:04 set buffer start soc: 0
evcc  | [site  ] DEBUG 2025/03/21 13:32:04 set priority soc: 95
evcc  | [site  ] DEBUG 2025/03/21 13:32:04 set battery discharge control: true
evcc  | [site  ] DEBUG 2025/03/21 13:32:04 set grid charge limit: 0.2
evcc  | [lp-1  ] DEBUG 2025/03/21 13:32:04 set smart cost limit: 0.1
evcc  | [lp-1  ] DEBUG 2025/03/21 13:32:04 phase timer inactive
evcc  | [lp-1  ] DEBUG 2025/03/21 13:32:04 pv timer inactive
evcc  | [lp-1  ] DEBUG 2025/03/21 13:32:04 max charge current: 1A
evcc  | [site  ] DEBUG 2025/03/21 13:32:04 ----
evcc  | [lp-1  ] DEBUG 2025/03/21 13:32:04 charge power: 3232W
evcc  | [lp-1  ] DEBUG 2025/03/21 13:32:04 charge currents: [14 0 0]A
evcc  | [site  ] DEBUG 2025/03/21 13:32:04 grid power: -6917W
evcc  | [site  ] DEBUG 2025/03/21 13:32:04 grid powers: [4 -3601 -3320]W
evcc  | [site  ] DEBUG 2025/03/21 13:32:04 grid currents: [-2.15 -15.2 -14.1]A
evcc  | [site  ] DEBUG 2025/03/21 13:32:04 pv 1 power: 5373W
evcc  | [site  ] DEBUG 2025/03/21 13:32:04 pv 2 power: 5373W
evcc  | [site  ] DEBUG 2025/03/21 13:32:04 pv power: 10746W
evcc  | [site  ] DEBUG 2025/03/21 13:32:04 battery 1 power: -16W
evcc  | [site  ] DEBUG 2025/03/21 13:32:04 battery 1 soc: 100%
evcc  | [site  ] DEBUG 2025/03/21 13:32:04 site power: -6933W
evcc  | [lp-1  ] DEBUG 2025/03/21 13:32:04 charge total import: 3072.849kWh
evcc  | [lp-1  ] DEBUG 2025/03/21 13:32:04 charger status: C
evcc  | [lp-1  ] DEBUG 2025/03/21 13:32:04 charge start energy: 3072.849kWh
evcc  | [lp-1  ] INFO 2025/03/21 13:32:04 car connected
evcc  | [lp-1  ] DEBUG 2025/03/21 13:32:04 vehicle api refresh
evcc  | [lp-1  ] DEBUG 2025/03/21 13:32:04 pv timer elapse
evcc  | [lp-1  ] DEBUG 2025/03/21 13:32:04 pv timer inactive
evcc  | [lp-1  ] DEBUG 2025/03/21 13:32:04 charge total import: 3072.849kWh
evcc  | [lp-1  ] DEBUG 2025/03/21 13:32:04 charge start energy: 3072.849kWh
evcc  | [lp-1  ] INFO 2025/03/21 13:32:04 start charging ->
evcc  | [lp-1  ] DEBUG 2025/03/21 13:32:04 wake-up timer: stop
[... 77 lines cut ...]
evcc  | [lp-1  ] DEBUG 2025/03/21 13:33:35 soc estimated: 21.08% (vehicle: 21.00%)
evcc  | [lp-1  ] DEBUG 2025/03/21 13:33:35 vehicle soc: 21%
evcc  | [lp-1  ] DEBUG 2025/03/21 13:33:35 vehicle soc limit: 80%
evcc  | [lp-1  ] DEBUG 2025/03/21 13:33:35 vehicle range: 104km
evcc  | [lp-1  ] DEBUG 2025/03/21 13:33:35 pv charge current: 21.9A = 12A + 9.86A (-6801W @ 3p)
evcc  | [lp-1  ] DEBUG 2025/03/21 13:33:35 max charge current: 16A
evcc  | [site  ] DEBUG 2025/03/21 13:34:05 ----
evcc  | [lp-1  ] DEBUG 2025/03/21 13:34:05 charge power: 3707W
evcc  | [lp-1  ] DEBUG 2025/03/21 13:34:05 charge currents: [16.1 0 0]A
evcc  | [site  ] DEBUG 2025/03/21 13:34:05 grid power: -5543W
[... cut ...]
evcc  | [lp-1  ] DEBUG 2025/03/21 13:34:05 charge total import: 3072.950kWh
evcc  | [lp-1  ] DEBUG 2025/03/21 13:34:05 charger status: C
evcc  | [lp-1  ] DEBUG 2025/03/21 13:34:05 soc estimated: 21.12% (vehicle: 21.00%)
evcc  | [lp-1  ] DEBUG 2025/03/21 13:34:05 vehicle soc: 21%
evcc  | [lp-1  ] DEBUG 2025/03/21 13:34:05 vehicle soc limit: 80%
evcc  | [lp-1  ] DEBUG 2025/03/21 13:34:05 vehicle range: 104km
evcc  | [lp-1  ] DEBUG 2025/03/21 13:34:05 pv charge current: 24A = 16A + 8.05A (-5553W @ 3p)`

andig, 2025-03-21:
> Die einzige Änderung was hinzu kommt, ist sie Umstellung von der "klassischen" Fronius Wattpliot Anbindung auf die neue OCPP.

Dann probier doch bitte die alte Anbindung nochmal.

PixelFresser, 2025-03-21:
> > Die einzige Änderung was hinzu kommt, ist sie Umstellung von der "klassischen" Fronius Wattpliot Anbindung auf die neue OCPP.
> 
> Dann probier doch bitte die alte Anbindung nochmal.

Mit der geht's nun wunderbar. Scheint wohl noch irgendwo mit der neuen OCPP Anbindung Probleme zu geben.

Jonnyblue85, 2025-03-21:
Also bei mir sieht es insgesamt gut aus und scheint wieder mit der aktuellen 0.201.1 gefixt zu sein. Ich beobachte weiter.

Danke für den schnellen fix.