evcc-io/evcc #19935: cfos Fehlermeldung (ohne Auswirkungen auf Funktion)

kvogel456, 2025-03-19:
### Describe the bug

Hallo!

habe seit einiger Zeit kurz nach Ladestart einen Fehler im Log, allerdings funktioniert alles wie gewohnt.
Es geht wirklich nur um die Fehlermeldung.

### Steps to reproduce

1. Ladevorgang starten

[image]

### Configuration details

```yaml
cfos template
```

### Log details

```text
cFos Master: climater: strconv.ParseBool: parsing \"off\": invalid syntax
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.201.0

andig, 2025-03-19:
> cfos template

Unklar :(

closed 2025-03-19 by andig

kvogel456, 2025-03-19:
> > cfos template
> 
> Unklar :(

geändert, wollte damit ausdrücken, dass ich das cfos template von evcc benutze

kvogel456, 2025-03-19:
> Danke. Ändert aber nix. Das Template nutzt Modbus. Der Fehler kommt aus JSON. Woher bleibt unklar da Du den Rest Deiner Config verschweigst. Das ist leider Zeitverschwendung.

Oh sorry, dachte der Verweis auf das Template reicht aus. Habe meine gesamte config reingestellt, möchte nichts verschweigen :)

andig, 2025-03-19:
Lesson learned: keine Issuebearbeitung ohne vollständige Config. Egal wie offensichtlich das Problem erscheint.

kvogel456, 2025-03-19:
> cFos Master ist der Name deines Loadpoints und wahrscheinlich steht in deinem mqtt `carconnectivity/0/garage/xxx/climatization/state` "off" und nicht true oder false.

genau so ist es.
werde die zeile rausnehmen. vielen dank!!

kvogel456, 2025-03-19:
> Lesson learned: keine Issuebearbeitung ohne vollständige Config. Egal wie offensichtlich das Problem erscheint.

yes sir! wird beim nächsten mal beachtet 👍