evcc-io/evcc #19936: min + solar overcharging

ronajon, 2025-03-19:
### Describe the bug

i have an Alfen wallbox connected on 3 phases, and a PV system that can deliver max 3500kW
The PV is connected to L1 of my home grid

this morning, there was around 2.5kW of solar energy produced by the PV system, other appliances in the house used around 2kW of power, so there was not enough power left for solar charging the car. 
Setting evcc to "min+solar" resulted in drawing more power from the grid than needed.

It seemed that the min amount for the car was to charge on all 3 phases the min amount (4.1kW), resulting in drawing 1.6kW from the grid (4.1kW - 2.5kW)  
When i changed the charge current from "auto-switching" to "1 phase" the power to the car reduced to the excess amount of power available for charging.

what i would expect:
when charging with "min + solar" the car would just charge with solar power, and only draw power from the grid when the solar system is not able to produces enough power to charge with at least 1.4kW. In my example above that would mean when the house uses 2kW and the car needs 1.4kW at least, with 2.5kW from the PV system, there is 0.9kW needed from the grid, and not 4.1kW 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
#network:
#  schema: http
#  host: evcc.local # .local suffix announces the hostname on MDNS
#  port: 7070

log: error
levels:
  cache: error

# unique installation id
plant: *****

#interval: 30s # control cycle interval

# https://docs.evcc.io/en/docs/sponsorship
#sponsortoken: *****
#sponsortoken: *****
#telemetry: false

meters:
- name: pv
  type: custom
  power:
    source: http
    uri: http://aaa.bbb.ccc.ddd:8123/api/states/sensor.envoy_122030062328_production_phase_1
    method: get
    headers:
      - Authorization: Bearer <secret>
    insecure: true
    jq: .state|tonumber
    timeout: 2s
  energy:
    source: http
    uri: http://aaa.bbb.ccc.ddd:8123/api/states/sensor.envoy_122030062328_lifetime_energy_production
    method: get
    headers:
      - Authorization: Bearer <secret>
    insecure: true
    jq: .state|tonumber
    timeout: 2s
    scale: 0.001
#- name: grid
#  type: template
#  template: slimmelezer
#  usage: grid
[... 602 lines cut ...]
#  type: fixed
#  price: 0.08 # EUR/kWh

#co2: # carbon intensity forecast data
#  type: template
#  template: grünstromindex
#  zip: <zip>"),("gridMeter","db:31"),("solarAccYield","{""pv"":{""accumulated"":133.8350739476157}}") ON CONFLICT (`key`) DO UPDATE SET `value`=`excluded`.`value` 16 <nil>
[site  ] DEBUG 2025/03/19 09:36:45 ----
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0200.9

StefanSchoof, 2025-03-19:
You are running with a grid meter in the mode like https://docs.evcc.io/en/docs/faq#can-i-use-evcc-without-a-grid-meter

> ```
> site:
>   title: Thuis
>   meters:
> #    grid: grid
>     pv:
>     - pv
> ```

ronajon, 2025-03-19:
> perhaps too impatient
> 
> ```
> [lp-1  ] DEBUG 2025/03/19 09:34:47 available power 1745W < 4140W min 3p threshold
> [lp-1  ] DEBUG 2025/03/19 09:34:47 phase scale1p in 2m0s
> ```

that seemed to be right 
i've reset all (wallbox, evcc, homeassistant), disconnected and reconnected the charging cable and restarted charging with "min+solar" 

> [lp-1 ] DEBUG 2025/03/19 11:34:45 available power 3054W < 4140W min 3p threshold
[lp-1 ] DEBUG 2025/03/19 11:34:45 phase scale1p in 0s
[lp-1 ] DEBUG 2025/03/19 11:34:46 switched phases: 1p
[lp-1 ] DEBUG 2025/03/19 11:34:46 phase timer inactive
[lp-1 ] DEBUG 2025/03/19 11:34:46 pv charge current: min 6A > 0.9A (1173W @ 1p, battery: false)

and the car is now charging on 1 phase

ronajon, 2025-03-19:
answered by @VolkerK62

closed 2025-03-19 by ronajon