evcc-io/evcc #19947: Homematic HM-ES-PMSw1-DR einbinden

martin1673, 2025-03-19:
### Describe the bug

Hallo,
ich versuche gerade meinen Homematic Hutschienen Actor HM-ES-PMSw1-DR im EVCC zu integrieren.
Ich bekomme aber immer invalid device als Fehlermeldung.
Wenn ich eine Homematic IP Schalt Mess Steckdose konfigurieren funktioniert alles.
Ich tausche hier nur die Seriennummer des Actors aus und passe den schalt und mess Kanal an.
Kann es sein das keine Homematic Komponenten unterstützt werden oder muss ich die Serien Nummer in irgendwelche Zeichen setzen? die Seriennummern der Homematic Aktoren sind auch kürzer als die der IP Aktoren.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
chargers:
  - name: my_smartswitch
    type: template
    template: homematic
    host: 192.168.0.107 # IP-Adresse oder Hostname
    port: 2010 # Port (optional)
    device: TEQ****100 # Homematic Geräte Id, wie im CCU Webfrontend angezeigt.
    user: ****** # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.) (optional)
    password: ******** # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen) (optional)
    standbypower: 15 # Leistung oberhalb des angegebenen Wertes wird als Ladeleistung gewertet (optional)
    meterchannel: 2
    switchchannel: 1
```

### Log details

```text
[lp-1  ] ERROR 2025/03/19 18:54:41 charger enabled: Invalid device (-2)
[lp-1  ] ERROR 2025/03/19 18:54:51 charge power: Invalid device (-2)
[lp-1  ] ERROR 2025/03/19 18:54:51 charge total import: Invalid device (-2)
[lp-1  ] DEBUG 2025/03/19 18:54:51 charger status: A
[lp-1  ] ERROR 2025/03/19 18:54:52 charger enabled: Invalid device (-2)
[lp-1  ] ERROR 2025/03/19 18:55:01 charge power: Invalid device (-2)
[lp-1  ] ERROR 2025/03/19 18:55:08 charge total import: Invalid device (-2)
[lp-1  ] DEBUG 2025/03/19 18:55:08 charger status: A
[lp-1  ] ERROR 2025/03/19 18:55:08 charger enabled: Invalid device (-2)
[lp-1  ] ERROR 2025/03/19 18:55:11 charge power: Invalid device (-2)
[lp-1  ] ERROR 2025/03/19 18:55:11 charge total import: Invalid device (-2)
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.201.0

Wisley7, 2025-03-19:
Versuch mal den Port 2001, ich glaube dass sollte dann gehen. Wenn nicht, dann muss ich morgen mal in meine config gucken

martin1673, 2025-03-19:
Ja das war es.
Super und danke für die schnelle Antwort

closed 2025-03-19 by martin1673