evcc-io/evcc #19950: evcc dump --cfg und trace.log enthalten noch zuviele sensible Daten

SolarPowerEV, 2025-03-19:
### Describe the bug

Als ich für meinen Bugreport gerade die Config mit `evcc dump --cfg` und das Logfile produziert hatte, habe ich in beiden Outputs noch ein paar nicht durch ****** ersetzte persönliche und m.E. potentiell sensible Daten gefunden.

I. Das sind die Daten, die ich noch im Output von `evcc dump --cfg` gefunden habe:
1. Im Bereich `vehicles:` für Teslas die `clientId` 
   (direkt vor den korrekterweise durch **** ersetzten accessToken und refreshToken)

2. Im Bereich _tariffs:_ unter `co2:` bei grünstromindex die Postleitzahl hinter `zip:` 
   (klar, da kann man jetzt geteilter Meinung sein, ob das noch ok, weil ungenau genug, ist oder nicht)

3. Ebenfalls im Bereich _tariffs:_ unter `solar:` die Werte für `lat:` und `lon:` 
   (die zeigen ja nun eindeutig direkt auf die persönliche Adresse von jemandem)

Ich vermute mal, dass hier einfach nur der Maskierungs-Teil von `evcc dump --cfg` nicht an die allerletzten Erweiterungen angepasst wurde.

II. Und im Logfile auf trace-Level habe ich noch die folgenden Daten gefunden. Kann sein, dass die harmlos oder schon irgendwie verschlüsselt sind, aber ich wollte sie zumindest mal angesprochen haben:

1. Tesla access_Token
2. Tesla Refresh Token
3. Tesla Secret Key
4. Tesla adminPassword
5. VW/Skoda client_id
6. VW/Skoda id_token
7. VW/Skoda user_id
8. VW/Skoda IdentityId und direkt dahinter noch ein Token:
9. VW/Skoda accessToken
10. Fiat userid
11. Fiat APIKey
12. Fiat displayName: (voller Vor- und Nachname)
13. Fiat profile: firstName: und lastName: und später noch nickname:

Ich kann mangels Detail-Kenntnisse natürlich nur mutmaßen und vielleicht sind diese ganzen IDs und Tokens alle harmlos, weil umgeschrieben oder verschlüsselt oder für sich alleine nichts wert. Aber wo ich dabei war, mein Logfile auf sensible Daten durchzusehen, bin ich halt dran hängen geblieben.

Zumindest besteht hier vielleicht potentiell das Risiko, dass jemand zu viele Daten preisgibt, wenn er ein vollständiges Logfile auf trace-Level hier in Github als Teil eines Bug-Reports postet,

### Steps to reproduce

1. yaml für Tesla mit MyTeslamate anlegen
2. `evcc dump --cfg` ausführen
3. Output anschauen

und

1. Logfile auf Trace-Level exportieren
2. Logfile öffnen
3. Im Logfile nach den von mir oben genannten Variablen wie z.B. accessToken suchen...

### Configuration details

```yaml
s.o.
```

### Log details

```text
[db    ] TRACE 2025/03/19 14:52:53 INSERT INTO `settings` (`key`,`value`) VALUES ("vehicle.0.targetSoc","82"),("vehicle.0.targetEnergy","0"),("vehicle.0.minSoc","0"),("vehicle.0.targetTime","0001-01-01T00:00:00Z"),("site.bufferStartSoc","0"),("vehicle.1.targetSoc","85"),("vehicle.1.targetEnergy","0"),("vehicle.1.minSoc","0"),("vehicle.1.targetTime","0001-01-01T00:00:00Z"),("vehicle.2.targetSoc","85"),("vehicle.2.targetEnergy","0"),("vehicle.2.minSoc","0"),("vehicle.2.targetTime","0001-01-01T00:00:00Z"),("telemetry","true"),("lp1.planEnergy","0"),("lp1.planTime","0001-01-01T00:00:00Z"),("lp2.planEnergy","0"),("lp2.planTime","0001-01-01T00:00:00Z"),("lp3.planEnergy","0"),("lp3.planTime","0001-01-01T00:00:00Z"),("bufferSoc","95"),("bufferStartSoc","0"),("prioritySoc","85"),("lp1.mode","pv"),("vehicle.model_y.planTime","0001-01-01T00:00:00Z"),("vehicle.model_y.planSoc","0"),("vehicle.enyaq.planTime","0001-01-01T00:00:00Z"),("vehicle.enyaq.planSoc","0"),("lp3.mode","pv"),("lp3.limitSoc","0"),("vehicle.fiat.planTime","0001-01-01T00:00:00Z"),("vehicle.fiat.planSoc","0"),("lp1.limitSoc","56"),("lp2.limitSoc","0"),("lp2.mode","pv"),("lp1.minCurrent","0.25"),("lp1.maxCurrent","8"),("lp2.minCurrent","6"),("lp2.maxCurrent","30"),("lp3.minCurrent","6"),("lp3.maxCurrent","30"),("lp2.phasesConfigured","0"),("lp1.phasesConfigured","3"),("lp3.phasesConfigured","0"),("tesla-command.7fa25aa6-bf1d-4852-b6b5-49060227f977","{""access_token"":""*****NvdW50X2lkIjoiZmQ0NTA5MmYtY2RmMC00ZjlhLWE0YzMtZGE5YTQzZDAwYzA1IiwiYXV0aF90aW1lIjoxNzQyMzkxNjg3LCJub25jZSI6bnVsbH0.bx6I5A5xuQDRV12onfLCah-XFBeObm0CaQbbq-eE9WtLbdWEqaXQD65WPDUaL3p2EzJ69o6xraUCxeoYmzMUbipZBQ7kwuY8vjbBl90taVMiY2hcb3Yc9-knrtQlLCxQJND6cP0fYIDXHdv78hiShHNt0Kb8jB0ntMU2vc7XGr2MCT7NGCuv1smwXX5HoIAM3OVHwipvj-sPNPToJi_UNERHE25aqvmfkF8iDaQT5ZJ01SVQScORkc7nTtPeveaal9dIm0gM8a-PSo7T_TGubynN-RMucwwnSallHCmGLxh_rns9PtxuXjsEL453IRJAeT4I4iExc8hP3VbPLLPKYA"",""token_type"":""Bearer"",""refresh_token"":""EU_*****a9b98252fcba3a38d2760728afd40"",""expiry"":""2025-03-19T22:41:27.646907935+01:00""}"),("batteryDischargeControl","true"),("jwtSecretKey","*****519213c7e2f22f3c25a5ff2"),("adminPassword","$*****IIzMbzuzURV/YkVJSkbMDA6pB8yAp7xZ2VwOde"),("interval","35000000000"),
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.201.0

StefanSchoof, 2025-03-20:
Ähnliches Issue https://github.com/evcc-io/evcc/issues/19691

premultiply, 2025-03-20:
See redaction list here:
https://github.com/evcc-io/evcc/blob/e87f4acd8d5f94680ed017862db642c4b82a95d8/cmd/helper.go#L50

PR welcome

andig, 2025-03-20:
Closed in 9a18c95fba3a1e1d72df04afbbb861f6a02884ca

closed 2025-03-20 by andig

SolarPowerEV, 2025-03-20:
What about adding clientId to the filter list? Or is that content not sensitive?

And what about also filtering the logfile on trace level? Is there a plan for that?

https://github.com/evcc-io/evcc/issues/19691 doesn't even talk about the trace.log

SolarPowerEV, 2025-03-25:
Könnte bitte noch jemand so nett sein, und meine letzte Frage beantworten?