evcc-io/evcc #19957: vehiclesoc not updated in influxdb

gordonlago, 2025-03-20:
### Describe the bug

I can see the correct vehiclesoc both in the UI and also in the debug trace. I do also see the correct value when accessing the evcc data via iobroker. So, evcc does indeed know the correct soc of my car. However, since 4 days this value does not seem to get updated in the influxdb. The last entry there is 4 days old and whatever I try (restart evcc, restart influxdb, change the pull property), the influxdb data point does not seem to get updated. All other values are updated fine in influxdb. Is this known already?

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
evcc.yaml

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: 

interval: 30s # control cycle interval

sponsortoken: 

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.341 # EUR/kWh
  feedin:
    type: fixed
    price: 0.069 # EUR/kWh
  solar:
    - type: template # Sued
      template: forecast-solar
      lat: 52.59819
      lon: 13.46134
      dec: 40 # 0 = horizontal, 90 = vertikal
      az: 0 # -180 = Norden, -90 = Osten, 0 = Süden, 90 = Westen, 180 = Norden
      kwp: 8.8
      interval: 1h
    - type: template # Nord
      template: forecast-solar
      lat: 52.59819
      lon: 13.46134
      dec: 40 # 0 = horizontal, 90 = vertikal
      az: 180 # -180 = Norden, -90 = Osten, 0 = Süden, 90 = Westen, 180 = Norden
      kwp: 8.8
#      horizon: '0,0,0,0,0,0,0,0,90,90,0,0' # Simuliert Verschattung durch Gelände, [mehr Informationen](https://doc.forecast.solar/horizon)
      interval: 1h

chargers:
- type: template
  template: easee
  user: 
  password:
  charger: EH851568
  timeout: 30s
  authorize: false
  name: easee_wallbox
- name: shelly_heizstab
  type: template
  template: shelly
  host: 192.168.3.230 # IP-Adresse oder Hostname
  user: 
  password: 
  heating: true
  integrateddevice: true
  icon: waterheater

vehicles:
- type: template
  template: enyaq
  title: Enyaq
  icon: car
  user: 
  password: 
  capacity: 77
  phases: 3
  cache: 15m
  mode: pv
  minCurrent: 6
  maxCurrent: 32
  timeout: 30s
  name: enyaq_car

loadpoints:
- title: Easee
  charger: easee_wallbox
  vehicle: enyaq_car
  mode: pv
  priority: 1
  soc:
    poll:
      mode: connected
      interval: 60m
    estimate: false
- title: Heizstab
  charger: shelly_heizstab
  enable:                       # erst einschalten, wenn mehr als 2 KW eingespeist werden
    threshold: -2000
  disable:
    threshold: 500

meters:
- name: maingrid
  type: template
  template: tasmota-sml
  usage: grid
  host: 192.168.3.224
- type: template
  template: enphase
  usage: grid
  host:  192.168.3.223
  token: 
  name: grid1
- type: template
  template: enphase
  usage: pv
  host: 192.168.3.223
  token:   name: pv1
- name: battery1
  type: custom
  power:
    source: http
    uri: https://192.168.3.223/ivp/livedata/status
    method: GET
    auth:
      type: bearer
      password: 
    insecure: true
    jq: .meters.storage.agg_p_mw
    scale: 0.001
  soc:
    source: http
    # uri: https://<Enter Enphase IQ Gateway IP here>/ivp/livedata/status
    # With "/ivp/ensemble/inventory" the soc is always up to date and independent from Enphase Live Status.
    uri: https://192.168.3.223/ivp/ensemble/inventory
    method: GET
    auth:
      type: bearer
      password:     insecure: true
    # jq: .meters.soc
    jq: .[].devices | map(.percentFull) | add / length
  capacity: 10.5 # kWh

site:
  title: Trese12
  meters:
    grid: maingrid
#    grid: grid1
    pv:
    - pv1
    battery:
    - battery1
#  residualpower: 100

influx:
  url: http://localhost:8086
  database: evcc
  # user:
  # password:
```

### Log details

```text
[lp-1 ] DEBUG 2025/03/20 13:10:30 next soc poll remaining time: 10m56s
[lp-1 ] DEBUG 2025/03/20 13:10:30 limitSoc reached: 80.0% > 80%
[lp-1 ] DEBUG 2025/03/20 13:10:58 charge power: 0W
[lp-1 ] DEBUG 2025/03/20 13:10:58 charge currents: [0.009 0.01 0.01]A
[lp-1 ] DEBUG 2025/03/20 13:11:28 charge power: 0W
[lp-1 ] DEBUG 2025/03/20 13:11:28 charge currents: [0.009 0.01 0.01]A
[lp-1 ] DEBUG 2025/03/20 13:11:31 charge total import: 11380.953kWh
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.201.0

andig, 2025-03-20:
Pls check influxdb trace log

gordonlago, 2025-03-20:
Anything specific to look at? I think I can see the writing to the db:

[httpd] ::1 - - [20/Mar/2025:12:24:28 +0000] "POST /api/v2/write?bucket=evcc&org=&precision=s HTTP/1.1 " 204 0 "-" "influxdb-client-go/2.14.0 (linux; arm64)" 44ce24b2-0586-11f0-9bc2-d83addfceecd 7608

[httpd] ::1 - - [20/Mar/2025:12:24:29 +0000] "POST /api/v2/write?bucket=evcc&org=&precision=s HTTP/1.1 " 204 0 "-" "influxdb-client-go/2.14.0 (linux; arm64)" 4566bfca-0586-11f0-9bc3-d83addfceecd 6536

[httpd] ::1 - - [20/Mar/2025:12:24:31 +0000] "POST /api/v2/write?bucket=evcc&org=&precision=s HTTP/1.1 " 204 0 "-" "influxdb-client-go/2.14.0 (linux; arm64)" 4697ef98-0586-11f0-9bc4-d83addfceecd 6633

andig, 2025-03-20:
The evcc log!

gordonlago, 2025-03-20:
Makes sense...I guess evcc is sending correct data and the error must be somewhere else. I guess topic can be closed here.

[influx] TRACE 2025/03/20 15:20:14 write vehicleSoc=map[value:80] (map[loadpoint:Easee vehicle:Enyaq])

gordonlago, 2025-03-20:
Oh, wait...now I see the following log with alternating values:

[influx] TRACE 2025/03/20 15:27:29 write vehicleSoc=map[value:0] (map[loadpoint:Easee vehicle:Enyaq])
[influx] TRACE 2025/03/20 15:27:33 write vehicleSoc=map[value:80] (map[loadpoint:Easee vehicle:Enyaq])

closed 2025-03-20 by andig