evcc-io/evcc #19961: Wallbox setting from from UI

timpetersson81, 2025-03-20:
### Describe the bug

Im moving from config file to UI and cannot connect my wallbox here. I keep getting: Request failed with status code 502 And creation failed

[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  port: 7070

log: debug

interval: 30s

meters:
  - name: my_grid
    type: template
    template: sungrow-hybrid
    usage: grid
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.1.63 # Hostname
    port: 502 # Port
    timeout: 10s # optional

  - name: my_pv
    type: template
    template: sungrow-hybrid
    usage: pv

    modbus: tcpip
    id: 1
    host: 192.168.1.63 # Hostname
    port: 502 # Port
    timeout: 10s # optional

  - name: my_battery
    type: template
    template: sungrow-hybrid
    usage: battery

    modbus: tcpip
    id: 1
    host: 192.168.1.63 # Hostname
    port: 502 # Port
    timeout: 10s # optional

vehicles:
  - name: Xpeng
    type: template
    template: offline
    title: # Will be displayed in the user interface (optional)
    capacity: 100

site:
  title: Zuhause
  meters:
    grid: my_grid
    pv: my_pv
    battery: my_battery

tariffs:
  currency: DKK
  grid:
    type: custom
    forecast:
      source: http
      uri: https://linkcia.dk/elpriser.php
```

### Log details

```text
[site ] WARN 2025/03/20 17:00:15 vehicle 'Xpeng' does not support automatic detection
[site ] WARN 2025/03/20 17:00:15 vehicle 'Nannas golf' does not support automatic detection
[EVB1-1] WARN 2025/03/20 17:03:26 failed triggering StatusNotification: cannot send request 2509817846, no client EVB1 exists
[792713-1] WARN 2025/03/20 17:06:39 failed triggering StatusNotification: cannot send request 2302985174, no client 792713 exists
[792713-1] WARN 2025/03/20 17:07:35 failed triggering StatusNotification: cannot send request 272859017, no client 792713 exists
[123-1 ] WARN 2025/03/20 17:13:34 failed triggering StatusNotification: cannot send request 1486938505, no client 123 exists
[1234-1] WARN 2025/03/20 17:14:35 failed triggering StatusNotification: cannot send request 540957244, no client 1234 exists
[1234-1] WARN 2025/03/20 17:20:33 failed triggering StatusNotification: cannot send request 2105146836, no client 1234 exists
[wb-1234-1] WARN 2025/03/20 17:22:21 failed triggering StatusNotification: cannot send request 2575073922, no client wb-1234 exists
[wb-1234-1] WARN 2025/03/20 17:23:17 failed triggering StatusNotification: cannot send request 2304364268, no client wb-1234 exists
[wb-1234-1] WARN 2025/03/20 17:23:51 failed triggering StatusNotification: cannot send request 248396607, no client wb-1234 exists
[wb1234-1] WARN 2025/03/20 17:27:43 failed triggering StatusNotification: cannot send request 1735669453, no client wb1234 exists
[wb1234-1] WARN 2025/03/20 17:27:51 failed triggering StatusNotification: cannot send request 1153055255, no client wb1234 exists
[SN 792713-1] WARN 2025/03/20 17:30:06 failed triggering StatusNotification: cannot send request 2657304285, no client SN 792713 exists
[SN792713-1] WARN 2025/03/20 17:30:20 failed triggering StatusNotification: cannot send request 2703367997, no client SN792713 exists
[SN792713-1] WARN 2025/03/20 17:30:33 failed triggering StatusNotification: cannot send request 2350456757, no client SN792713 exists
[SN-792713-2] WARN 2025/03/20 17:31:35 failed triggering StatusNotification: cannot send request 1268925933, no client SN-792713 exists
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.201.1

VolkerK62, 2025-03-20:
I am not sure, but I guess, OCPP ist actually not possible via UI-Config.

andig, 2025-03-20:
> my wallbox

Which one would that be?

timpetersson81, 2025-03-20:
This is from my old config file:
chargers:
  - name: my_charger
    type: template
    template: pulsarplus-fw5

That worked. I have a wallbox commander s2

timpetersson81, 2025-03-20:
> > my wallbox
> 
> Which one would that be?

This is from my old config file:
chargers:

name: my_charger
type: template
template: pulsarplus-fw5
That worked. I have a wallbox commander s2

closed 2025-03-21 by andig