evcc-io/evcc #19963: Ladeplanung funktioniert seit 0.201.0 nicht mehr (EASEE Home)

brmien, 2025-03-20:
### Describe the bug

Seit 0.201.0 funktioniert die Ladeplanung bei mir nicht mehr richtig. Ich habe dies nun in zwei Situationen beobachtet.

1. Plan steht auf 60%, dieser Zustand soll am kommenden Tag morgens um 7:00 erreicht werden. Fahrzeug wird angeschlossen, gegen 18:00 Uhr am Vorabend, und der SOC liegt unter dem minimalen Ladezustand (25%) bei Ankunft. evcc fängt korrekterweise an zu laden, stoppt jedoch nicht bei 25% temporär (und startet wieder so, dass das Fahrzeug um 7:00 geladen ist), , sondern lädt **sofort** durch bis auf die 60%.
2. Ich habe gerade noch einen Versuch gemacht, auch, um einen Screenshot produzieren zu können. Das Fahrzeug wurde bei 45% (17:55) angeschlossen, und die Ladeplanung wurde auf 50% bis 19:00 gestellt. evcc startet sofort das Laden, meldet aber die Fertigstellung gegen 18:18 (siehe Screenshot).

[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
tom@raspberrypi:~ $ evcc dump --cfg
[main  ] INFO 2025/03/20 18:17:02 evcc 0.201.1
[main  ] INFO 2025/03/20 18:17:02 using config file: /etc/evcc.yaml
[db    ] INFO 2025/03/20 18:17:02 using sqlite database: /var/lib/evcc/evcc.db
[lp-2  ] WARN 2025/03/20 18:17:05 poll mode '{connected 1h0m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[lp-1  ] DEBUG 2025/03/20 18:17:07 charge total import: 0.000kWh
[lp-2  ] DEBUG 2025/03/20 18:17:07 charge total import: 3129.787kWh
 
Konfiguration (/etc/evcc.yaml):
 
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070
 
database:
  type: sqlite
  dsn: /var/lib/evcc/evcc.db
 
log: debug
levels:
  cache: error
 
# unique installation id
plant: *****
 
interval: 30s # control cycle interval
 
sponsortoken: *****
 
# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false
 
meters:
- type: template
  template: openems 
  usage: grid  
  host: 192.168.2.21  
  name: grid1
- type: template
  template: openems 
  usage: pv  
  host: 192.168.2.21  
  name: pv2
- type: template
  template: openems 
  usage: battery  
  host: 192.168.2.21  
  name: battery3
[... 645 lines cut ...]
[lp-2  ] DEBUG 2025/03/20 17:55:56 charger status: C
[lp-2  ] DEBUG 2025/03/20 17:55:56 soc estimated: 46.17% (vehicle: 45.00%)
[lp-2  ] DEBUG 2025/03/20 17:55:56 vehicle soc: 46%
[lp-2  ] DEBUG 2025/03/20 17:55:56 vehicle soc limit: 80%
[lp-2  ] DEBUG 2025/03/20 17:55:56 vehicle range: 202km
[lp-2  ] DEBUG 2025/03/20 17:55:56 plan: charge 17m47s between 2025-03-20 17:55:57 +0100 CET until 2025-03-20 19:00:00 +0100 CET (power: 11040W, avg cost: 153.580)
[site  ] DEBUG 2025/03/20 17:55:56 solar forecast: accumulated 1.134kWh, produced 1.394kWh, scale 1.229
[site  ] DEBUG 2025/03/20 17:56:26 ----
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.201.1

andig, 2025-03-20:
> und der SOC liegt unter dem minimalen Ladezustand (25%) bei Ankunft. evcc fängt korrekterweise an zu laden, stoppt jedoch nicht bei 25% temporär (und startet wieder so, dass das Fahrzeug um 7:00 geladen ist), , sondern lädt sofort durch bis auf die 60%.

Wo sehe ich davon was im Log?

brmien, 2025-03-20:
Leider gar nicht. Sorry, hätte ich erwähnen sollen. Das war heute Nacht irgendwann. Ich habe nur für den zweiten Teil das Logfile.

andig, 2025-03-20:
Was hat das jetzt mit deiner Fehlerbeschreibung aus dem Issue zu tun?

brmien, 2025-03-20:
Okay, nun hat evcc doch bei 50% angehalten. Dennoch ist es doch sehr komisch, dass evcc für die letzten 5% Ladung bis 18:00 morgen Abend schon um 9:00 morgens mit dem Laden beginnen will.

[image]

naltatis, 2025-03-20:
> manchmal möchte ich mein Fahrzeug zu 100% aufladen, dann soll er dies jedoch so spät wie möglich machen, um die Batterie zu schonen

Geht heute noch nicht automatisch mit dem Planer. Müsstest du externe Automation (API) für nutzen.

Der Feature Request ist aber bereits Teil von https://github.com/evcc-io/evcc/issues/11654

closed 2025-03-20 by naltatis

thecem, 2025-09-09:
Die Ladedauer muss eine Länge von min 10min haben, dann geht es. Kurze slot kann man über die Absenkung des Ladestrom verlängern.