evcc-io/evcc #19964: Kein Überschussladen WB Daheim Laden Smart

Mafiazwerg, 2025-03-20:
### Describe the bug

Auto wird angesteckt mit Karte wird Wallbox gestartet. Modus PV oder min+PV ist vorher in evcc eingestellt worden. Ladung startet und evcc regelt nach Vorgaben alles prima. 
Jetzt kommt der Fall keine Sonne oder es trifft sonst irgendein anderes Eingabe ein das evcc die Wallbox auf 0 Strom regeln muss. Evcc regelt auf null aber wenn aus diesen Zustand wieder geladen werden soll werden Fehler im Log gezeigt : charger status: invalid status: 0.
Egal was ich hier mache in evcc ich bekomme die Wallbox erst wieder zum Laden wenn ich manuell mit Karte den Startvorgang einleite. 
Bin ich der einzigste mit dieses Verhalten?
Firmware ist Software Version (H)
AC_M3W_3.09STPn3.P105
Software Version (P)
AC_NEW_1.05

Daheim laden Support sagt Nachladen ist freigeschaltet aber bleib mit mein Problem alleine gelassen bis dato.

Vielleicht findet sich hier Hilfe.

[evcc-20250320-160301-debug.log.txt](https://github.com/user-attachments/files/19372526/evcc-20250320-160301-debug.log.txt)
[evcc-20250319-161211-debug.log.txt](https://github.com/user-attachments/files/19372527/evcc-20250319-161211-debug.log.txt)

### Steps to reproduce

1. Wird geladen
2. kein Überschuss 
3. Sollte wieder laden da überschuss etc.
4. Fehler im Log 
5. Kein laden mehr bis manuell neu gestartet an WB
...

### Configuration details

```yaml
interval: 30s # control cycle interval

meters:
- type: template
  template: huawei-sun2000 
  id: 1  
  host: 192.168.178.93  
  port: 502  
  usage: grid  
  modbus: tcpip  
  timeout: 20s  
  name: grid1
- type: template
  template: huawei-sun2000 
  id: 1  
  host: 192.168.178.93  
  port: 502  
  usage: pv  
  modbus: tcpip  
  timeout: 20s  
  name: pv2
- type: template
  template: huawei-sun2000 
  id: 1  
  host: 192.168.178.93  
  port: 502  
  usage: battery  
  modbus: tcpip  
  timeout: 20s  
  name: battery3

chargers:
- type: template
  template: daheimladen-mb 
  host: 192.168.178.164  
  port: 502  
  name: wallbox6

vehicles:
- type: template
[... 358 lines cut ...]
[lp-1  ] DEBUG 2025/03/19 14:58:56 charger status: C
[lp-1  ] DEBUG 2025/03/19 14:58:56 soc estimated: 74.00% (vehicle: 65.00%)
[lp-1  ] DEBUG 2025/03/19 14:58:56 vehicle soc: 74%
[lp-1  ] DEBUG 2025/03/19 14:58:56 vehicle soc limit: 80%
[lp-1  ] DEBUG 2025/03/19 14:58:56 vehicle range: 224km
[lp-1  ] DEBUG 2025/03/19 14:58:56 pv charge current: min 6A > 4.7A (896W @ 3p, battery: true)
[site  ] DEBUG 2025/03/19 14:59:19 ----
[lp-1
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.201.1

andig, 2025-03-21:
> charger status: invalid status: 0

Hast du mal bitte ein trace log?

VolkerK62, 2025-03-21:
Im "alten" Modbus Dokument gab es keinen Status 0.
Im "neuen" Dokument ist Status 0 = "unknown state"

Von der DaheimLaden Homepage:
[ModBusTCP-Smart.pdf](https://github.com/user-attachments/files/19383488/ModBusTCP-Smart.pdf)

Mafiazwerg, 2025-03-21:
Logs reiche ich heute Nachmittag nach wenn Auto wieder geladen wird.

andig, 2025-03-21:
Da muss leider daheimladen ran. Keine Ahnung, was evcc mit einem „unknown“ Status anfangen soll?

Mafiazwerg, 2025-03-21:
Hier der Log :  Auto wird geladen danach gestoppt in evcc durch keine Sonne etc. wenn jetzt wieder genug Sonne dann startet Ladung nicht mehr Automatisch. 

[evcc-20250321-131436-trace.log.txt](https://github.com/user-attachments/files/19388926/evcc-20250321-131436-trace.log.txt)

andig, 2025-03-21:
/cc @StefanSchoof die WB meldet über Modbus einen unklaren Status 0 der sich dann auch nicht mehr ändert- Firmwarefehler?

StefanSchoof, 2025-03-21:
@andig Ich glaube du wolltest jemanden anderen anpingen.

andig, 2025-03-21:
Argh, sorry!

Andre44260, 2025-03-21:
Hallo, 
Ich haben den Selben Fehler, zudem ist mir Aufgefallen das EVCC die Anzeige der Geladenen Energie um genau Faktor 10 zuviel Anzeigt zudem hat sich vom Alten Modbus zum Neuen ein paar dinge geändert ( Zb. Register 28/27 ) Nach welchem Register ist es bei EVCC hinterlegt ? 

[image]
[image]

[DaheimLaden ModBus TCP Spezifikation 1.4 (1).pdf](https://github.com/user-attachments/files/19397076/DaheimLaden.ModBus.TCP.Spezifikation.1.4.1.pdf)

[ModBusTCP-Smart.pdf](https://github.com/user-attachments/files/19397079/ModBusTCP-Smart.pdf)

andig, 2025-03-21:
Es wird also für ein wunderbar funktionierendes Gerät einfach das Registerlayout geändert? Das ist aber arg… unglücklich!

Wie wärs mit FW Downgrade?

VolkerK62, 2025-03-22:
Ich habe die Implementierung mit der neuen Modbus Dokumentation verglichen. 
Da gibt es keine Abweichungen.
Lediglich `dlRegCardId          = 54  // Chr[16] RO UTF16` gibt es nicht mehr. 
Aber das funktionierte von Anfang an nicht.

VolkerK62, 2025-03-22:
@Andre44260 
>zudem ist mir Aufgefallen das EVCC die Anzeige der Geladenen Energie / Zählerstand um genau Faktor 10 zuviel Anzeigt

Ich kann dir nicht ganz folgen.
Laut evcc hast du 45 Minuten geladen. Wenn das stimmt, dann wären 693Wh seeehr wenig.
Woher kommt diese Zeile "Ladezusammenfassung"?

Mafiazwerg, 2025-03-22:
Also in meinem Fall kann ich das nur bestätigen mit dem falschen werten wie bei @Andre44260  3.63 kWh würde passen aber 36.3 kWh wie in meinem Screenshots passt nicht. Es ist auch interessant wie @andig sagt das die WB ein super funktionierendes Gerät ist genau deswegen weil ich eigentlich nichts negatives gelesen habe hab ich meine alte WB gegen diese getauscht und seit Tag 1 funktioniert Überschussladen nicht. Egal mit welcher Firmware ( daheim laden support hat mehrere probiert ohne Erfolg ). 

[image]

VolkerK62, 2025-03-22:
Ok. Das ist jetzt offenischtlich. Da scheint es mit der neuen "PRO" wohl Probleme zu geben.

Ich habe seit über 3 Jahren den DaheimLader. Ist wie der Käfer, läuft und läuft und läuft.

VolkerK62, 2025-03-22:
Sorry, aber die Dokumentation alt vs neu ist identisch.
Da stimmt dann etwas in der PRO-Firmware nicht. 
De facto hat deine Wallbox also nur 51,3kWh "auf der Uhr"?

Mafiazwerg, 2025-03-22:
Richtig ! @VolkerK62

Andre44260, 2025-03-22:
@Mafiazwerg  wie hast du die DaheimLaden in dein EVCC implementiert, per evcc.yaml ging sie bei mir auch nur ohne "Ladesteuerung" ( als ich habe sie gesehen aber ich konnte sie nicht in ihrem Ladeverhalten beeinflussen. )

Probier mal per WebUI Konfiguration hinzuzufügen. 

**WebUI -> Konfiguration -> Experimentell -> An - Ladenpunkte Hinzufügen**. 

so hat es bei mir dann plötzlich Funktioniert. 

@VolkerK62  ja, ich hatte einen Plugin mit minimalster ladegeschwindigkeit geladen, meine Daheim Laden hat einen Zählerstand von 30,94 KWH ( ganz neu )  in EVCC wird 309,4 KWH als Zählerstand angegeben. 

Daheim Laden hat mir ebenfalls eine ältere FW aufgespielt 

Hardware Version               **YEVAXYJTX13E-B**
Software Version (H)          **AC_M3W_3.09STPn3.P105**
Software Version (P)          **AC_NEW_1.05**
LCD Version                       **YEVAXYJ7001X-A**
Modbus RTU Version         **V1.01**
Web Version                      **V1.0.1**

Mafiazwerg, 2025-03-22:
@Andre44260 naja ich hab direkt am Anfang das genau so gemacht mit dem einfügen wie du mit meinem Homeassistent mit dem Ergebnis kein Überschussladen möglich. Der Support sagte probieren Sie evcc auf einem Raspberry Pi da dies genau ihre Testanforderungen sind , gesagt getan, Ergebnis, keine Besserung. Sobald WB die Anweisungen bekommt von evcc keinen Strom ins Auto zulieferern macht Sie das zwar aber wenn dann die Anweisungen kommt wieder Strom zu Liefern streikt die Wallbox. Anscheinend bleibt die in eine Art Schlafmodus. Wenn bei dir mit genau der gleichen Firmware alles funktioniert außer der Fehler mit der falschen kWh Anzeige Frage ich mich was bei mir falsch ist. Wo hast du evcc den installiert ?

VolkerK62, 2025-03-22:
Was mich wundert, im Handbuch der PRO findet sich keine Einstellmöglichkeit bzgl. "Nachladen"

Andre44260, 2025-03-22:
@Mafiazwerg  
Entschuldige, wenn die Wallbox auf 0 A Regelt aufgrund zu wenig PV beginnt sie nicht wieder mit dem Laden. 
Ich habe verstanden das die PV Regelung von evcc nicht funktioniert. 

@VolkerK62  
ja, das geht nur über den Support und heißt bei der Touch nichtmehr Nachladen sondern RSDA, die haben das auf ihrer Website mit einem Blogbeitrag, wobei ich auch nicht ganz verstehe warum das nicht von haus aus Aktiviert ist.