evcc-io/evcc #19965: teslalogger vehicle integration not working

bill340, 2025-03-20:
### Describe the bug

When using teslalogger vehicle integration in evcc I get the following error:

chargeStatus: unexpected status: 404 (Not Found)

### Steps to reproduce

Add a new vehicle with teslalogger integration in the evcc gui.

### Configuration details

```yaml
regardless of evcc.yaml
The error is already when adding the vehicle in a clean config.
```

### Log details

```text
The error is already when adding the vehicle in a clean config.
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-03-21:
Please add a trace log. Closing until available.

closed 2025-03-21 by andig

bill340, 2025-03-21:
Attached is the trace log.
Here is the relevant part:

[db:17 ] TRACE 2025/03/21 08:13:59 GET http://192.168.2.2:5010/currentjson/1
[db:17 ] TRACE 2025/03/21 08:13:59 URL Not Found!
[main ] ERROR 2025/03/21 08:13:59 vehicle status: unexpected status: 404 (Not Found)

[evcc-20250321-081540-trace.log](https://github.com/user-attachments/files/19383805/evcc-20250321-081540-trace.log)

andig, 2025-03-21:
Wenn es die URL nicht gibt kann evcc da auch nix abholen. Ggf. hilft ein Ticket auf https://github.com/bassmaster187/TeslaLogger oder @Adminius hat eine Idee.

Adminius, 2025-03-21:
I'm still using it, it's working

bill340, 2025-03-21:
So any idea what the problem could be? Could it be because teslalogger is running in docker?

Adminius, 2025-03-21:
Looks like example IP "192.168.2.2"
@bill340 Please check your settings

bill340, 2025-03-21:
The IP is correct.
It is my unraid server where Teslalogger runs in docker.
GUI, database, etc. of teslalogger is running fine on this IP.

Adminius, 2025-03-21:
ID "1" is correct?

Can you call http://192.168.2.2:5010/currentjson/1 via browser? 
Admin Panel: http://192.168.2.2:5010/admin

bill340, 2025-03-21:
Admin Panel I can get with
http://192.168.2.2:8888/admin

Can you call http://192.168.2.2:5010/currentjson/1 via browser?
-> I also get "URL Not Found!"

the line is in my docker-compose.yml

version: "3"
services:

  teslalogger:
    build: docker/teslalogger/.
    restart: always
    volumes:
      - ./TeslaLogger/www:/var/www/html
      - ./TeslaLogger/bin:/etc/teslalogger
      - ./TeslaLogger/GrafanaDashboards/:/var/lib/grafana/dashboards/
      - ./TeslaLogger/GrafanaPlugins/:/var/lib/grafana/plugins
      - ./docker/teslalogger/Dockerfile:/tmp/teslalogger-DOCKER
      - teslalogger-tmp:/tmp/
    depends_on:
      - database
    environment:
      - TZ=Europe/Berlin
    ports:
      - 5010:5000

  database:
    image: mariadb:10.4.7
    restart: always
    env_file:
      - .env
    volumes:
      - ./TeslaLogger/sqlschema.sql:/docker-entrypoint-initdb.d/sqlschema.sql
      - ./TeslaLogger/mysql:/var/lib/mysql
    ports:
      - 3307:3307
    environment:
      - TZ=Europe/Berlin

  grafana:
    image: grafana/grafana:10.0.1
    restart: always
    environment:
      - GF_SECURITY_ADMIN_PASSWORD=teslalogger
      - GF_PLUGINS_ALLOW_LOADING_UNSIGNED_PLUGINS=natel-discrete-panel,pr0ps-trackmap-panel,teslalogger-timeline-panel
      - TZ=Europe/Berlin
    ports:
      - 3000:3000
    volumes:
      - ./TeslaLogger/bin:/etc/teslalogger
      - ./TeslaLogger/GrafanaDashboards/:/var/lib/grafana/dashboards/
      - ./TeslaLogger/GrafanaPlugins/:/var/lib/grafana/plugins
      - ./TeslaLogger/GrafanaDB:/var/lib/grafana/
      - ./TeslaLogger/GrafanaConfig/datasource.yaml:/etc/grafana/provisioning/datasources/datasource.yml
      - ./TeslaLogger/GrafanaConfig/sample.yaml:/etc/grafana/provisioning/dashboards/dashboards.yml
    depends_on:
      - database

  webserver:
    build: docker/webserver/.
    restart: always
    volumes:
      - ./docker/webserver/php.ini:/usr/local/etc/php/php.ini
      - ./TeslaLogger/www:/var/www/html
      - ./TeslaLogger/bin:/etc/teslalogger
      - ./docker/teslalogger/Dockerfile:/tmp/teslalogger-DOCKER
      - ./TeslaLogger/GrafanaConfig/datasource.yaml:/tmp/datasource-DOCKER
      - teslalogger-tmp:/tmp/
    ports:
      - 8888:80
    environment:
      - TZ=Europe/Berlin

volumes:
    teslalogger-tmp:

Adminius, 2025-03-21:
> -> I also get "URL Not Found!"

So, here the is the problem. The question is why?
- Restart TL docker and try agaein
- Check if any NAT rules are blocking on UnRaid/Router
- 
I'm not a docker specialist, can't really help here :/ 

Anyway, EVCC is working as expected.

bill340, 2025-03-21:
Already did several restarts...
No NAT blocking between unraid and the docker on the same machine... As far as I know...
Can anyone else help here?

Adminius, 2025-03-21:
Is ID=1 a right one?

[image]

bill340, 2025-03-21:
That was the problem...
I thought it is "1" because it is the only car in teslalogger right now. 
But from previous usage there were somehow other IDs that I actually deleted already...

However: IT IS WORKING NOW!! Thanks for your help!!!