evcc-io/evcc #19966: Solax Template Grid Meter um Phasenströme erweitern

FSXPET, 2025-03-21:
Hallo
Ich wollte nun das Lastmanagement nutzen, um meinen Hausanschluss nicht zu überlasten.
Mit "maxPower" geht das, mit "maxCurrent" nicht, da das meter grid keine Ströme misst.
Sicherer und besser ist aber ja "maxCurrent".

Solax stellt über Modbus die Phasenströme zur Verfügung:
Hybrid X1: 0x0001
Hybrid X3:
- L1: 0x006B
- L2: 0x006F
- L3: 0x0073

Kann das mit ins Template aufgenommen werden?

Zum Solax Template hatte ich auch schon diesen Post, gerne mit anschauen/aufnehmen: #18580

andig, 2025-03-21:
Gerne PR

FSXPET, 2025-04-01:
Leider ist es wohl doch nicht generell möglich. Ich hatte die Register in der Modbus-Beschreibung gefunden. Aber bei mir funktionierten sie nicht.

Antwort von Solax:
Register werden nur unterstützt, wenn der Kunde einen Chint-Meter verwendet, und es ist nicht möglich, wenn er nur an den CT anschließt.

FSXPET, 2025-04-02:
Weitere Antwort von Solax:
Derzeit ist es nicht möglich, Register für den CT zu verwenden.
Ich habe unser IT-Team bereits über die Register für den CT informiert, sodass wir diese Funktion in Zukunft haben werden, aber wann, kann ich Ihnen nicht sagen.

closed 2025-04-14 by github-actions[bot]