evcc-io/evcc #19970: Cupra Tavascan not detected as plugged in

nilseckert, 2025-03-21:
### Describe the bug

The Cupra Tavascan is correctly configured via the UI (not evcc.yaml) and the state of charge (SOC) is retrieved successfully. However, EVCC does not recognize the vehicle as plugged in, even though the vehicle is physically connected to the charger.

In the official Cupra app, the status is shown as:

“Charging cable connected and not locked. External power supply: not available.”
This suggests the cable is detected by the backend, but charging is not yet active (typical standby state before charging).

Using the Home Assistant "Cupra WeConnect" integration, the plug status is correctly shown as "connected" at the same time.

As a result, EVCC does not select the car, instead "Guest Vehicle" is selected and charging does not start automatically.

We previously owned an Audi and an ID.3, both worked flawlessly.

### Steps to reproduce

Buy a Tavascan
Setup in EVCC
Plug car in ;-)

### Configuration details

```yaml
Cupra Connect V4 API
Credentials (validated correctly)
VIN entered
```

### Log details

```text
Status from cupra with log level trace

{
  "engines": {
    "primary": {
      "fuelType": "electric",
      "rangeKm": 377.0,
      "levelPct": 75.0
    }
  },
  "services": {
    "charging": {
      "status": "Error",
      "targetPct": 80,
      "currentPct": 75,
      "chargeSettings": "default",
      "chargeMode": "manual",
      "preferredChargeMode": "manual",
      "active": false,
      "remainingTime": 0,
      "progressBarPct": 94.0
    },
    "climatisation": {
      "status": "Off",
      "targetTemperatureCelsius": 22.0,
      "targetTemperatureFahrenheit": 72.0,
      "climatisationTrigger": "off",
      "active": false,
      "remainingTime": 0,
      "progressBarPct": 0.0
    },
    "windowHeating": {
      "status": "Off",
      "active": false
    }
  }
}
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.201.1

mfuchs1984, 2025-03-21:
Evcc benutzt `Services.Charging.Status` und interpretiert `connected` und `readyforcharging` als Status B (angesteckt) und `charging` als Status C (laufend). Deiner meldet `error`. Ggf. könnte man den Status auch als Status B interpretieren, da er in dem Fall auch angesteckt sein wird und nicht lädt.

mfuchs1984, 2025-03-21:
Als Workaround könnte `welcomecharge: true in der vehicle config funktionieren.

andig, 2025-03-21:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2025-03-21 by andig

nilseckert, 2025-03-21:
> Als Workaround könnte `welcomecharge: true` in der vehicle config funktionieren.

Das habe ich auch schon probiert, aber das Standard-Fahrzeug ist ja Gast-Fahrzeug und da gibt es kein WelcomeCharge.

nilseckert, 2025-03-21:
> Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

Ich kann gerne ein detailliertes Log bereitstellen. Reicht es, wenn ich Cupra auf debug stelle? Ich hatte es auf trace und da sind dann auch die Bearer Token mit drin, das wollte ich nicht posten, weil ich sonst ggf. eine Stelle übersehe und mein Account dann exponiert ist ;-)

andig, 2025-03-21:
debug sollte reichen

mfuchs1984, 2025-03-21:
> > Als Workaround könnte `welcomecharge: true` in der vehicle config funktionieren.
> 
> Das habe ich auch schon probiert, aber das Standard-Fahrzeug ist ja Gast-Fahrzeug und da gibt es kein WelcomeCharge.

Sobald ein konfiguriertes Fahrzeug welcomecharge konfiguriert hat, startet evcc den welcome charge.

nilseckert, 2025-03-21:
Ich probiere es morgen früh mal mit dem nightly build aus und gebe dann Bescheid, ob es so funktioniert!

PR #19982 (2025-03-21): fix: treat charging status "error" as StatusB for cupra provider
Fix #19970