evcc-io/evcc #19973: Dynamic tariff poll interval

speertje, 2025-03-21:
**Is your feature request related to a problem? Please describe.**
The 3h interval in enever.yaml is to frequent. Enever has a query cap of 250 request/per month. If you pull grid and feedin prices you run out of queries in 15 day's. 

**Describe the solution you'd like**
Is it possible to add an poll interval option to tariffs in evcc.yaml?

**Describe alternatives you've considered**
Currently I stopped using dynamic energy prices

naltatis, 2025-03-21:
What would be your suggestion? 8h?

speertje, 2025-03-21:
> What would be your suggestion? 6h?

6h would leave little or no room for restarts of evcc. I think 10/12 hours would be more than enough

naltatis, 2025-03-21:
The above PR makes this interval configurable (as you proposed).

closed 2025-03-23 by andig