evcc-io/evcc #19980: Custom vehicule's "maxcurrent" expected data type is not matching documentation

jaytonic, 2025-03-21:
### Describe the bug

In the [documentation](https://docs.evcc.io/en/docs/devices/vehicles#custom) , I'm supposed to be able to configure the maxcurrent as an int.

I've currently a test (fake) vehicule:

```
vehicles:
  - name: vehicle_1
    title: Fake car
    type: custom
    soc:
      source: js
      vm: shared
      script: |
        var lp = state.loadpoints[0];
        if (lp.chargepower > 0) lp.vehicleSoc+=0.1; else lp.vehicleSoc-=0.1;
        if (lp.vehicleSoc < 15) lp.vehicleSoc = 80;
        if (lp.vehicleSoc > 80) lp.vehicleSoc = 15;
        lp.vehicleSoc;
    range:
      source: js
      vm: shared
      script: |
        var lp = state.loadpoints[0]
        var range = (100 * lp.vehicleSoc) / 15;
        range
    capacity: 100
    maxcurrent: 16
```

And when I'm trying to activate the config, I get the following error message:

> vehicle: cannot create vehicle 'vehicle_1': cannot create vehicle type 'custom': decoding failed due to the following error(s): 'MaxCurrent' expected a map, got 'int'

I'm not sure what/how to put in this field to match the "map" format?

### Steps to reproduce

1. Create a vehicule with the example provided
2. Restart
3. Check the UI to see the error message

### Configuration details

```yaml
meters:
  - name: my_grid
    type: template
    template: enphase
    usage: grid
    host: 192.168.0.196 # IP address or hostname
    token: AAA

  - name: my_pv
    type: template
    template: enphase
    usage: pv
    host: 192.168.0.196 # IP address or hostname
    token: AAA

tariffs:
  currency: CHF
  grid:
    type: fixed
    price: 0.2917 # EUR/kWh (default)
    zones:
      - days: Mo-So
        hours: 21-0
        price: 0.1881 # EUR/kWh (weekdays 2-6 hours)
      - days: Mo-So
        hours: 0-7
        price: 0.1881 # EUR/kWh (weekdays 2-6 hours)
chargers:
  - name: easee_home
    type: template
    template: easee
    user: AAA
    password: BBB
    charger: CCC
    timeout: 60s # optional
    authorize: 

vehicles:
  - name: vehicle_1
    title: Fake car
    type: custom
    soc:
      source: js
      vm: shared
      script: |
        var lp = state.loadpoints[0];
        if (lp.chargepower > 0) lp.vehicleSoc+=0.1; else lp.vehicleSoc-=0.1;
        if (lp.vehicleSoc < 15) lp.vehicleSoc = 80;
        if (lp.vehicleSoc > 80) lp.vehicleSoc = 15;
        lp.vehicleSoc;
    range:
      source: js
      vm: shared
      script: |
        var lp = state.loadpoints[0]
        var range = (100 * lp.vehicleSoc) / 15;
        range
    capacity: 100
    maxcurrent: 16

    
site:
  title: Maison
  meters:
    grid: my_grid
    pv: my_pv

loadpoints:
  - title: Couvert
    charger: easee_home
    mode: pv
    vehicle: vehicle_1

sponsortoken: XXX
interval: 60s
```

### Log details

```text
[main ] INFO 2025/03/21 15:17:47 evcc 0.201.1
[main ] INFO 2025/03/21 15:17:47 using config file: /root/.evcc/evcc.yaml
[db ] INFO 2025/03/21 15:17:47 using sqlite database: /root/.evcc/evcc.db
[main ] INFO 2025/03/21 15:17:47 listening at :7070
[main ] FATAL 2025/03/21 15:17:48 vehicle cannot create vehicle 'vehicle_1': cannot create vehicle type 'custom': decoding failed due to the following error(s): 'MaxCurrent' expected a map, got 'int'
[main ] FATAL 2025/03/21 15:17:48 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.201.1

jaytonic, 2025-03-21:
@andig But then if we have cars with different charging capabilities, we should still be able to have different values. Otherwise the plans for the one with the lowest maxcurrent will be wrong, right?

andig, 2025-03-21:
Right, but that's https://docs.evcc.io/en/docs/reference/configuration/vehicles#onidentify. Seems the documentation is a bit scattered for that aspect and maybe not entirely consistent.