evcc-io/evcc #19990: Charger not waiting before departure on plan

frankb-CZ, 2025-03-21:
### Describe the bug

I have set up a plan for 7AM departure. EVCC has clearly planned to start around 5AM. But when I select the mode PV or minPV it charges immediately. 

Why?

### Steps to reproduce

[evcc-20250321-173045-debug.log](https://github.com/user-attachments/files/19393562/evcc-20250321-173045-debug.log)

1. Schedule a plan
2. Set up static off peak times
3. Select PV mode and activate plan

...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  meter: trace

# unique installation id
plant: *****

interval: 30s # control cycle interval

meters:
- type: template
  template: wattsonic 
  id: 255  
  host: 192.168.1.71  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: wattsonic 
  id: 255  
  host: 192.168.1.71  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- name: battery1
  type: template
  template: wattsonic 
  id: 255  
  host: 192.168.1.71  
  port: 502  
  usage: battery  
  modbus: tcpip
  capacity: 11

chargers:
- type: template
  template: ellipro 
  ski: *****
  ip: 192.168.1.222  
  name: wallbox1

vehicles:
- type: template
  template: vw 
  title: Tiguan  
  user: *****
  password: *****
  vin: *****
  capacity: 20  
  timeout: 10s  
  name: ev1

loadpoints:
- title: Wallbox
  charger: wallbox1
  mode: pv

site:
  title: Zelenec
  meters:
    grid: grid1
    pv:
    - pv1
    battery:
    - battery1

eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
       *****
          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
         *****
          -----END CERTIFICATE-----
```

### Log details

```text
Logs in the attachment
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

VolkerK62, 2025-03-21:
`[lp-1  ] DEBUG 2025/03/21 17:24:13 smart cost active: 7.87`
you set a price limit and the actual price is below of that

frankb-CZ, 2025-03-22:
@VolkerK62 Yes but this is the off peak time which is valid for a whole night. Why the EVCC charges the car immediately after the price is lower than setpoint? It is a static price, it knows it wont change and charging can wait just shortly before departure… This is not a feature.

@andig Can you reopen please?