evcc-io/evcc #19993: Support for Luxtronik Heatpumps

Venxir, 2025-03-21:
I would like to see the Luxtronik system for heatpumps added to evcc so that I can determine when to start heating the water boiler or when to start my disinfection cycle. By setting certain parameters on Surplus Solar of low prices we could fully control when to start these heatpumps.

Luxtronik uses a TCP/IP based system on a custom protocol which uses some sort of registers which can be read and written. Currently these brands already use the Luxtronik controller:
- Alpha Innotec
- Siemens Novelan
- Roth
- Elco
- Buderus
- Nibe
- Wolf Heiztechnik

I looked into using relay's but these only give a basic control while actually controlling luxtronik will give much more precise control.

For Home Assisstant a similiar interface has been built with alot of information: https://github.com/BenPru/luxtronik.
Alot of info is based from: https://wiki.fhem.de/wiki/Luxtronik_2.0.

I also located some parameter mappings:
- https://github.com/Bouni/python-luxtronik/blob/main/luxtronik/parameters.py
- https://github.com/Bouni/python-luxtronik/blob/master/luxtronik/calculations.py
- https://github.com/Bouni/python-luxtronik/blob/master/luxtronik/visibilities.py

The most basic parameters to start with should be:
- **ID_Ba_Hz_akt** The mode of operation of the heating circuit, possible values are "Automatic", "Second heatsource", "Party", "Holidays", "Off"
- **ID_Ba_Bw_akt** The mode of operation of the hot water circuit, possible valus are "Automatic", "Second heatsource", "Party", "Holidays", "Off"
- **ID_Soll_BWS_akt** The set point for hot water generation, for example 50.0 for 50.0°C
- **ID_Einst_BA_Kuehl_akt** The mode of operation of the cooling circuit, possible values are "Automatic", "Off"

Venxir, 2025-03-21:
I might already have found a suitable modbus interface since the latest firmware version..Let me have look into this before i present this information...

bjospre2388, 2025-03-26:
I am using some additional features via HomeAssistant that might be interesting for implementation in evcc:

1. Time based (4h before sunset) warm water cycle (parameter: **ID_Soll_BWS_akt** / value: 60) to save warm water cycle by night
 --> evcc could do this better by using the PV forecast
2.  Parallel shift of the heat-curve between sunrise and sunset (parameter: **ID_Einst_WK_akt** / value: 1) to save PV-energy in the house
 --> evcc could do this better by using PV forecast and maybe in combination with dynamic pricing
3. Parallel shift of the heat-curve by night (time based) (parameter: **ID_Einst_WK_akt** / value: 0.5) to use low prices by night
 --> evcc could do this much better by using dynamic prices and charge planner

maybe these ideas help to develop more of the incredible nice evcc-features. Thank you so much for your work

Ph1975, 2025-04-02:
i own an Alpha Innotec Heatpump and can be beta tester if needed

closed 2025-04-15 by github-actions[bot]

daniel309, 2025-05-24:
Die luxtronik kann jetzt modbus. Nennt sich SHI Funktion und muss manuell aktiviert werden. Wurde wohl wegen ENWG 14a gemacht.

Details und Register Doku:

https://github.com/raibisch/mylibs/blob/main/LuxModbusSHI/LuxtronikSHI.md

https://github.com/raibisch/mylibs/blob/main/LuxModbusSHI/Modbusregister.md

daniel309, 2025-05-29:
Ok, mich hats doch gejuckt. Bitte helft beim testen: https://github.com/evcc-io/evcc/pull/21516