evcc-io/evcc #20008: Ladebereit. Warte auf Fahrzeug - Easee Wallbox / Hyundai Ioniq

undauntedjogging, 2025-03-22:
### Describe the bug

Hallo Zusammen,

ich habe momentan vermehrt das Problem, dass mein EVCC die Meldung anzeigt 

Ladebereit. Warte auf Fahrzeug 
Setup: Tibber Pulse mit dynamischen Tarif
WB: Easee Wallbox
PV: Huawei Sun Box
Fahrzeug: Hyundai Ioniq 6

[evcc-20250322-122517-debug.log](https://github.com/user-attachments/files/19402581/evcc-20250322-122517-debug.log)

[image]

[image]

[image]

[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
vehicles:
  - name: Ioniq
    type: template
    template: hyundai
    title: Hyundai Ioniq
    user: x
    password: x
    language: en # 'de' für Deutsch und 'en' für Englisch (optional)
    capacity: 77
```

### Log details

```text
[main ] INFO 2025/03/22 12:11:40 evcc 0.201.1
[main ] INFO 2025/03/22 12:11:40 using config file: /config/evcc.yaml
[db ] INFO 2025/03/22 12:11:40 using sqlite database: /data/evcc.db
[main ] INFO 2025/03/22 12:11:40 listening at :7070
[main ] ERROR 2025/03/22 12:11:57 creating vehicle Ioniq failed: cannot create vehicle type 'template': cannot create vehicle type 'hyundai': login failed: Post "https://prd.eu-ccapi.hyundai.com:8080/api/v1/user/oauth2/token": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[lp-1 ] DEBUG 2025/03/22 12:11:57 set title: Garage
[lp-1 ] DEBUG 2025/03/22 12:11:57 set priority: 0
[lp-1 ] DEBUG 2025/03/22 12:11:57 set smart cost limit: 0.1
[lp-1 ] DEBUG 2025/03/22 12:11:57 set thresholds: {Enable:{Delay:1m0s Threshold:0} Disable:{Delay:3m0s Threshold:0}}
[lp-1 ] DEBUG 2025/03/22 12:11:57 set plan energy: 0kWh @ 0001-01-01 00:53:28 +0053 LMT
[lp-1 ] DEBUG 2025/03/22 12:11:57 set session energy limit: 0
[lp-1 ] DEBUG 2025/03/22 12:11:57 set session soc limit: 0
[lp-1 ] DEBUG 2025/03/22 12:11:57 set soc config: {Poll:{Mode:charging Interval:1h0m0s} Estimate:0x4000fb2cf8}
[lp-1 ] DEBUG 2025/03/22 12:11:57 set default mode: pv
[lp-1 ] DEBUG 2025/03/22 12:11:57 set phases: 3
[lp-1 ] DEBUG 2025/03/22 12:11:58 charge total import: 485.847kWh
[site ] INFO 2025/03/22 12:11:58 site config:
[site ] INFO 2025/03/22 12:11:58 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2025/03/22 12:11:58 grid: power ✓ energy ✗ currents ✓
[site ] INFO 2025/03/22 12:11:58 pv 1: power ✓ energy ✓ currents ✗
[site ] INFO 2025/03/22 12:11:58 battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✓
[site ] INFO 2025/03/22 12:11:58 vehicles:
[site ] INFO 2025/03/22 12:11:58 vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site ] INFO 2025/03/22 12:11:58 tariffs:
[... 371 lines cut ...]
[lp-1 ] DEBUG 2025/03/22 12:23:36 set charge mode: off
[site ] DEBUG 2025/03/22 12:23:36 ----
[lp-1 ] DEBUG 2025/03/22 12:23:36 charge power: 0W
[lp-1 ] DEBUG 2025/03/22 12:23:36 charge currents: [0.011 0.011 0.011]A
[site ] DEBUG 2025/03/22 12:23:36 grid power: 283W
[site ] DEBUG 2025/03/22 12:23:36 grid currents: [0 0 0]A
[site ] DEBUG 2025/03/22 12:23:37 battery 1 power: 2529W
[site ] DEBUG 2025/03/22 12:23:38 pv 1 power: 1938W
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

201.1

undauntedjogging, 2025-03-22:
**[lp-1 ] WARN 2025/03/22 12:44:48 charger logic error: current mismatch (got 32A, expected 16A)**
[lp-1 ] DEBUG 2025/03/22 12:44:48 plan: charge 2h15m52s between 2025-03-22 12:44:48 +0100 CET until 2025-03-22 18:00:00 +0100 CET (power: 11040W, avg cost: 0.189)
[lp-1 ] DEBUG 2025/03/22 12:44:50 max charge current: 16A
[site ] DEBUG 2025/03/22 12:45:17 ----

Woher kommt, dass er 32A erhält?

undauntedjogging, 2025-03-22:
Das Problem scheint die Authorisierung der Wallbox zu sein , evcc gibt die Aktivierung nicht richtig weiter 
Mache ich diese aus in der Easee App scheint es zu klappen.

andig, 2025-03-22:
> dass mein EVCC die Meldung anzeigt

Was für eine Meldung?