evcc-io/evcc #20014: evcc cannot boot / start if charger is not connected

Krusty1977, 2025-03-22:
### Describe the bug

[image]

### Steps to reproduce

1. disconnect charger
2. restart evcc
3. it does not start.
...

### Configuration details

```yaml
configured over gui
```

### Log details

```text
See screenshot
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-03-22:
Yes, thats the current status.

closed 2025-03-22 by andig

Krusty1977, 2025-03-22:
works as designed? Will the add-on come up if the charger will be connected? or do I have to restart evcc manually again?